cncjs/cncjs #242: Cannot zero the z work coordinate

ENGR-mike, 2017-12-16:
I'm having trouble with operating my CNC mill using the Shop_Floor_Tablet UI and the TinyWeb UI.
The following are specifics about my installation:

CNCjs: 1.9.12
Node.js: 8.7.0
NPM: 5.4.2
I installed it via NPM. I am using an xPRO v3 controller, but that is irrelevant at this point since I cannot get cncjs to run on the pi. My pi is a brand new Raspberry Pi 3 model B running Raspbian (i.e Debian) Stretch (I think Stretch is version 9).

When setting the work zero position using either UI, I cannot zero the Z coordinate.  Also, the Z coordinate display does not change when I jog the Z axis on my machine.  However, when I jog the Y axis, not only do the Y coordinates change in the display, but the Z coordinates change.

Without being able to see my Z axis position, I cannot establish the work zero position.  I'm having the problem with both the Shop_Floor_Tablet UI and the TinyWeb UI.

MitchBradley, 2017-12-17:
I just fixed that bug in cncjs-shopfloor-tablet.  Release v1.0.4 should work correctly in that respect.

For cncjs-pendant-tinyweb, the problem was fixed 2 weeks ago in release v1.2.3, but you might as well get release v1.2.4 instead, which fixes both that and also the redirect bug.

ENGR-mike, 2017-12-17:
Thanks!  I'll install the updates later.
I'm heading out to dinner right now.  When I get some time, I'd like to discuss the sequence of steps you go through when setting up a job from a PC, then switching to the tablet to run the job at the mill.  I had some problems loading the g code in CNCjs on the PC, then switching to the tablet.  After setting my work zero, I had difficulty getting the g code to execute when pressing the start button.  If you have any thoughts about this, please feel free to share them, but I won't respond immediately.

ENGR-mike, 2017-12-17:
I installed the updated files for tablet and tinyweb and all works as expected now regarding the Z coordinate.  However, I'm still having problems starting an operation from the tablet once I've loaded a g code fle from the desktop.  My process is as follows:
1. I open the tablet UI on my iPad and setup the work zero
2. Go to my desktop, open the full UI and establish a connection
3. Load the g code file
4. Go back to the iPad in the shop and try to run the g code file by pressing start
After step 4 above, sometimes the tablet UI sort of freezes and won't respond to pressing start to run the g code file.  Sometimes I cannot even jog the machine or start a homing cycle.  Usually I have to back out of the tablet UI close the connection and reconnect, or sometimes refresh the tablet UI page.

Should I do things in a different order?
I'm going to bed now, so I'll check back tomorrow.

MitchBradley, 2017-12-17:
I don't know.  It works for me.  It is hard to debug "sometimes", especially when it doesn't reproduce here.

One thing you might try, to be closer to my workflow:  I typically load files directly from the tablet UI, instead of using the full UI as part of the process.  The way to do that is to set up a "watch directory" on the Pi and put your GCode files there.  To set up a watch directory, edit the file /home/pi/.cncrc and add this line as the next to last line, before the closing brace:

``` 
   "watchDirectory": "/home/pi/GCode"
```

Then create that directory with:

```
mkdir /home/pi/GCode
```

After you do that and restart cncjs, you can use tablet's file dropdown to pick a file to run.  Of course you need to put some gcode files in that directory.  I personally use WinSCP as an easy way to copy files from my PC to the Pi.

ENGR-mike, 2017-12-17:
I'm glad you mentioned the watch directory.  I was wondering about that.  I'll have to set it up and do it that way.

ENGR-mike, 2017-12-17:
Are there any other lines of coded I should put in the .cncrc file?  The following webpage on NPM shows an extensive number of lines in the .cncrc file.
https://www.npmjs.com/package/cncjs

MitchBradley, 2017-12-17:
That's sort of an open-ended question since I know very little about what you want to do.  I will try to explain what the cncrc options do so you can decide for yourself.

- ports - Lets you specify the names of any serial ports that, for whatever reason, are not found by the automatic serial port listing code.  The reason for this is because, on Linux and some other systems, there is no easy way to find every device name that is for a serial port.  There are some naming conventions that usually work, but every now and then somebody will choose an off-the-wall name that doesn't follow the conventions.  The ports key lets you work around that by manually declaring that a given device is in fact a serial port.  One specific use for this is for the GRBL simulator which uses a port named /dev/ttyFAKE ; unfortunately the GRBL simulator doesn't work well enough to be very useful.  It would be nice if you could, in addition to adding ports, restrict the set of ports to the ones that you know are actually connected to CNC controllers. But, alas, that is not how this config key works.
- baudrates - Lets you specify additional non-standard baud rates for oddball controllers. Again, it would be nice if you could restrict to the actual baud rate that your CNC controller is known to use, but that's not how this works.
- watchDirectory - Lets you specify a directory on the server that contains GCode files.  Very useful.
- accessTokenLifetime - How long before the security token expires.  You can probably make the system more secure in some theoretical sense by shortening this from the default of 30 days.
- allowRemoteAccess - If you set this to **true**, you can access the cncjs server from a computer on a different network - perhaps even across the Internet.  Of course that opens you up to attackers.
- controller - Valid values are "", "Grbl", "Marlin", "g2core", "TinyG", "Smoothie". I think the way this works is that, if the value is anything other than "", only the specified controller code is loaded.  If the value is "", the code for all of the controllers is loaded.  It seems like a good idea to me to "turn off" all the controllers that are not actually present on your system, thus avoiding confusing the operator with options that are cannot possibly work on a given system.
- state - This has the sub-property "checkForUpdates".  In the full cncjs UI, there is a Settings (gears) button in the upper left corner.  There is a General tab with an "Automatically check for updates" checkbox.  The state/checkForUpdates property is where that checkbox setting is saved.
- commands - This is where the values for the Settings/Commands tab (see above) are stored.  This appears to be a feature for running specific OS commands, but I'm not sure how to invoke that feature.  I don't see any facility in the UI for using it.
- events - This is an apparently-undocumented feature for running specified system commands or issuing GCode commands when certain things happen.  You can set up events from the Settings/Events tab.
users -This is where the users that are set up via Settings/My Account are stored.
macros - This is the storage for macros that are set up via the Macros widget in the full UI.

cheton, 2017-12-19:
To invoke commands, you can open the dropdown to execute a specific program on the server.
[image]

closed 2017-12-23 by ENGR-mike