cncjs/cncjs #243: 1.9.12 problems after upgrade from 1.9.11

flebas, 2017-12-17:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [X] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [X] Windows
- [ ] Mac
- [ ] Linux

I had cncjs 1.9.11 working well with TinyG. 
I upgraded to 1.9.12. Program starts but it seems there is a communication problem. 
Com port is open ans I can see in console : 
_CNCjs 1.9.12 [TinyG]
Connected to COM5 with a baud rate of 115200_
but commands from CNCjs are ignored. The only CNCjs button that is working is Reset. 
Commands typed in console are ignored too. 
I suspect a problem in communication layer, since it seems to have changed. 
I have unsinstalled and reinstalled CNCjs, manually cleared cache directories, and restarted computer, without improvement.

cheton, 2017-12-18:
/cc @reconbot

I upgraded [node-serialport](https://github.com/node-serialport/node-serialport) package from v4 to v6 in CNCjs 1.9.12, and I did discover an issue that only happened on Windows after the upgrade.

I made several tests among three latest releases of node-serialport v4, v5, and v6. The node-serialport v6 seems to have problems on Windows when connecting to an Arduino Due board, but it's working good with my Arduino Uno and Smoothieboard.

Following are my test results:

#### serialport@4.0.7 (Arduino Due + g2core)
[image]

#### serialport@5.1.0-beta5 (Arduino Due + g2core)
[image]

#### serialport@6.0.4 (Arduino Due + g2core)
No response
[image]

#### serialport@6.0.4 (Arduino Uno R3 + Grbl)
[image]

cheton, 2017-12-18:
Hi @reconbot
Is there any options I can try to investigate what may cause this problems to node-serialport v6? Thanks.

flebas, 2017-12-18:
Hi Cheton, 
It looks like no character is sent to TinyG from CNCjs. 
When starting CNCjs or opening serial port I can see in console : 

CNCjs 1.9.12 [TinyG]
Connected to COM5 with a baud rate of 115200
{ej:1}
{jv:4}
{qv:1}
{sv:1}
{si:100}
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,spe:t,spd:t,sps:t,cof:t,com:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
{sys:n}
{mt:n}
{pwr:n}
{qr:n}
{sr:n}

But the commands I am typing are ignored. 

I can try to investigate with serialport-list and serialport-term and TinyG but I don't know how to get or build them. I have VS 2015.

cheton, 2017-12-18:
All above json commands are sent by CNCjs while a serial connection is established, see the code at https://github.com/cncjs/cncjs/blob/master/src/app/controllers/TinyG/TinyGController.js#L582-L659

It's probably a serial communication issue with node-serialport v6, although I can temporarily roll back to v4 or v5, I would like to know if there is any workaround option to mitigate this issue.

flebas, 2017-12-18:
I have not found any workaround. I can investigate the problem but I need some help to get the project running in Visual Studio. I have already done some serial port development with other languages.

cheton, 2017-12-19:
@flebas 

It might be related to this issue: https://github.com/node-serialport/node-serialport/issues/1399, and it's also documented in https://github.com/node-serialport/node-serialport/blob/master/UPGRADE_GUIDE.md.

I will try to enable `rtscts` in open options by tonight.

flebas, 2017-12-19:
@cheton 
I think you found it. I have set TinyG to use Xon/Xoff and I changed serial port flow control in Windows to Xon/Xoff, and version 1.9.12 seems to work.

cheton, 2017-12-19:
Serial port options is currently not configurable in CNCjs 1.9.12. I will make another release to provide the following flow control options:

* **rtscts** _optional, default: false_
* **xon** _optional, default: false_
* **xoff** _optional, default: false_
* **xany** _optional, default: false_

cheton, 2017-12-19:
Enabling "rtscts" option to turn on hardware flow control can fix this issue on Windows.

[image]

I will turn on this option for future releases.

flebas, 2017-12-19:
It works. Thank you for fast and efficient support.

swpaxton66, 2017-12-29:
Hi.  I tried your 1.9.13 release and no matter what I do I cannot connect to tinyG.  Same issue as at the start of this thread.  I've tried all options available (I think) and still cannot communicate with the tinyG.

Any suggestions?

Thanks,

S

swpaxton66, 2017-12-29:
Sorry.  It is connecting now.  Not sure what the issue was.

Thanks

cheton, 2017-12-29:
The 1.9.13 release you downloaded is a development build, not an official release, but you can still use it for testing. Thank you.

cheton, 2017-12-29:
Just to let you know you have to "Enable hardware flow control" from the Connection widget, it's not turned on by default since it may cause problems for certain boards.

swpaxton66, 2017-12-29:
Thank you.  I am using it because I need to turn off the error for tinyG.  Is there an official release that I can use to turn off errors in tinyG?

Thanks

swpaxton66, 2017-12-29:
and for clarification I use the desktop app

cheton, 2017-12-29:
I need to make some enhancements to the Axes widget in the following days, the official release of 1.9.13 may be available after one or two weeks. You can try the pre-release build in advance, and report me issues if you encountered any problems.

cheton, 2017-12-29:
The latest build (a.k.a. nightly build) is available here:
https://github.com/cncjs/cncjs/releases/tag/master-latest

cheton, 2018-02-03:
Fixed in 1.9.13

closed 2018-02-03 by cheton