cncjs/cncjs #244: G38.2 probe with TinyG

flebas, 2017-12-18:
#### Description

CNCjs G38.2 probe sequence does not work well with TinyG.

The CNCjs sequence is 👍 

; Z-Probe
G91
G38.2 Z-15 F20 (probe depth)
G90
; Set the active WCS Z0
G10 L20 P2 Z10 (touch plate thickness)
; Retract from the touch plate
G91
G0 Z4
G90

TinyG G38.2 implementation is run in G53 machine coordinate system. The G91 and G90 has no effect. 

If the machine Z zero is on top, and current Z position is say -30, the command _G38.2 Z-15 F20_ will move the Z axis up. 

The _G10 L20 P2 Z10 (touch plate thickness)_ command add touch plate thickness to machine origin but ignores the Z position when touching. 

I think that the correct pseudo code for TynyG would be : 

; Z-Probe
G38.2 Z[GetCurrentAbsoluteZPosition-ProbeDepth] F20
; Set the active WCS Z0
G10 L20 P2 Z[GetCurrentAbsoluteZPosition-TouchPlateThickness]
; Retract from the touch plate
G91
G0 Z4
G90

It would be nice to be able to probe on X and Y axis too. 

#### Versions

- CNCjs: 1.9.11
- Node.js: N/A
- NPM: N/A

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2017-12-19:
TinyG 0.97 (firmware build 449.01) added support for `G10 L20`. In previous master version (incl. TinyG 0.96 build 380.08 and TinyG 0.97 build 440.18), the `G10 L20` command appears to work just like `G10 L2` (offset).

http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L2_
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L20

What's your TinyG firmware version and build?

flebas, 2017-12-19:
My TinyG firmware is 0.97 build 440.20. This is the current master version.

flebas, 2017-12-19:
There is an error in pseudo code in my first message, I type G10 L20 instead of G10 L2. It should read : 

; Z-Probe
G38.2 Z[GetCurrentAbsoluteZPosition-ProbeDepth] F20
; Set the active WCS Z0
G10 L2 P2 Z[GetCurrentAbsoluteZPosition-TouchPlateThickness]
; Retract from the touch plate
G91
G0 Z4
G90

cheton, 2017-12-19:
It's fine with me to run **G38.2** command in absolute position. I will make the change in CNCjs 1.9.13.

Will you have chance to upgrade TinyG firmware to 0.97 build 449.02? This has added support for `G10 L20` command.

https://github.com/synthetos/TinyG/blob/master-449/firmware/tinyg/tinyg.h#L51-L52

flebas, 2017-12-19:
I can upgrade TinyG firmware to 449.02 but I don't know how to get it. The only edge version listed on http://synthetos.github.io/ is 449.01

cheton, 2017-12-19:
449.01 is also good to use.

flebas, 2017-12-19:
I have installed 449.01 and Z-Probe is OK. 
Thank you.

cheton, 2017-12-25:
@flebas What's about the G38.2 implementation in firmware build 449.01? Will it take G90 and G91 into consideration?

flebas, 2017-12-25:
G90 ang G91 are not taken in consideration with build 449.01.
The axis value in G38.2 command is in absolute coordinate system. 
https://github.com/synthetos/TinyG/files/139431/TinyG.wiki.-.G38_2.Probe.pdf
Generally speaking, I have noticed strange behavior with 449.01 (don't know if it come's from TinyG or CNCjs 1.9.12) but I would prefer to be able to use CNCjs Probe commands with build 440.20 (use of G10 L2 with appropriate computed value).

cheton, 2017-12-26:
@flebas 

Currently I don't have a Z-probe wire connected to my board for physical testing. Could you try to create a macro as shown below for testing? I will merge it back to the probe widget if it works for you. Thanks.

```
; Wait until the planner queue is empty
%wait

; Set user-defined variables
%PROBE_DEPTH = 10
%PROBE_FEEDRATE = 20
%TOUCH_PLATE_THICKNESS = 10
%RETRACTION_DISTANCE = 4

; Move in a straight line to the target point in the absolute coordinate system
G90
G38.2 Z[posz - PROBE_DEPTH] F[PROBE_FEEDRATE]

; A dwell time of one second
G4 P1

; Set Z offset for the specified coordinate system
G10 L2 P1 Z[posz - TOUCH_PLATE_THICKNESS]

; Retraction
G91
G0 Z[RETRACTION_DISTANCE]
G90
```

flebas, 2017-12-26:
Hi Cheton, 
I cannot test before the end of the week. I shall report results. 
Thank you.

cheton, 2017-12-26:
Great! It will definitely help if you or someone can help test this macro. Thanks.

flebas, 2017-12-30:
Hi Cheton, 

Hi have switched back to firmware 440.20 and this fixed strage behaviour. Since it is related to motion and acceleration, I prefer not use edge firmware. 

I have tested the macro. It does move tool correctly for z-probe but the Z offset setting is wrong. 
Below is the console capture of macro running. I have removed extra {"sr":{"posz":...}} lines. 
Starting position is Z-35. 

G4 P0.5 (%wait)
{"qr":32}
G90
{"rx":247}
{"r":{},"f":[1,102,6,7584]}
G38.2 Z-45 F20
{"err":{"code":102,"msg":"Bad number format"}}
G4 P1
{"r":{},"f":[1,0,4,4397]}
G10 L2 P1 Z-45
G91
{"r":{},"f":[1,0,15,74]}
G0 Z4
G90
{"sr":{"coor":0,"stat":7}}
{"qr":32}
{"qr":31}
{"sr":{"posz":-35.024,"vel":20.00}}
{"sr":{"posz":-35.056}}
...
{"sr":{"posz":-38.886}}
{"sr":{"posz":-38.910,"vel":0.02}}
{"r":{"prb":{"e":1,"x":20.000,"y":20.000,"z":-38.910,"a":0.000,"b":0.000,"c":0.000}},"f":[1,0,0,1397]}
{"sr":{"vel":0.00,"coor":1,"stat":3}}
{"qr":32}
{"r":{},"f":[1,0,6,4399]}
{"qr":32}
{"r":{},"f":[1,0,15,74]}
{"sr":{"posz":6.090}}
{"r":{},"f":[1,0,4,4397]}
{"sr":{"dist":1}}
{"r":{},"f":[1,0,6,4399]}
{"sr":{"vel":0.00,"stat":5}}
{"qr":31}
{"r":{},"f":[1,0,4,4397]}
{"sr":{"posz":6.119,"vel":58.69}}
...
{"sr":{"posz":10.089,"vel":3.75}}
{"sr":{"posz":10.090,"vel":0.00,"dist":0,"stat":3}}
{"qr":32}

It seems that TinyG does not wait for G28.2 command completion  before executing following instructions, and G10 L2 command is run with posz value at the beginning of the macro and not after touch. 

I have found a workaround in splitting probe in 2 macros : 

1st macro for probe : 

; Wait until the planner queue is empty
%wait

; Set user-defined variables
%PROBE_DEPTH = 10
%PROBE_FEEDRATE = 20

; Move in a straight line to the target point in the absolute coordinate system
G38.2 Z[posz - PROBE_DEPTH] F[PROBE_FEEDRATE]

2nd macro to set Z offset : 

; Wait until the planner queue is empty
%wait

; Set user-defined variables
%TOUCH_PLATE_THICKNESS = 10
%RETRACTION_DISTANCE = 4

; Set Z offset for the specified coordinate system
G10 L2 P1 Z[posz - TOUCH_PLATE_THICKNESS]

; Retraction
G91
G0 Z[RETRACTION_DISTANCE]
G90

cheton, 2017-12-31:
I found an error code `{"err":{"code":102,"msg":"Bad number format"}}` in the status report. Can you help identify which command will cause such error?

Second, will it work by increase G4's P value to a large number? It's right after the line of G38.2.

flebas, 2017-12-31:
Hi Cheton, 
I have tested the macro. It does not work. 

I think that posz is not updated after G38.2 command, so G10 L2 command

cheton, 2017-12-31:
I just came up with an idea. How about replacing **G4 Px** with **M0**? It will pause the job and prompt a modal dialog asking user to resume execution.

flebas, 2017-12-31:
Error code {"err":{"code":102,"msg":"Bad number format"}} is caused by yhe line 
G4 P0.5 (%wait)
which is generated by the %wait macro command. 
G4 P0.5 works well, I think TinyG does not like the comment after the number. 
Replacing %wait with G4 P1 fix the error but does not change result. 

Increasing delay after G38.2 command does not change result. The console output is below. 
It seems that there is a bug in TinyG G38.2 implementation and that G38.2 command is not blocking and that following GCode instruction are executed before the G38.2 command terminates. 
G4 P1
{"r":{},"f":[1,0,6,4399]}
G90
G38.2 Z-46 F20
{"qr":32}
G4 P30
{"r":{},"f":[1,0,4,4397]}
G10 L2 P1 Z-46
G91
{"r":{},"f":[1,0,15,74]}
G0 Z4
G90
{"sr":{"coor":0,"stat":7}}
{"qr":32}
{"qr":31}
{"sr":{"posz":-36.026,"vel":20.00}}
{"sr":{"posz":-36.059}}
...
{"sr":{"posz":-39.790}}
{"sr":{"posz":-39.807,"vel":0.02}}
{"r":{"prb":{"e":1,"x":20.000,"y":20.000,"z":-39.807,"a":0.000,"b":0.000,"c":0.000}},"f":[1,0,0,7632]}
{"sr":{"vel":0.00,"coor":1,"stat":3}}
{"qr":32}
{"r":{},"f":[1,0,7,4400]}
{"qr":32}
{"r":{},"f":[1,0,15,74]}
{"sr":{"posz":6.193}}
{"r":{},"f":[1,0,4,4397]}
{"sr":{"dist":1}}
{"r":{},"f":[1,0,6,4399]}
{"sr":{"vel":0.00,"stat":5}}
{"qr":31}
{"r":{},"f":[1,0,4,4397]}
{"sr":{"posz":6.218,"vel":51.62}}
...
{"sr":{"posz":10.192,"vel":5.36}}
{"sr":{"posz":10.193,"vel":0.00,"dist":0,"stat":3}}
{"qr":32}

It works if I replace the wait after G38.2 with a tool change M6. 
I don't know if it is possible to customize the M6 message. 

The complete working macro for TinyG is : 

; Wait until the planner queue is empty
G4 P0.5

; Set user-defined variables
%COORD_SYSTEM = (modal.wcs.slice(1) - 53)
%PROBE_DEPTH = 10
%PROBE_FEEDRATE = 12
%TOUCH_PLATE_THICKNESS = 10
%RETRACTION_DISTANCE = 4

; reset Z-offset
G10 L2 P[COORD_SYSTEM] Z0
; slow small move to update posz after G10 L2 (TinyG bug ?)
G91
G1 Z-0.01 F[PROBE_FEEDRATE]
G90

;wait for posz update
G4 P1

; Move in a straight line to the target point in the absolute coordinate system
;G38.2 implemented with absolute coordinates (G53) in TinyG, no need to G90
G38.2 Z[(posz - PROBE_DEPTH).toFixed(3) * 1] F[PROBE_FEEDRATE]

;wait for probe completion 
M6

; Set Z offset for the specified coordinate system
G10 L2 P[COORD_SYSTEM] Z[(posz - TOUCH_PLATE_THICKNESS).toFixed(3) * 1]

; Retraction
G91
G0 Z[RETRACTION_DISTANCE]
G90

flebas, 2017-12-31:
I have had a similar idea with M6. 
It works the same with M0 instead of M6. 
Is it possible to customize the message ?

cheton, 2017-12-31:
Sure. I will use M0 or add a "%pause" command in 1.9.13 that allows to display custom message to inform tool probe completion.

cheton, 2017-12-31:
Will G38.2 perform differently if not doing so after resetting Z-offset?

```
; slow small move to update posz after G10 L2 (TinyG bug ?)
G91
G1 Z-0.01 F[PROBE_FEEDRATE]
G90
```

flebas, 2017-12-31:
Resetting Z offset is needed since G38.2 use absolute coordinates, and posz is given in the current work coordinate system. See below. 
The small move is used as a workaround to update posz, that was not updated after Z-offset reset, ven with a G4 wait command. 

My machine has a Z axis with 85 mm travel. Zero is on top : ztm=0 and ztn=-85. 
First probe : I manually set the probe within 10 mm above touch plate : Z-36
Z-probe is run from current position : G38.2 Z-46 F20
Z-offset is adjusted after touch G10 L2 P1 Z-46.144
Z position after running the probe macro is 14 (touch plate thickness + retractation). 

If I run again the probe macro without resetting Z-offset, the target Z will be 4 (14 - 10) and command will be : G38.2 Z4 F20
This Z value is outside limits. 
Even if there is no limit problems, the Z axis can move upward during probe if Z value in G38.2 is above absolute position.

flebas, 2018-01-11:
Hello, 

I think I have found why the initial macro is not working. 

G-code for G10 L2 Pn line is generated using the posz value when the macro is started, not the posz value after completion of G38.2 command. 
Adding a M0 after G38.2 works but need manual input on CNCjs. 

Is it possible to get the macro engine to wait for current Gcode line completion before evaluating and sending the next macro line ?

cheton, 2018-02-06:
@flebas 

Just found a way to avoid manual input while waiting for operation completion. I will let you know when it works.

cheton, 2018-07-18:
Use `%wait` command to wait for completion without the need for user input

bcoate, 2020-01-13:
It might be because of the length of the planner buffer, but using g2core the "%wait" approach wasn't working for me when using tool change macros that others have posted. Adding M0 pauses to prevent values being stored before probing completed worked, but isn't ideal.

The workaround I came up with that has been working well for me was to load up a bunch of "garbage" gcode after the probe command, and before the z-axis value is stored, so the planner buffer would be full and the z-height value wouldn't be stored until after probing completed.

I used ~50 lines of alternating between G90 and G91. I'm sure there are many other ways to do this.