cncjs/cncjs #245: Confused by releases

MitchBradley, 2017-12-18:
Which system is cncjs-1.9.12-linux-armv7.deb supposed to work on?

I tried installing it on a Raspberry Pi 2 
```
pi@laser:~$ uname -m
armv7l
pi@laser:~$ dpkg --print-architecture
armhf
```
 but that failed because of unmet dependencies on libnotify4:armv7l and other packages that are actually installed, but they are listed as having the "armhf" architecture.  I think that armv7l is in the kernel architecture namespace, not the package architecture namespace.

So I edited the control file in the .deb file and replaced "armv7l" with "armhf".  After that, the installation succeeded, but when I tried to run cncjs, it just exited immediately with no messages.  I tried running it under strace but did not see any obvious reason for its exit.

I can upgrade using npm, but was hoping to avoid that because it often takes hours to complete, and sometimes dies in the middle of the upgrade due to network errors.

cheton, 2017-12-19:
If you're trying to install cncjs-1.9.12-linux-armv7.deb on a Raspberry Pi, you're supposed to run a graphical Linux desktop application. Actually, I haven't had chance to test the armv7l package on my Raspberry Pi. It was built with electron-builder running on Travis CI:

[image]

There are some discussions about `armhf` support:
https://github.com/electron-userland/electron-builder/issues/1230
https://github.com/electron/electron/issues/6055
https://github.com/electron/electron/issues/5706

So far I have no way to do unless new target platform is supported by electron-builder.

MitchBradley, 2017-12-19:
We need to find some good way to document all the different ways to run cncjs, and which release artifacts correspond to which usage scenarios.  As things currently stand, the situation is utterly confusing.

Now that I understand that the cncjs_1.9.12_armv7l.deb is an Electron build, I tried running cncjs inside an xterm on a Pi 2.  The first problem I encountered was a message saying that serialport.node is for a 64 bit system.  Further inspection showed that, not only is it 64-bit, it is x86_64, not ARM at all.  So somehow the electron-builder for armv7l is inserting an x86_64 binary instead of an ARM binary.

I got past that problem by rebuilding serialport.node from source on ARM and substituting that for the x86_64 one.  Then it failed with

```
A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find module 'babel-runtime/regenerator'
   at Module._resolveFilename (module.js:470:15)
   at Function.Module._resolveFilename (/opt/CNCjs/resources/electron.asar/common/reset-search-paths.js:35.12)
   ... and several more lines referencing main.js and module.js
```

I think the armv7l .deb should be removed from the released files for now.  I don't see how it can possibly work on any system in its current state.  Maybe when the electron team get serious about ARM support it can be added back.

cheton, 2017-12-20:
I'm busy with my day work recently and only have limited time for this project (may be one hour per day), so far I would like to put my major focus on development rather than spending time on completing documentation. I will be really grateful if you and others could help update Wiki documentation, especially the User Guide and Troubleshooting section, that will be helpful for novice users.

For the corrupted ARM build, I may have missed something that accidentally produced x86_64 binary, I will remove it from the releases page to avoid misuse.

AustinSaintAubin, 2017-12-20:
I will aim to give the wiki attention. I have had the same issue with time but will try to step-up.  I welcome others to help update the Wiki as well.

MitchBradley, 2017-12-20:
I am happy to help with the documentation.  Unfortunately I am often unsure about how things are supposed to work.  The documentation contains a lot of stale information so it can be difficult to figure out what is obsolete and what is not.  Sometimes I can work out the information from reading the source code, but there is a lot of that, and a lot of interaction with external software, hardware, and frameworks.  Piecing it all together is challenging and error prone.

For example, there is a broken link in the Docker section.  Is there a better link, or is the use with docker obsolete?

MitchBradley, 2017-12-20:
One thing that would help would be better naming conventions.  The names "cnc" and "cncjs" seem to be used for different things.  For starters, how about naming the server ("app") component "cncjs-server" to distinguish it from the "all-in-one" Electron app?

AustinSaintAubin, 2017-12-20:
@MitchBradley Good ideas. It might be worth while to have [a wiki page to track what needs to be updated on the wiki](https://github.com/cncjs/cncjs/wiki/Wiki-Todo). Please add and comment. I will try to manage the wiki rework and get this going... strong documentation is what makes projects strong.

cheton, 2017-12-20:
@MitchBradley 
How about changing the filename to **cncjs-desktop-{version}-{platform}-{arch}.{ext}**? That's pretty easy than changing project's folder structure.

@AustinSaintAubin 
GitHub Team Discussions is a place to share information with team members, it would be a better place for discussions.
https://github.com/orgs/cncjs/teams/cncjs-dev

MitchBradley, 2017-12-20:
Moving this discussion to the team place ...

cheton, 2017-12-26:
Drop armv7 support (https://github.com/cncjs/cncjs/commit/2a0a776e71e8e6e565d53ecdefd16d7af725796a)

closed 2017-12-26 by cheton