cncjs/cncjs #246: CNCjs detecting error in g code and tablet UI won't allow resuming

ENGR-mike, 2017-12-22:
CNCjs is detecting errors in lines of g code where there does not seem to be an error.  The lines of code are as follows:
G2 Y50.6825 Z-0.25 I0.3175
G2 Y30.6825 Z-0.25 I0.3175
G2 Y10.6825 Z-0.25 I0.3175 
Yet it processed a previous similar line without problems.  The similar line is as follows:
G2 X15.6362 Y70.6825 Z-4.25 I0.3175
The only difference I see is that the three lines of code causing the error don't have an x coordinate and the previous line does have an x coordinate.  Should that be a problem?
My installation is as follows:
CNCjs: 1.9.12
Node.js: 8.7.0
NPM: 5.4.2
I installed it via NPM. I am using an xPRO v3 controller, but that is irrelevant at this point since I cannot get cncjs to run on the pi. My pi is a brand new Raspberry Pi 3 model B running Raspbian (i.e Debian) Stretch (I think Stretch is version 9).

Also, when simulating this in the full UI, the full UI will allow me to resume and the g code file continues where it paused at the error.  However, when running the g code file with the tablet UI, it goes back to the beginning of the g code, so when I press resume it starts the operation over from the beginning of the g code file.

ENGR-mike, 2017-12-22:
Update:  I tried adding X0 into the three lines of code and it did not help.  An error was still detected by CNCjs.

MitchBradley, 2017-12-23:
It's impossible to interpret a G2 Arc without knowing the starting position that was established by preceding context.

To specify an arc you need 3 points - start end center. Even in the best case where all of X Y Z are listed in the G2 command, only the end point is fully known without preceding context. 

In addition to the inability to determine start and center without context, it's also impossible to interpret the Arc without knowing the modal work plane that is in effect

MitchBradley, 2017-12-23:
When you say "CNCjs is detecting errors", what exactly do you mean?  Are you seeing error reports in the serial console window?  If so, what exactly is the report text?

That information is crucial because, in many cases, CNCjs does not actually "detect" the error directly, but rather passes on error reports that are generated by Grbl or TinyG or whatever - and the "detection" of the error occurs in Grbl.  If that is what is happening, there is no way to determine from the CNCjs code what is going wrong; rather one must deep dive into the Grbl code - and for that one needs to know the Grbl version and sometimes the detailed machine configuration.

Including screen captures in error reports can be the difference between an easy and an impossible debug session.

ENGR-mike, 2017-12-24:
Yes, in the serial console window and it halts the processing of the g code asking if I want to resume.  It reports an error in the code and reports the line of code, which are the three lines I mentioned above.  If I see it again, I'll take a snip of it and post it.  I'll be glad to post the issue on the GRBL page.  However, I modified the operation and I'm no longer seeing it.

AustinSaintAubin, 2017-12-24:
If you are using `pm2` on your raspberry pi you might run `pm2 monit` before you start a job and watch to see if something in interfering with the cncjs instance.

MitchBradley, 2017-12-24:
So, since the problem is no longer reproducible, is it okay to close this ticket?

See #233 for some recent work that is probably relevant.  It is CNCjs policy to, by default, pause when the controller reports an error, but it is now possible to configure selective ignoring of specific errors.

Since the error occurred during a gcode arc, the most likely error is STATUS_GCODE_INVALID_TARGET (33) from these lines in grbl's "gcode.c"

```
            // Compute difference between current location and target radii for final error-checks.
            float delta_r = fabs(target_r-gc_block.values.r);
            if (delta_r > 0.005) { 
              if (delta_r > 0.5) { FAIL(STATUS_GCODE_INVALID_TARGET); } // [Arc definition error] > 0.5mm
              if (delta_r > (0.001*gc_block.values.r)) { FAIL(STATUS_GCODE_INVALID_TARGET); } // [Arc definition error] > 0.005mm AND 0.1% radius
            }
```
Basically what that means is that grbl is trying to work out the specifics of the arc from the starting point, ending point, and center, but it appeared that the radius from center to start is different than the radius from center to end.  One way that can happen is when roundoff error in the computation makes things appear different even though they are really close.  The code tries to handle "close enough", but the math for doing that can get tricky.

cheton, 2017-12-24:
> Also, when simulating this in the full UI, the full UI will allow me to resume and the g code file continues where it paused at the error. However, when running the g code file with the tablet UI, [...]

Hi @MitchBradley 

According to fact mentioned by @ENGR-mike, I can reproduce this issue by manually inserting an unsupported command in the G-code file. If there is an error encountered, I can resume the program execution from the main UI:

[image]

However, if I try to resume the job from tablet UI, it will always trigger the "gcode:start" command, therefore it will go back to the beginning of the G-code.
https://github.com/cncjs/cncjs-shopfloor-tablet/blob/master/src/index.html#L81
[image]

To fix this issue in the tablet UI, you have to determine whether to start or resume the job with respect to current workflow state, the workflow state can be retrieved by the 'workflow:state' event, see below:
https://github.com/cncjs/cncjs-shopfloor-tablet/blob/master/src/controller.js#L103

```js
// Workflow State
var WORKFLOW_STATE_RUNNING = 'running';
var WORKFLOW_STATE_PAUSED = 'paused';
var WORKFLOW_STATE_IDLE = 'idle';

// Start
if (cnc.controller.workflowState === WORKFLOW_STATE_IDLE) {
    cnc.controller.command('gcode:start');
    return;
}

// Resume
if (cnc.controller.workflowState === WORKFLOW_STATE_PAUSED) {
    cnc.controller.command('gcode:resume');
    return;
}
```

cheton, 2017-12-24:
You might have to update the code in index.html using above example:

```html
<div class="col-xs-2">
    <button type="button" class="btn btn-default btn-start" onclick="cnc.controller.command('gcode:start')">
        Start
    </button>
</div>
```

cheton, 2017-12-24:
@ENGR-mike 

There is a known issue I need to mention here, CNCjs uses the character-counting streaming protocol for sending G-code program to Grbl in an aggressive way, so the reported line of error might not be the exact line in your G-code file. The error might happen before or after the reported number of line.

MitchBradley, 2017-12-24:
@cheton - Thanks for looking into this.  I just woke up and was about to look at it myself.  I want to study all the different interactions between controller state and workflow state.   Ideally, it would be nice if the controllers were abstracted so completely that the UI code itself would not need to know which controller is in effect, but I think that will take some work to achieve.  Meanwhile, I want to handle the cases of of pausing/stopping/resuming both GCode flows and long jog operations.

ENGR-mike, 2017-12-24:
I guess that means the issue will remain open?

MitchBradley, 2017-12-24:
I'm starting to understand the full complexity of the situation.  The buffering in the CNC controller can lead to interesting state combinations.  For example, consider this trivial GCode program:

```
G21 G1 X40 F400
G0 X0
M30
```
The G1 command takes a finite amount of time to run on the CNC controller (6 seconds in this case, but in the general case a command could take a really long time).  Meanwhile, CNCjs has proceeded to the M30 command and has already entered the workflow:paused state.  So the total state, for a substantial amount of time, is "grbl:state = Run" and "workflow:state = paused".

A similar complicated situation can occur when Grbl encounters errors in the GCode program.

It is difficult to synchronize GCode execution in the Grbl CNC controller with status reporting and control in the CNCjs computer.  That's an inherent weakness of any system that is implemented with multiple computers.  The limitations of Grbl - Arduino resource constraints and serial line flow control difficulties - make the problem especially hard to solve well.  That's one reason why I use g2core on my mill.  g2core runs on an ARM processor with some memory and processor speed headroom, so there is at least some hope of fixing deep problems.

ENGR-mike, 2017-12-24:
Merry Christmas everyone!  Thanks for all of the help using CNCjs.  I've begun to make chips and am learning a lot about defining operations relative to mill capability, bit size, etc.  I will continue to do everything I can to help make CNCjs better.

ENGR-mike, 2017-12-24:
P.S.  I also submitted an issue over on the Shopfloor-Tablet page.

cheton, 2017-12-25:
Hi @MitchBradley 

To make things simple, you can rely on the workflow state to change each button's disabled state for controlling START (or RESUME), PAUSE, STOP actions, like below:
https://github.com/cncjs/cncjs/blob/master/src/web/widgets/Visualizer/WorkflowControl.jsx#L82-L139
```js
const isReady = canClick && gcode.ready;
const canRun = this.canRun();
const canPause = isReady && (workflow.state === WORKFLOW_STATE_RUNNING);
const canStop = isReady && (workflow.state === WORKFLOW_STATE_PAUSED);
const canClose = isReady && (workflow.state === WORKFLOW_STATE_IDLE);
```

The only most complicated one is the `canRun` function, you can refer to the code at https://github.com/cncjs/cncjs/blob/master/src/web/widgets/Visualizer/WorkflowControl.jsx#L82-L129 to see how it's implemented.

Below are the handler functions that are used for workflow control:
https://github.com/cncjs/cncjs/blob/master/src/web/widgets/Visualizer/index.jsx#L351-L419
```js
handleRun: () => {
    const { workflow } = this.state;

    if (workflow.state === WORKFLOW_STATE_IDLE) {
        controller.command('gcode:start');
        return;
    }
    if (workflow.state === WORKFLOW_STATE_PAUSED) {
         controller.command('gcode:resume');
    }
},
handlePause: () => {
     const { workflow } = this.state;
     controller.command('gcode:pause');
 },
 handleStop: () => {
    const { workflow } = this.state;
    controller.command('gcode:stop', { force: true });
},
handleClose: () => {
    const { workflow } = this.state;
    controller.command('gcode:unload');
    pubsub.publish('gcode:unload'); // Unload the G-code
}
```

cheton, 2017-12-25:
@MitchBradley 
For the synchronization issue due to buffering, it's not a Grbl's problem. Grbl provides two choices for sender to stream a G-code program. One is the simple **send-response streaming protocol**, and the other is the **character-counting streaming protocol**. I copied part of the description from Grbl's [wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-a-g-code-program-to-grbl):

---

#### Send-Response Streaming Protocol
The host PC interface simply sends a line of G-code to Grbl and waits for an ok or error: response message before sending the next line of G-code. So, no matter if Grbl needs to wait for room in the look-ahead planner buffer to finish parsing and executing the last line of G-code or if the the host computer is busy doing something, this guarantees both to the host PC and Grbl, the programmed G-code has been sent and received properly. 
#### Character-Counting Streaming Protocol
The main difference between this protocol and the others is the host PC needs to maintain a standing count of how many characters it has sent to Grbl and then subtract the number of characters corresponding to the line executed with each Grbl response. Suppose there is a short G-code program that has 5 lines with 25, 40, 31, 58, and 20 characters (counting the line feed and carriage return characters too). We know Grbl has a 128 character serial receive buffer, and the host PC can send up to 128 characters without overflowing the buffer. If we let the host PC send as many complete lines as we can without over flowing Grbl's serial receive buffer, the first three lines of 25, 40, and 31 characters can be sent for a total of 96 characters. When Grbl sends a response message, we know the first line has been processed and is no longer in the serial read buffer. As it stands, the serial read buffer now has the 40 and 31 character lines in it for a total of 71 characters. The host PC needs to then determine if it's safe to send the next line without overflowing the buffer. With the next line at 58 characters and the serial buffer at 71 for a total of 129 characters, the host PC will need to wait until more room has cleared from the serial buffer. When the next Grbl response message comes in, the second line has been processed and only the third 31 character line remains in the serial buffer. At this point, it's safe to send the remaining last two 58 and 20 character lines of the g-code program for a total of 110.

_RESERVATION:
If a g-code line is parsed and generates an error response message, a GUI should stop the stream immediately. However, since the character-counting method stuffs Grbl's RX buffer, Grbl will continue reading from the RX buffer and parse and execute the commands inside it. A GUI won't be able to control this. The interim solution is to check all of the g-code via the $C check mode, so all errors are vetted prior to streaming. This will get resolved in later versions of Grbl._

---

With current implementation, I chose to use character-counting as the default streaming protocol for Grbl and Smoothie to achieve high performance especially for low-cost Arduino boards, but it can be switched to use send-response protocol by changing only one line of code:

from
```js
this.sender = new Sender(SP_TYPE_CHAR_COUNTING, { /* options */ });
```
to 
```js
this.sender = new Sender(SP_TYPE_SEND_RESPONSE, { /* options */ });
```

If you all think this is necessary to be a configurable option, I can make it configurable so users can change the streaming protocol setting for Grbl or Smoothie according to their needs.

MitchBradley, 2017-12-25:
> To make things simple, you can rely on the workflow state to change each button's disabled state for controlling START (or RESUME), PAUSE, STOP actions, like below:

I wish it were that simple.  The problem is that the workflow state does not reflect jogging and MDI operations.  My milling machine has a 350 mm bed and can rapid at 500 mm/min, so long traverses can take many seconds.  It's important to be able to pause or stop jogs in case of a mistake; otherwise you might break an expensive bit or mill a gouge in a table, vise, or fixture.

My best analysis so far is:
```
// workflow.state == RUNNING does not account for jog and MDI
canPause = grbl.activeState == RUN;  
canStop =  grbl.activeState == HOLD || grbl.activeState == RUN;
canResume = (grbl.activeState == HOLD || (grbl.activeState == IDLE && workflow.state == PAUSED);
```

MitchBradley, 2017-12-25:
> For the synchronization issue due to buffering, it's not a Grbl's problem. 

I'm not talking about the serial buffer flow control, but rather the problem of determining how far the CNC controller has progressed in its execution.  That affects the interpretation of error messages and highlighting of gcode lines.  Grbl's serial buffer, planner buffer and machine movement speed all contribute to that.  TinyG has a way to report which GCode lines have actually been executed, but even that has the problem of how you identify lines in the absence of Nxx line numbers.

cheton, 2017-12-25:
So far line number detection is only added in TinyGController.js, but not in other controllers' code.

https://github.com/cncjs/cncjs/blob/master/src/app/controllers/TinyG/TinyGController.js#L292-L295
```js
// Replace line numbers with the number of lines sent
const n = this.sender.state.sent;
line = ('' + line).replace(/^N[0-9]*/, '');
line = ('N' + n + line);
```

Not sure if it's worth doing for other controllers, but it's easy to implement if you find it useful.

cheton, 2017-12-25:
> I wish it were that simple. The problem is that the workflow state does not reflect jogging and MDI operations. My milling machine has a 350 mm bed and can rapid at 500 mm/min, so long traverses can [...]

I suggest you move <kbd>Feed Hold</kbd>, <kbd>Cycle Start</kbd>, <kbd>Unlock</kbd>, <kbd>Reset</kbd> buttons from the dropdown menu to the top navigation for providing immediate operation to the controller, and the workflow buttons (i.e. START/RESUME, PAUSE, STOP) will only affect (or be affected by) the workflow state, so it won't mix workflow state and controller state together. What's about your thought?

MitchBradley, 2017-12-25:
From a user perspective, having so many different controls is confusing.  It requires the user to understand many details of the system implementation, and to think carefully under pressure.  CNC machines can be very dangerous.

My ultimate goal for my production UI is that, when running GCode, touching anywhere on the screen will immediately pause.  For jogging and MDI, the pause button will be somewhat smaller, but still very prominent and easy to hit.

CNC operators do not want to know that there are multiple computers with different requirements.  The user interface needs to focused on their needs, not on the implementation structure of the system.

cheton, 2017-12-26:
Close this issue as the state management will be lately enhanced in cncjs-shopfloor-tablet.

closed 2017-12-26 by cheton