cncjs/cncjs #250: Macro questions

flebas, 2017-12-30:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

Is there any documentation for macros other than online help displayed when clicking on "Macro variables" in Edit Macro dialog ?

Current questions are : 
1) how can I get P parameter value from %WCS (e.g. If WCS is G55 and I want to adjust offset with G10 L2 command, I need the P2 parameter). 

2) is there a way to round calculated values. 
Macro line G38.2 Z[posz - PROBE_DEPTH] F[PROBE_FEEDRATE]
is sent to TinyG as G38.2 Z0.34999999999999964 F20

3) how can I refresh the position value (posz) after changing offset ?

cheton, 2017-12-31:
1. There is a workaround way to get P by accessing the modal variables:
    ```
    %P = (modal.wcs.slice(1) - 53)
    G10 L20 P[P] X0
    ```
    *  `str.slice(1)` will extract the string and return a new starting from the begin index 1 (see [slice](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/String/slice))
    * Lines started with the % character will be treated a simple JavaScript expression for variable assignment.
    * An expression inside by `[]` will be replaced with its evaluated result.

2. Use `G38.2 Z[(posz - PROBE_DEPTH).toFixed(3) * 1] F[PROBE_FEEDRATE]` to round calculated values.
    * The [toFixed()](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Number/toFixed) method formats a number using fixed-point notation.
    * `number.toFixed(3) * 1` will format "1.000" to "1".

3. You cannot change the value of `posx`, `posy`, and `posz`, these will be overrided by current work position. It is hard to determine current work position at the right time, you may have to use a G4 dwell command to wait a certain amount of time for the position to be updated.

flebas, 2017-12-31:
Thank you very much.

thymjan, 2018-02-06:
Is it / would it be possible to call macros or MDI commands from within a cnc g-code file like sort of functions?

cheton, 2018-02-26:
@thymjan 

I didn't provide this option because it may cause unexpected result if a macro or MDI command failed to complete for some reasons, and the g-code file will have compatibility issues when running with other G-code senders.

closed 2019-05-09 by cheton