cncjs/cncjs #251: Machine position reports 0 in Axes widget

flebas, 2017-12-30:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

In the Axes widget, machine position values are always 0.000 for the 3 axis. Work position values are reported correctly. Is there any setting to adjust in CNCjs or TinyG ?

cheton, 2017-12-31:
When you jogged the machine, did you see "mpox", "mpoy", and "mpoz" reported to the console widget?

[image]

flebas, 2017-12-31:
No. Console output below. 

g91
{"r":{},"f":[1,0,4,4397]}
{"sr":{"dist":1}}
g0 x20 y20
{"r":{},"f":[1,0,11,70]}
{"sr":{"posx":0.003,"posy":0.003,"vel":57.85,"stat":5}}
{"qr":31}
{"sr":{"posx":2.550,"posy":2.550,"vel":2828.43}}
{"sr":{"posx":5.696,"posy":5.696}}
{"sr":{"posx":9.007,"posy":9.007}}
{"sr":{"posx":12.318,"posy":12.318}}
{"sr":{"posx":15.463,"posy":15.463}}
{"sr":{"posx":18.774,"posy":18.774}}
{"sr":{"posx":20.000,"posy":20.000,"vel":0.00,"stat":3}}
{"qr":32}
g91
{"r":{},"f":[1,0,4,4397]}

flebas, 2018-01-01:
I have tried to add mpox, mpoy and mpoz in status report without success. Maybe a limitation of 440.20 firmware.

cheton, 2018-01-02:
Could you try below status report commands to check if it will perform different result?

#### Full status report (incl. spe, spd, sps, cof, and com for edge-0.82.10)
```
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,spe:t,spd:t,sps:t,cof:t,com:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
```

#### Full status report (incl. spe, spd, sps, cof, and com for edge-0.82.10)
Same as above, but move **spe**, **spd**, **sps**, **cof**, and **com** to the last.
```
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,spe:t,spd:t,sps:t,cof:t,com:t}}
```

#### Remove edge support
```
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
```
####

flebas, 2018-01-02:
It does not change output. TinyG always respond OK but nothing change. 
It seems this is a TinyG problem.

flebas, 2018-01-03:
Update : if I send JSON config commands with CoolTerm, it works, and I get mpox, mpoy and mpoz in status reports when sending move commands. 
It seems the is with CNCjs when sending JSON commands : TinyG always respond OK without message.

cheton, 2018-01-04:
Ok! I will install 440.20 firmware to my TinyG board to check if I can reproduce this issue.

cheton, 2018-01-29:
Sorry for the late reply.

TinyG will report "Unrecognized command or config name" with current status report commands:

```
> {sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,spe:t,spd:t,sps:t,cof:t,com:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
{"r":{"sr":{"line":true,"vel":true,"feed":true,"stat":true,"cycs":true,"mots":true,"hold":true,"momo":true,"coor":true,"plan":true,"unit":true,"dist":true,"frmo":true,"path":true,"spe":true}},"f":[1,100,190,4864]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
```

Then, I found that the machine position will work if I send the line without **spe**, **spd**, **sps**, **cof**, and **com**
```
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
```

To resolve this issue, I think it may need to send "{sys:n}" command first to determine the firmware build, and then send different status report commands accordingly.
```
> {sys:n}
{"r":{"sys":{"fb":440.20,"fv":0.970,"hp":1,"hv":8,"id":"5W3249-QUS","ja":100000,"ct":0.0100,"sl":0,"st":0,"mt":2.00,"ej":1,"jv":4,"js":1,"tv":1,"qv":1,"sv":1,"si":100,"ec":0,"ee":0,"ex":1,"baud":5,"net":0,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,8,7844]}
```

flebas, 2018-01-29:
Hi Cheton, 
Thank you for searching. 
It should work. 
For an unknown reason I cannot send JSON commands from CNCjs console (no reply or only tinyg ok), but if I send the modified status report command from CoolTerm and then start CNCjs, I get the machine position in the Axes widget. 
Maybe the JSON console problem is related with serial port, I am using CNCjs 1.9.12 with a patch to set rtscts to true.

MitchBradley, 2018-01-29:
@cheton - Long term, it is better to ask specific questions about what is supported, instead of looking for a version ID or firmware  build number and inferring from that.  If you infer from build numbers, you ultimately end up with a maintenance problem - unwieldy version number tables that require unending updates - as build numbers cycle, especially if the number schema is complicated or changes with private forks.

In this case, I think you can "probe" for which report elements are supported, as follows:

> {sr: {spe:t}}

Then look for an error response to see if that element is supported.

cheton, 2018-01-30:
@MitchBradley 

Thanks for your suggestion. That will definitely better than comparing the build version.

As for current implementation, it does not stream initialization commands to the feeder, I will try to find a simple way for probing supported commands.

cheton, 2018-02-03:
Fixed in 1.9.13

closed 2018-02-03 by cheton