cncjs/cncjs #252: Support for Send-Response Streaming Protocol in Grbl controller for troubleshooting

andrewismoody, 2018-01-05:
#### Description

Looking at the code, it appears that GrblController is hard-coded to Character Counting Streaming Protocol.  I have compiled Grbl to enable Echo commands on my Arduino, but this has proven to be useless in Character Counting mode, as the Echo does not correspond with the most recently sent command (as expected).  Would like to be able to switch between Character Counting and Send-Response protocol for Grbl.  The overall idea here is to compare the command that was sent with the echo response from Grbl to determine whether any data corruption is occurring.  Ideally, this could be coupled with a special mode in the software (troubleshooting mode?) to automatically compare the sent and received gcode and alert the user if there is a mismatch.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.12.3
- NPM: 5.6.0
- Raspbian 4.9.35-v7+ #1014
- Grbl v1.1 [REPORT_ECHO_LINE_RECEIVED enabled]

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux
- [x] Raspbian

andrewismoody, 2018-01-05:
Brilliant! I'll give it a try.  I'm crap at Node development, apparently, because I didn't see in the controller where SP_TYPE_SEND_RESPONSE was implemented.  Thanks for the heads up.

MitchBradley, 2018-01-05:
The coding learning curve for this app is really steep.  I have been programming in C and many other languages for 45 years, and it has been a real slog getting up to speed on the current state of JavaScript development.  You don't just program in JS, instead you also have to cope with Node and NPM and Babel and lodash and jQuery and bootstrap and webpack and numerous other libraries - not to mention the ancillary services like eslint and stylint and coveralls and Travis CI.  You have to switch mental gears to a different knowledge domain every 5 minutes. It's getting to the point that you have to live and breathe industrial-strength JS development to have a prayer of keeping up.

MitchBradley, 2018-01-05:
I switched to SP_TYPE_SEND_RESPONSE just to make sure it basically works; unsurprisingly, it does.  I didn't stress test anything.

andrewismoody, 2018-01-05:
Awesome, this does seem to work.  Now to run a job and diff the log to see where the data corruption occurs.

MitchBradley, 2018-01-05:
Why do you suspect data corruption?  Are you seeing strange error reports, or incorrect machine motion?

andrewismoody, 2018-01-05:
Random cut failures for no apparent reason.  I suspect it's not placing the tool head in the right position.  I don't get errors from Grbl, so it's not significant, but enough to make things go wonky for no apparent reason.  I had lots of issues previously to the point where Grbl would throw errors, but ferrite cores, eliminating USB hub, powering spindle on different circuit all seemed to have improved this significantly.  Still fighting some unknown issue though, and want to KNOW for sure that data corruption is not still haunting me.  The troubleshooting section in the wiki is largely the result of my previous woes.

AustinSaintAubin, 2018-01-05:
I have the same issue every winter. It’s an issue of poor grounding, or improper grounding (like having created a grounding loop).

andrewismoody, 2018-01-05:
I would agree with you only I've connecting every component to a common grounding block and attached that ground to the receptacle ground.  This includes grounding the machine chassis and the spindle chassis.  There should be no loops or potential difference on any of the grounds.  Unless I've completely misunderstood something.  I first realized my ground loop issues when I installed a relay for the spindle power and couldn't get it to trip due to excessive potential difference on the ground between the Arduino and the chassis.  The grounding block and common ground point for all components resolved that issue, so I assume it should have eliminated any other related issues.

andrewismoody, 2018-01-05:
Did a dry run (no stock, but full spindle/motor power) and used this simple powershell script to parse out any mismatched lines.  Only thing that came back were comments (which is expected).  Going to try with a live cut next.  I may have completely misdiagnosed whatever the issue is.

```
Param (
[String] $LogPath = ".\cnc-out-0.log"
)
Write-Host "Reading log file '$LogPath'"
$log = get-content $LogPath
Write-Host "Finding Matches"
$Result = $log|Select-string "\< \[echo\: [^$]|l >"|Select-Object -ExpandProperty Line|%{ if ($_.Contains(">")) { $_.Split(">")[1].Trim() } else { $_.Substring($_.LastIndexOf(":") + 1).Trim("]").Trim() } }|Group
Write-Host "Collating results"
$Result|?{$_.Count % 2 -ne 0 }
```

MitchBradley, 2018-01-05:
Wow, sounds like an annoying problem.

Reminds me of one of the most interesting problems I ever faced, which I call "The Case of the Van de Graaff Lab Bench".

I was responsible for the boot firmware for Sun's "SPARCstation 1" computer (and pretty much every SPARC machine thereafter).  While that machine was still in the prototype phase, we had one unit that would randomly reboot; all the other units worked flawlessly.  Everybody blamed the firmware, for no other reason than firmware always gets the blame, so it fell to me to solve the problem.

It took weeks to debug the problem.  Long story short: That unit was sitting on a lab bench, powered by a Hewlett Packard programmable power supply that cost several thousand dollars.  The  other units just had ordinary switching power supplies.  I finally discovered that the bench power supply would sometimes glitch the 5V off for about a millisecond.  Boom, reboot.

But why did it glitch?  The lab bench had a chrome-plated cylindrical footrest bar.  The bar was not bolted to the metal side supports, but instead rested in circular holes in those supports.  There was paint between the bar and the sides.  People would sit at the bench, put their feet on the bar, and without thinking anything about it, spin the bar with their rubber-soled shoes.  The combination of rubber soles, an insulated metal object, and mechanical energy, is essentially a Van de Graaff generator!  Eventually the static charge on the bar would build up until it arced across the paint to the bench sides, then discharged down the power supply's power cord, at which time the power supply would glitch its 5V output.

MitchBradley, 2018-01-05:
Care to grace us with a photo of your setup?

andrewismoody, 2018-01-05:
[https://drive.google.com/open?id=1Dt0ZE8MRZyabtatIjuH29l7l_axmCCLi2Q](url)
The cable management is a bit embarrassing, but I didn't bind any of the cables together intentionally to ensure there weren't any induction side effects.

MitchBradley, 2018-01-06:
What I would try:

Keep the signal grounds completely separate from the chassis grounds.
Run all the chassis grounds back to the electrical box.
Do not directly connect the limit switch grounds to the machine chassis.  You might even need to opto-isolate the limit switches.
Give every limit switch a separate common back to the Arduino.
Opto-isolate the spindle relay.  I use an SSR-40DA solid state relay, whose built-in control input is optocoupled.
My first try would be to not connect the stepper frame back to the driver board.  If you are using shielded cable for the stepper wiring, connect the shield at the driver end but not at the motor end.
Consider the possibility that the stepper motor power supply might be intermittently current-limiting.

Effective grounding in situations like this requires thinking about where ground current needs to flow, and giving it a path that does not interfere with some other circuit.  Just grounding everything to a single point won't always accomplish that.  Sometimes you have to make "ground islands" that are tied together by small-value resistors.  The resistors prevent large potentials between the islands, but also "steer" the ground current by making some paths "not the path of least resistance".

Apologies if I am belaboring points you already know.

Good luck.

andrewismoody, 2018-01-06:
These all seem like really good suggestions and mostly over my head.  I've read a lot of posts about using resistors on grounds, more frequently with limit switches.  When I started out on this adventure 4 years ago, I just wanted a low-cost hobby CNC machine to do some projects with.  I didn't expect or even desire to engineer the machine itself.  This all gets very tiring, the amount of additional knowledge I need to acquire at every turn to just cut some stuff.  Whining aside:

When you say:
>Give every limit switch a separate common back to the Arduino. 
Do you mean to run the open leg of the switch back to the Arduino ground pin?

And I think you mean that the Arduino and stepper driver should share a ground block with a resisted connection to the chassis ground block that then wires up to the gang box.

Additionally, I think you're suggesting that I ground the stepper shields to the driver ground (or should this be a separate grounding block with a resisted connection to the driver/Arduino ground block?

I'm currently grounding the limit switch shield to the chassis ground block - should this go along to the driver/Arduino block as well?

Any suggestions on resistor size?

I'm fairly confident that the motor power is not current-limiting, as I've ensured the amperage is above the collective motor draw amperages.

andrewismoody, 2018-01-06:
I just completed a full cut pass with no issues and the log doesn't show any mismatched echo responses.
I have 4 more passes to cut on this part, I'll work through them one-by-one tomorrow.

cheton, 2018-01-06:
Not sure if you've read this wiki: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

In addition to the pull-up resistors, connecting 100nF capacitors to GND might help for noise filtering.

[image]

[image]

andrewismoody, 2018-01-06:
Thanks, Cheton. I have read that, but I'm not actually using limit switches, just homing switches. And I don't really understand it.
I guess I'm focusing on the wrong details here.  I wouldn't expect noisy homing switches to affect tool head losing its position in the middle of a cut.
Seems like I need to decide which is less expensive: an electronics engineering degree or a high-quality pre-engineered cnc machine. 
It seems that "hobby grade cnc" means that the machine is going to be your hobby and whatever you want to cut is a pipe dream you'll never achieve.
I guess I'll see if I can cobble these resisted circuits together and see if that improves reliability any.  Thanks to everyone for the input.

MitchBradley, 2018-01-06:
I agree, DIY CNC is a labor of love, with the emphasis on "labor".  There is a reason why professionals get paid to make things "just work".  That said, commercial quality results are possible.  I use a GRBL/CNCjs-powered cheapo laser engraver to put logos on my product, and g2core/CNCjs to drive a Taig mill to do the basic shaping of my product.

It turns out that I do have an EE degree :-)  I will draw you a diagram showing physical connectivity.  Give me a few hours to take care of some other stuff then I'll post a drawing.

The wiring shown in the picture above is absolutely awful from a noise susceptibility standpoint.  The promised diagram will show how it should be done.

Limit and homing switches are the same thing - homing is just limit on one of the two ends.

MitchBradley, 2018-01-06:
Here's a swag at how things should be wired.  Let me know if this makes sense.

[CNCMachineWiring.pdf](https://github.com/cncjs/cncjs/files/1608495/CNCMachineWiring.pdf)

andrewismoody, 2018-01-06:
Thanks, Mitch. Most of this makes sense and much of it aligns with where I'm already at (with a few tweaks). The biggest gap for me is the 'RC Details' section. Is this like a break-out board or something?  Ill have to google up some schematic definitions, but it looks like the 5v power is connected to the same circuit as the ground through a resistor? This blows my mind.  Any chance that these little guys are prefabbed somewhere I can just order them?

andrewismoody, 2018-01-06:
I did two identical live cuts this morning with my machine as-is (none of Mitch's suggestions implemented yet).  One was 100% successful, the other was a complete failure (no broken bits thankfully) with no mismatched commands (using my echo logging/comparison method outlined previously).  I think this confirms that my issue is not serial communication corruption like I had first supposed, but something else.  This is good news, in a way, that I have definitively solved the serial communication issues.
Cheton mentioned that 'homing switches and limit switches are the same', but I don't have soft or hard limits enabled in my Grbl config:
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
So false triggering of home switches during a cut should not impact the tool head location, should it?
I can't help but think that the homing switches are unrelated to the cut failure, but I might be naive on this point.
I think I'm back to the drawing board, either way.  I might explore current-limiting hint that Mitch mentioned earlier and make sure I haven't made any errors with amperage requirements.  This definitely feels like missed steps now, though.  If only there was a good way to confirm that and what caused it.

MitchBradley, 2018-01-06:
Since your limit switches are not actually enabled in Grbl, we can eliminate false triggering as a cause for your cut problems.  It is still remotely possible that noise picked up by the home switch wiring could be coupling back into the Arduino, but that seems very unlikely to me.  You could test that easily, however, by simply disconnecting the home switch wires at the Arduino.  Based on the data so far, I wouldn't worry about the RC networks at this point.  We might revisit them later if necessary.

Is it possible that your feedrate is too high?  That is by far the most common cause of bad cuts that I have experienced.  Sometimes, with the feedrate too high, it will work just fine, but then all of a sudden some steps will be missed, if it doesn't stall altogether.  The difference between good and bad can be caused by all sorts of things, from motor overheating, to driver overtemperature cutout, to transient power droops, to harder places in the wood.

Looking at your photo, it seems that your stepper motors might be a little on the small side relative to the mass of the router and gantry.  I wouldn't expect them to be able to plow-through tough cuts at a fast rate. Belt-drive gantries are just not that stiff.

My experience with a belt-driven gantry machine is a laser engraver with a very lightweight head.  It can be finicky.  I have to be very careful with motion rates, and careful with balance.  Once I added a small downward-blowing fan to the head.  That unbalanced it enough that, instead of smooth lines, I got wiggles due to mechanical oscillations of the head during direction changes.  With your router, there is going to be a fair amount of sideways loading during cuts, especially on corner cuts.  One solution is to go slower.  Another is to use toolpaths with careful lead-in and lead-out moves to reduce transient loads on the tool.

andrewismoody, 2018-01-06:
Thanks again for your thoughtful responses.
His is year #4 working on this machine which started out life as a Shapeoko 2. I've done tons of ridiculous things to it to try and achieve some stability with cuts. I'm actually cutting aluminum, not wood, which is likely why I'm having so much trouble. The motors are Nema 23 160 oz in. I did fight driver and motor thermal overloads early on, but trim pot adjustments and carefully placed ventilation fans seem to have quelled that (motors and drivers maintain room temperature throughout long cuts).  I've tried various combinations of cut speeds and acceleration settings with nominal results. Going too slow on the cut gauls up the bit. I've reduced my cut depth to 0.2mm to maintain cut speed, chip load, and prevent excessive friction from flutes rubbing the already cut material.  I've tried every feed and speed calculator I could find, various flute angles, various flute counts, various coatings.
I almost ordered some 240 oz in. NEMA 23 motors to try those, but they aren't cheap and I can't seem to identify if torque is the actual issue.
I'm about to cut my losses and call it quits.  Cheton' s work on this software gave me some light at the end of the tunnel early last year as I worked through the serial comms issues that only cncjs was able to make visible.  Other software masked the issue by not properly reporting the grbl errors.
The stupid thing is that I do actually achieve some amazing cuts.  Then everything comes crashing down again for no obvious reason.  I've grown weary of staring at the machine for hour long cuts to try and diagnose it, so I stopped.  I was just accepting the failures and replacing the stock over and over again, but it's not cheap and I'm fed up.  I thought I might discover something obvious when I started this thread, but it seems to habe been a waste of everyone's time.  I appreciate all who've contributed to my silliness. If I ever get it working, I'll be sure to let everyone know what the problem was.

MitchBradley, 2018-01-06:
I'm impressed that you can cut aluminum with that kind of machine, even some of the time.  I'm used to a much stiffer machine - a Taig mill - with a lower-speed, higher-torque pulley spindle.  It's unclear that higher torque motors would help a lot, given the belt drive and the inherent stiffness characteristics of a gantry.  Best wishes.

andrewismoody, 2018-01-07:
You're using arduino / grbl / driver shield / cncjs to control your Taig?

MitchBradley, 2018-01-07:
Taig is Arduino Due  , gShield, g2core, cncjs on R Pi 0 .
Laser engraver is Grbl Arduino some driver board cncjs on R Pi 2.  Sometimes I just send laser gcode from CMD line with a python script.

andrewismoody, 2018-01-07:
And you're cutting aluminum on Taig What torque motors / driver amperage?

MitchBradley, 2018-01-07:
Mostly I cut bamboo but I have done aluminum and the Taig had no problem with it.  Steppers are 180 on X and Z, 165 on Y. They run 1.5 to 2A.  30 V 10 A power supply.  Lead screws, wish I had ball screws.