cncjs/cncjs #255: Probe is generating an ALARM every time ???

hublerdesignworks, 2018-01-14:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.13
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application (Windows 64)

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core (g2 100.26)

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows 10
- [ ] Mac
- [ ] Linux

Scratching my head for a couple of days and still I can't figure this out. I'm getting an ALARM every time when using the PROBE feature. To my understanding, the Z-AXIS is supposed to move down and trigger the Z-MIN signal wired to the grounded plate and retract. All of the HOMING (XYZ) functions work fine. Umm? What am I missing here? 

Thanks in advance.

ril3y, 2018-01-14:
I think you need to post your config for g2 also.

cheton, 2018-03-07:
If you still have problems with Z-probe, you can post your console logs or raise your issue at https://github.com/synthetos/g2/issues

closed 2018-03-07 by cheton