cncjs/cncjs #256: Load gcode without serial connection

pchanial, 2018-01-17:
### Description

If I am not mistaken, it is not possible to load a gcode file before establishing a serial connection.
It would be nice to be able to visualize the gcode toolpaths without such a connection.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2018-01-18:
One option is to use a low cost Arduino clone board and load GRBL on it.
https://m.banggood.com/ATmega328P-Nano-V3-Controller-Board-Compatible-Arduino-p-940937.html?gmcCountry=USD&currency=USD

cheton, 2018-01-18:
Duplicate of #114

arsdragonfly, 2019-08-23:
See [This GRBL simulator](https://github.com/grbl/grbl-sim)

emcniece, 2022-03-27:
Closing as duplicate!

closed 2022-03-27 by emcniece