cncjs/cncjs #257: No open ports after upgrade

Timberwolfe, 2018-01-18:
#### Description:
I was using CNCJS for a while and really liked it. I had two versions and everything was fine, then I downloaded and installed CNCJS 1.9.12 win X-64 from 1.9.11 win X-64 desktop for windows. Running windows 10, newer machine less then a 6 months old. Have plenty of open usb ports. It seemed to run just fine for the 1st time. After shutting down the computer then a couple days later restarting the computer CNCJS  would not run properly. Kept telling me that there were no open ports and I also got a warning from smart screen about the program. After doing some looking and figuring out what smart screen was I allowed CNCJS to run and I still get the same message about no open ports. I tried two other programs and everything is fine. I uninstalled CNCJS and tried to install the earlier version that did work CNCJS 1.9.11 and now all I get is a black screen where all the graphics should be. Top to bottom left to right. Can not find anything on here about this problem so if indeed this is a repeat, I did look 1st. I also have a real hard time finding the right page for the download but that is my problem. I find pages about how to install and set up and all that but not the download page. I will find it again.  I took a copy off another machine that is working fine and installed it but the same problems. Same browser (google chrome)
I am not sure what Node.js is or NPM  I just found the download page like before and got the win-64 and it took care of the rest. Sorry for being stupid about some of this. There are several .zip files and a Hex file but I just do not know crap about some of that.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-01-18:
Please roll back CNCjs to 1.9.11 temporarily until 1.9.13 is released. I upgraded the node-serialport library in 1.9.12, but it changed the default behavior as hardware glow control will now be turned off by default especially on Windows platform. I added a “Enable hardware flowcontrol” option in 1.9.13, it will allow you to turn on/off CTS/RTS manually to address this issue.

For the blank screen problem, you can close the window and open it again to workaround it. I should have identified the root cause, it will be fixed in 1.9.13 as well.

Timberwolfe, 2018-01-18:
Thank you so much. I will try this out and see what happens. Keep up the GREAT work.

Timberwolfe, 2018-02-01:
Hi how you doing. I think I did not explain myself well enough about the black screen. Not only did I have a black screen that allowed the white typing to come through, yes by reloading it is brought back the white screen but what I was getting was a complete black screen and no typing at all.Everything went away, only a Black screen.  I have been able to correct that some how I am not sure but it now works much better and the ports now work After 5 uninstalls and trying to re-install an earlier version that did not help.   You just never know what you are going to get sometimes no matter how hard to try to make sure it works for everyone. You are doing a GREAT job. I do have another problem that you might not be able to fix but I will ask anyway. I use the Triquetra three axis touch off block and usually use inches for work offset. After using the touch off block the units in CNCJS changes from mm to inches and has caused several crashes. Driving my bit down into the table or moving faster and farther then I had expected since CNCJS had been in mm before doing the touch off. Is that something that can be changed or is that just something that I need to be very careful of and maybe try using mm for the touch off plate. I am not use to mm but I know I need to learn  the difference. Just wondering. Thanks you so much.!! I used GRBL Pannel when I was having trouble with CNCJS but find it lacking in several areas that CNCJS has covered.  Where can I send some money to help support your work. ?

cheton, 2018-02-01:
If you ran into a black screen problem, you can press **Ctrl+R** or go to **View > Reload** to reload the whole UI. So far it's the only way to workaround this issue.

Can you provide your complete steps or G-code that was running with Triquetra touch plate? If possible, a video capture would help a lot.

Timberwolfe, 2018-02-01:
I had tried View>reload a bunch of times but it did not fix the problem. As I explained it was not the normal black screen with white typing showing through. It was a TOTAL black screen. could not see anything. I got this resolved so lets just let it go. If no one else has said anything then forget it.
The code for the edge finder is as follows.
G92 X0
G92 Y0
G92 Z0
G20
G38.2 X-0.5 F2
G92 X1.0625
G91 G0 X0.5 F10
G91 G0 Y1 F10
G91 G0 X-1 F10
G38.2 Y-1 F2
G92 Y1.0625
G91 G0 Y0.125 F10
G91 G0 Z1.0 F50
G91 G0 Y-0.625 F10
G38.2 Z-1 F4
G91 G0 Z0.05 F3
G38.2 Z-0.2 F1
G92 Z1.0000
G91 G0 Z0.125 F10
G91 G0 X1 F30
G91 G0 Y1 F30
M02

what this does is change the setting in CNCJS from MM to inches so when I go to jog the machine it drives my milling tool into the table breaking the bits. I was just wondering if anyone else was having this happen to them or to you. There is an easy way to fix this and that is to just run the Triquetra edge finder in mm and that will stop the problem. I am just not use to mm but I can certainly run this in mm and cure the problem.  Its not that big of a deal. DO NOT spend a lot of time on this. I can lean to change my ways. Thanks.

Timberwolfe, 2018-02-01:
Forgot to tell you I did find the page with all the updates on it so that problem is now gone too.

MitchBradley, 2018-02-01:
You could put a G21 at the end of your program - just before the M02 line - to switch back to mm.

cheton, 2018-02-02:
If you always use inches for your work, you can configure Grbl’s startup block with`$N0=G20` to force inches mode every time you power in Grbl or reset Grbl, and set `$13=1` to report position in inches. So you don’t need to check every time whether it is in inches or mm

Timberwolfe, 2018-02-02:
Thank you both for your replies. This is a huge learning curve for me and straight simple answers like these makes it so much easier for someone like me. Short, sweet, simple, and to the point. Thanks again. You are my new hero's.

cheton, 2018-02-03:
Close this issue as some issues has been fixed in 1.9.13. You can continue to discuss if you have any further questions.

closed 2018-02-03 by cheton