cncjs/cncjs #258: Feature Request

hublerdesignworks, 2018-01-19:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.13
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

I have a couple of requests that I would like to see in CNCjs.

In 3D View - options to change the color of the background, cnc paths, tool, and etc.

In 3D View - options to display 3D, GCode, or 3D/GCode side-by-side.

By the way, CNCjs is awesome and thanks for developing the software!

alfredanil, 2018-02-14:
Adding time lapse recording would the great 👍

AustinSaintAubin, 2018-02-14:
I have a way to do time lapse and recording automatically on start of a job. I will post in the wiki how to set this up soon. It uses existing CNCjs functionality and ffmpeg to accomplish all this.

AustinSaintAubin, 2018-02-18:
@alfredanil Take a look at [Setup Guide: Raspberry Pi | Recording Triggered by Events](https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Recording-Triggered-by-Events)

thymjan, 2018-02-18:
How can one flip or rotate the recorded video stream?
I tried the following options: -vf -hf -rot 180, but had no luck.
Can you give me a hint?

cheton, 2018-03-15:
Taking notes
https://www.webrtc-experiment.com/RecordRTC/
https://github.com/muaz-khan/RecordRTC
https://github.com/szwang/recordrtc-react/blob/master/src/RecordPage.react.js