cncjs/cncjs #266: move error direction

hznupeter, 2018-01-23:
#### Description
the fireware is [GRBL1.1f](https://github.com/gnea/grbl/archive/v1.1f.20170801.zip), and i set $10=2 as your wiki instruation.
Then i use cncjs to control the motor,the move direction is different with the command.
when i press Y+,it goes to Y+ and X+.
when i press Y-,it goes to Y- and X-.
when i press X- Y+,it goes to Y+.

HOW to fix it?

if i change the fireware to GRBL0.9,it's ok.

and, another question,how to change the X/Y range. In the cncjs ,the Y[max] is 650mm,but actually is 280mm.
thank u.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl1.1f
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-01-23:
When jogging the machine, what commands did you see on the console widget? I guessed you may have incompatible EEPROM settings after upgrading Grbl 0.9 to 1.1f. You can run the `$RST=*` command to clear out the EEPROM.

https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#my-grbl-settings-and-parameters-are-all-funky-after-flashing-grbl-how-do-i-clear-my-eeprom-to-start-from-a-clean-slate

You can check your Grbl settings ($$) before running the "$RST=*" command.

hznupeter, 2018-01-23:
thank you ,it seems OK now.
my machine is CoreXY type, so i should modify the code of GRBL 1.1

closed 2018-01-23 by cheton