cncjs/cncjs #268: language cant change

hznupeter, 2018-02-04:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.3

#### How Do You Install CNCjs?

- [ ] NPM
- [ x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [ x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ x] Windows
- [ ] Mac
- [ ] Linux
cant change language in the 1.9.3
i change the language option in the Settings ,but it still eng.
and i reboot the cncjs,it still cant work.

cheton, 2018-02-04:
It looks like a CI build problem that generates empty resource files, I will try to resolve this issue soon.

cheton, 2018-02-04:
So far only Windows apps are affected with this issue, others seem fine:

Executables | Status
------------- | -------
NPM installation | OK
cncjs-app-1.9.13-linux-i386.AppImage | OK
cncjs-app-1.9.13-linux-i386.deb | OK
cncjs-app-1.9.13-linux-ia32.tar.gz | OK
cncjs-app-1.9.13-linux-x64.tar.gz | OK
cncjs-app-1.9.13-linux-x86_64.AppImage | OK
cncjs-app-1.9.13-mac-x64.dmg | OK
cncjs-app-1.9.13-win-ia32.exe | Missing resource files
cncjs-app-1.9.13-win-x64.exe | Missing resource files

cheton, 2018-02-04:
Rebuilding the Windows binary on AppVeyor CI, I will let you know when it's ready to download.

cheton, 2018-02-04:
Fixed in https://github.com/i18next/i18next-scanner/commit/a5be5b8747e650d5ce29a89f6405e1c47be689d7 and https://github.com/cncjs/cncjs/commit/0134cc28afa48f2ec8fb601b569274aa64067725

You can download the app again:
* cncjs-app-1.9.13-win-ia32_20180204.exe
* cncjs-app-1.9.13-win-x64_20180204.exe

Thanks for reporting this issue.

closed 2018-02-04 by cheton

hznupeter, 2018-02-04:
thank you for your hardworking!