cncjs/cncjs #270: Only 10 defined Custom Commands are safed

thymjan, 2018-02-05:
#### Description

I want to define more than 10 Custom Commands. Well, I can. But only ten are saved and usable.

#### Versions

- CNCjs: 1.9.13
- Node.js: 4.8.7
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-02-05:
Fixed in the master branch. I will publish a new release this week. Thank you for reporting the bug.

thymjan, 2018-02-05:
Thanks again for your great work!

thymjan, 2018-02-06:
Just realized it's the same problem with makros. After defining 10 makros the elevenths isn't saved no more.

thymjan, 2018-02-06:
Feature Request: Would it be possible to give the custom command buttons **a custom background color**?

cheton, 2018-02-10:
Fixed in 1.9.14

For customizing the background color of command buttons, I will put it into the backlog, but it might not be a high priority task right now. Thank you for your understanding.

closed 2018-02-10 by cheton