cncjs/cncjs #271: serialport@beta v6.1.0

reconbot, 2018-02-06:
#### Description

As of `serialport@6.1.0` we now have custom buadrate support for linux. This will allow buadrates on linux of 250k and maybe even ones that are faster. I'd love if people here could give it a try and let me know if it work for them. I'm most curious about all the flavors of linux that run on embedded devices.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x
- Serialport 6.1.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [x] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [x] Desktop or Laptop
- [x] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2018-02-06:
Thank you for the information. The 250k baud option is supported in CNCjs. I will upgrade serialport to 6.1.0 with the next release of CNCjs 1.9.14.

reconbot, 2018-02-06:
If we are out of beta I do recommend that. I want to keep testing for a few
weeks.

cheton, 2018-03-07:
Upgraded serialport to v6.1.1 in CNCjs v1.9.15

closed 2018-03-07 by cheton