cncjs/cncjs #273: Error: Error: Permission denied, cannot open /dev/ttyAMA0

eciramella, 2018-02-06:
#### Description

After following all the steps in the FAQ, I'm still hitting this error.

I'm wondering if I've done something wrong or misunderstood how this operates.

The desire it to connect the shapeoko 3 via USB to a Raspberry PI.

#### Versions

- CNCjs: 1.9.13
- Node.js: 6.12.3
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2018-02-08:
/dev/ttyAMA0 is a built in serial connection on the Rasp Pi. 
But is [must be enabled](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/enabling-serial-console) if you want to use it... but you dont want to use this, you want to use your USB serial connection.

You should see and use your USB serial connection ( /dev/ttyUSB0 ).

[image]

eciramella, 2018-02-10:
Hmm - I never see ttyUSB0, I ended up using/seeing ttyACM (as per this discussion, seems to be ok http://community.carbide3d.com/t/shapeoko-3-xxl-raspberry-pi-3-cncjs-tablet/2458/25).

If this all makes sense, and OK, then we can probably close this ticket.

If I should be using /dev/ttyUSB0, I'm wondering if we can't still figure this out and update the documentation together?

One final thought, I made up a project with Carbide Create, just a basic 2" circle that gets cut .5" deep.  When I cut it with Carbide Motion, it's smooth and fast.  When I cut it with CNCjs, it's herky/jerky and much slower.  Is this POTENTIALLY because I'm not using /dev/ttyUSB0?

I can smooth it out a bit by adjusting these settings (at the substantial cost of speed):

[image]

But I'm not sure what those things even are at this point (Feed rate, speed, and R?).

Also - thank you so much for the feedback, I'd love a shot a taking all the lessons learned and improving your documentation for the next person.

eciramella, 2018-02-10:
I see this in syslog:

[image]

eciramella, 2018-02-11:
Welp - I think we can close this ticket.

I hooked up a Dell10v (yeah yeah, me and my commodity hardware...) and it runs just fine.  No stuttering, no stumbling.

The original raspberry pi I was using was a Model B, version 1 (https://en.wikipedia.org/wiki/Raspberry_Pi).  I even overclocked it up to 900mhz and still saw the stuttering.  The issue was the planner buffer was getting saturated (I could see it get full in the grbl widget).  As I understand it, that means that the Shapeoko Grbl board is waiting for instructions.  I understand it's just waiting a few milliseconds, but it's stuttering none the less.

The Raspberry Pi 3's have a 64 bit quad core 1.2 Ghz processor.  That's 2x the power of this old Dell 10v (roughly).

Again, the goal is to create a dust proof CNC setup so I'm off to find one of those bad boys.

Closing...

closed 2018-02-11 by eciramella