cncjs/cncjs #274: Toogle cameras

alfredanil, 2018-02-06:
#### Description
I cannot toggle between different cameras. I mean I have connected several cameras to my PC but only one of them is mounted on the tool head. Is there a feature that I can change the camera port or choose from existing cameras connected to the pc. 

#### Versions

- CNCjs: 1.9.13
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-02-07:
It's not supported in current versions, you cannot toggle detected cameras when choosing the first option _"Use a built-in camera or a connected webcam"_. If you're using IP cameras, you can fork the webcam widget and choose another media source.

I will put it into the backlog of 1.9.x releases:
- [ ] Adds a dropdown to choose one of the connected cameras.

cheton, 2018-02-10:
Added in 1.9.14. Now you can toggle multiple video inputs from the dropdown.
[image]

To view different cameras concurrently, you can fork the webcam widget and choose another video input.

closed 2018-02-10 by cheton