cncjs/cncjs #275: Some button doesn't seem to work with iOS

thymjan, 2018-02-10:
#### Description

With iOS/Safari some buttons doesn't seem to work. For example the button to colapse a widget or the button to toggle the webcam.

#### Versions

- CNCjs: 1.9.14
- Node.js: 6.12.3
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

eciramella, 2018-02-10:
Which widget were you using?

I couldn't deal with all the tiny tiny tiny text and ended up mounting a different widget (I use this one on my phone -> https://github.com/cncjs/cncjs-pendant-tinyweb).  If you're using a raspberry pi and a shapeoko, this makes it SUPER easy to zero!

thymjan, 2018-02-10:
Thanks for the tip. Just a minute ago I tested the pendant. Works fine.
I just noticed in the original GUI some buttons work an some do not (even if you pinch to zoom).

cheton, 2018-02-21:
@thymjan 

I can reproduce this issue for browsers running on iOS, however, it's hard to troubleshoot what's going wrong with a mobile device. You can either use [cncjs-pendant-tinyweb](https://github.com/cncjs/cncjs-pendant-tinyweb) or [cncjs-shopfloor-tablet](https://github.com/cncjs/cncjs-shopfloor-tablet) for mobile/tablet devices, which has better usability than the main UI.