cncjs/cncjs #277: FR: Start from specific GCode Line

VirtualEnder, 2018-02-15:
#### Description
I am using grbl to drive production grade 6040 CNC to cut carbon fiber with fairly small bits for the fine detail. Unfortunately part of this is that bits tend to snap fairly easily. The software I was using previously had an option to restart the gcode from a particular line, which is extremely useful after swapping in a new bit as the machine usually continues on after the break for a few lines. The program I was using previously would allow you to select a line of code and it would show visually where in the tool path that line started. As an alternative, it would be work to simply let the user specify the number of moves to rewind before resuming after a feed hold. CNCjs much more polished solution and the development of the other project seems to have fallen off. This is the only feature at the moment preventing me from using CNCjs for all my cuts. I would love to see it implemented. Awesome work on this software! Glad to see that active development here.

#### Versions

- CNCjs: 1.9.14 app

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-02-19:
May I know what software you were using for restarting the G-code from a particular line? Will it move to a safe Z height when resuming execution?

VirtualEnder, 2018-02-19:
Yes, I was using Candle.  It has a set of commands that return it the safe point and plunge before resuming.

MitchBradley, 2018-02-19:
In general, restarting GCode requires tracking the modal state and possibly issuing commands to reestablish that state.

TobyGWilliams, 2018-03-07:
Hi,

It would be really helpful to be able to easily start from an M0, M1, or M6 line. 

My fusion 360 post processor takes care of restarting all the modals once the tools change/M0 have been cleared and the machine can carry on cutting. At the moment, I have to re-post from Fusion every time I need to re-run one of my tool paths, which is very irritating when I have a complex part with many tools and different tool paths.

I've been fiddling around with the source code, seeing if I can insert such features myself but haven't gotten anything workable yet. 

Extra style points for: 
- Filtering the 3D render by tool change, so that I can see what the current tool/strategy in the spindle will do (compared to ALL tools at the moment).
- Some kind of cool navigating thing in the gcode, so that we can easily jump between the breaks in the code.

Many thanks
Toby

water-rhatt, 2020-12-23:
I would like to see support for this too.  I notice that UGS has a "run from line..." option. So does Mach3. I don't know how they do it internally, but they seem to execute some code where they move to the xy coordinates of the preceding instruction, read the most recent spindle and feed information, turn on the spindle, move to the previous z coordinate, and start executing from there.