cncjs/cncjs #278: No control with Marlin 1.1.01 firmware

BPAL71, 2018-02-15:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

Ramps 1.4, Arduino mega 
Marlin 1.1.01

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows 10
- [ ] Mac
- [ ] Linux

CNCJs makes connection to board Using Marlin tab, Com3, 115200 Baud Rate, but displays garbled text in console window. Will not control stepper motors or accept any commands. I can use my LCD screen to control stepper motors, i.e. move 10mm, home but not from CNCjs screen. What I am doing wrong?

BPAL71, 2018-02-16:
Solved it, had to change my baud rate to 250000. Works great now

cheton, 2018-02-18:
The default baudrate for Marlin is 250000. 

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L114
```h
#define BAUDRATE 250000
```

In addition to the baud rate option, you can also turn on/off "Enable hardware fow control" to check if it has any differences on Windows platform.

closed 2018-02-18 by cheton