cncjs/cncjs #279: Compiling source code with Node.JS...?

hublerdesignworks, 2018-02-16:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.14
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl 1.1f
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows 10
- [ ] Mac
- [ ] Linux

Please bear with me as I am new to Node.JS

I am able to install Node.JS and CNCjs on my Windows 10 Laptop. I am able to run CNCjs in Google Chrome. I can connect to the Arduino Uno with GRBL 1.1f and run some Gcode files.

I made some 'text' changes to in the source code for test. I cannot figure out the appropriate steps to compile the source code. Can someone show me the steps to do it?

Thanks in advance.

BPAL71, 2018-02-16:
Solved it, had to change my baud rate to 250000. Works great now

cheton, 2018-02-16:
For detailed build instructions, you can see following sections in [CONTRIBUTING.md](https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md):

#### Running Local Development Server
https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#running-local-development-server

#### Running Production Build
https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#running-production-build

closed 2018-02-16 by cheton

hublerdesignworks, 2018-02-16:
I followed the instructions "Running Production Build" and I am getting errors:

C:\Users\huble\Downloads\cncjs-master>npm install

> cncjs@1.9.14 prepare C:\Users\huble\Downloads\cncjs-master
> npm run clean && npm run lint && npm run test && npm run build

> cncjs@1.9.14 clean C:\Users\huble\Downloads\cncjs-master
> bash -c 'rm -rf ./dist ./output'

'bash' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 clean: `bash -c 'rm -rf ./dist ./output'`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 clean script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-02-16T18_40_33_347Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 prepare: `npm run clean && npm run lint && npm run test && npm run build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 prepare script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-02-16T18_40_33_469Z-debug.log

C:\Users\huble\Downloads\cncjs-master>

I went to the Troubleshooting section and tried the suggested work around and I am still getting 'ELIFECYCLE' errors.

hupf0211, 2019-04-12:
Does the author solve the problem? I also encountered a similar problem? Help me