cncjs/cncjs #280: Request Tiny g

nesquik011, 2018-02-16:
post processor for Tiny G coz i did try it over arduino due and it didnt work and add please A and B control to move it 

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

1.19.14

- CNCjs: 1.9.14
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2018-02-16:
I use TinyG/g2core on Arduino Due with CNCjs, and it works just fine.

nesquik011, 2018-02-17:
maybe i did something  wrong , so tell me what post processor do you use ?! 
and i want 5 axis post processor any idea ?

cheton, 2018-03-07:
CNCjs is a G-code sender, you have to find a 5-axis post processor for your CAM software to convert  G-code files.

closed 2018-03-07 by cheton