cncjs/cncjs #281: ELIFECYCLE error...

hublerdesignworks, 2018-02-16:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.13
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

I followed the instructions from "Running Production Build" and I am getting errors:

C:\Users\huble\Downloads\cncjs-master>npm install

    cncjs@1.9.14 prepare C:\Users\huble\Downloads\cncjs-master
    npm run clean && npm run lint && npm run test && npm run build

    cncjs@1.9.14 clean C:\Users\huble\Downloads\cncjs-master
    bash -c 'rm -rf ./dist ./output'

'bash' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 clean: bash -c 'rm -rf ./dist ./output'
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 clean script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! C:\Users\huble\AppData\Roaming\npm-cache_logs\2018-02-16T18_40_33_347Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 prepare: npm run clean && npm run lint && npm run test && npm run build
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 prepare script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! C:\Users\huble\AppData\Roaming\npm-cache_logs\2018-02-16T18_40_33_469Z-debug.log

C:\Users\huble\Downloads\cncjs-master>

I went to the Troubleshooting section and tried the suggested work around and I am still getting 'ELIFECYCLE' errors.

MikeColes, 2018-02-17:
Hi Hubler,

 The error is saying that the bash shell isn't installed or in your path. I'm not familiar with the Windows install. Was there a step to install bash?

hublerdesignworks, 2018-02-17:
I don't see the steps to install bash... How?

hublerdesignworks, 2018-02-17:
I think figured it out(?).

I installed Ubuntu from the Windows Store and followed the steps from www.windowscentral.com/how-install-bash-shell-command-line-windows-10.

(command prompt)...
...npm install

Looks like it's compiling...

...[build-prod-web] [i18next-scanner] transform: count=9, file="widgets\\Webcam\\Settings.jsx"
                                                         [build-prod-web] [i18next-scanner] transform: count=6, file="widgets\\Webcam\\Webcam.jsx"
                          [build-prod-web]
                                           [build-prod-web] > cncjs@1.9.14 postbuild-prod-web C:\Users\huble\Downloads\cncjs-master
           [build-prod-web] > bash -c 'mkdir -p dist/cnc/web; cp -af src/web/{favicon.ico,i18n,images,textures} dist/cnc/web/'
      [build-prod-web]

Then it just sat there...

Umm?

cheton, 2018-02-18:
You can install **Git for Windows** (https://git-scm.com/download/win), and launch a Git bash terminal to build the code:

```bash
Cheton Wu@Home MINGW64 ~/GitHub/cncjs (master)
$ npm run prepare

> cncjs@1.9.14 prepare C:\Users\Cheton Wu\GitHub\cncjs
> npm run clean && npm run lint && npm run test && npm run build

npm WARN invalid config loglevel="notice"

> cncjs@1.9.14 clean C:\Users\Cheton Wu\GitHub\cncjs
> bash -c 'rm -rf ./dist ./output'

npm WARN invalid config loglevel="notice"

> cncjs@1.9.14 lint C:\Users\Cheton Wu\GitHub\cncjs
> npm run eslint && npm run stylint

npm WARN invalid config loglevel="notice"

> cncjs@1.9.14 eslint C:\Users\Cheton Wu\GitHub\cncjs
> eslint --ext .js --ext .jsx *.js src bin test

    :    :    :

[build-prod-web] [i18next-scanner] transform: count=9, file="widgets\\Visualizer\\WorkflowControl.jsx"
[build-prod-web] [i18next-scanner] transform: count=12, file="widgets\\Webcam\\index.jsx"
[build-prod-web] [i18next-scanner] transform: count=9, file="widgets\\Webcam\\Settings.jsx"
[build-prod-web] [i18next-scanner] transform: count=6, file="widgets\\Webcam\\Webcam.jsx"
[build-prod-web]
[build-prod-web] > cncjs@1.9.14 postbuild-prod-web C:\Users\Cheton Wu\GitHub\cncjs
[build-prod-web] > bash -c 'mkdir -p dist/cnc/web; cp -af src/web/{favicon.ico,i18n,images,textures} dist/cnc/web/'
[build-prod-web]
[build-prod-web] npm run build-prod-web exited with code 0

Cheton Wu@Home MINGW64 ~/GitHub/cncjs (master)
$ ./bin/cnc -vv
2018-02-18T04:18:03.132Z - info: init Loading configuration from "C:\\Users\\Cheton Wu\\.cncrc"
2018-02-18T04:18:03.280Z - debug: app app.settings: {"x-powered-by":false,"etag":"weak","env":"production","query parser":"extended","subdomain offset":2,"trust proxy":true,"views":["C:\\Users\\Cheton Wu\\GitHub\\cncjs\\dist\\cnc\\web","C:\\Users\\Cheton Wu\\GitHub\\cncjs\\dist\\cnc\\app\\views"],"jsonp callback name":"callback","view cache":true,"verbose errors":false,"case sensitive routing":true,"strict routing":false,"view engine":"html"}
2018-02-18T04:18:03.365Z - debug: app assets: name=web, asset={"routes":["/ae5588a0/","/"],"path":"C:\\Users\\Cheton Wu\\GitHub\\cncjs\\dist\\cnc\\web","maxAge":31536000000}
2018-02-18T04:18:03.365Z - debug: app > route=web /ae5588a0/
2018-02-18T04:18:03.366Z - debug: app > route=web /
2018-02-18T04:18:03.375Z - debug: service:cncengine Loaded controllers: Grbl,Marlin,Smoothie,TinyG
2018-02-18T04:18:03.418Z - info: init Starting the server at http://192.168.1.109:8000
```

hublerdesignworks, 2018-02-19:
I crossed my fingers and tried what you suggested and 'eslint' errors...

Duane@DESKTOP-TCI8GAO MINGW64 ~/documents/github/cncjs (master)
$ npm run prepare

> cncjs@1.9.14 prepare C:\Users\huble\documents\github\cncjs
> npm run clean && npm run lint && npm run test && npm run build

> cncjs@1.9.14 clean C:\Users\huble\documents\github\cncjs
> bash -c 'rm -rf ./dist ./output'

> cncjs@1.9.14 lint C:\Users\huble\documents\github\cncjs
> npm run eslint && npm run stylint

> cncjs@1.9.14 eslint C:\Users\huble\documents\github\cncjs
> eslint --ext .js --ext .jsx *.js src bin test

'eslint' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 eslint: `eslint --ext .js --ext .jsx *.js src bin test`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 eslint script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-02-19T01_36_14_800Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 lint: `npm run eslint && npm run stylint`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 lint script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-02-19T01_36_14_833Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.14 prepare: `npm run clean && npm run lint && npm run test && npm run build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.14 prepare script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-02-19T01_36_14_872Z-debug.log

cheton, 2018-02-19:
You have to install node_modules with the `npm install` command to install all package dependencies.

hublerdesignworks, 2018-02-20:
I gave up trying to compile cncjs by following the suggested steps from the posts above. 

Here are the steps which I successfully compiled cncjs:

   - Uninstalled node.js and GitHub (Git Bash, etc.)
   - Reinstalled node.js, (https://nodejs.org), version 8.9.4 LTS
   - Landed into https//github.com/cncjs/cncjs and clicked 'clone' to ~/Documents/GitHub/cncjs
     (note: after clicking 'clone' and if GitHub was not installed, a prompt will ask if you wish to install it) 
   - Fired up Git Bash 
   - In Git Bash, jumped into ~/Documents/Github/cncjs
   - In Git Bash, typed: npm install -g cncjs
   - Compiled!
   - In Git Bash, typed: cnc
   - In Git Bash: ...info: init Loading configuration from "C:\\Users\\huble\\.cncrc"
                        ...info: init Starting the server at http://192.168.0.12:8000
   - Fired up Google Chrome and entered: http://192.168.0.12:8000
   - Presto! CNCjs!

Thank you all for your suggestions.

cheton, 2018-02-20:
`npm install -g cncjs` is not necessary if you just want to clone and compile the source code. After compiling, you can run "./bin/cnc" under the "~/Documents/GitHub/cncjs" directory.

closed 2018-03-07 by cheton

snapedumbledore, 2018-12-18:
@hublerdesignworks 
Hi，hublerdesignworks, I am newbie to cncjs ,can you tell me what commands should i type on git bash to compile source code? 
Thank you very much.