cncjs/cncjs #284: Desktop app for Raspberry Pi

yasamoka, 2018-02-19:
Hello,

Would it be possible to provide a desktop app built for ARMv6/7 that runs on the Raspberry Pi?

I am currently running CNC.js locally on the Raspberry Pi. Other machines have access via the web interface. However, Chromium, even the version optimized for the Raspberry Pi, is somewhat slow and not very responsive, even on a Raspberry Pi 3. A desktop app, I presume, would take care of that instantly.

Thank you very much for your awesome software.

MitchBradley, 2018-02-19:
The way that CNCjs desktop apps work is that the browser and the CNCjs code are bundled together into a single executable - so any deficiencies of Chromium on a given platform affect the app too.

cheton, 2018-02-19:
Hi @yasamoka,

It's not recommended to run the desktop app on a Raspberry Pi, it will consume a large amount of CPU and memory resources for rendering the normal UI, especially the 3D view.

My suggestion: If you want to access the web interface via Chromium on a Raspberry Pi, you can use a simplified CNCjs UI especially designed for tablet (https://github.com/cncjs/cncjs-shopfloor-tablet), it only needs a small amount of recoures than the normal UI since it does not use 3D graphics.

closed 2018-02-19 by cheton

yasamoka, 2018-02-19:
I see. Thank you very much. I will try the suggested UI.