cncjs/cncjs #285: Marlin support not really working

RaraAvisRO, 2018-02-19:
#### Description

While the connection with the board (a Duet3D Ethernet) is working normally and homing is correctly done, when trying to jog the commands are queued and sent only when asking for status through one of the "Get..." commands in the Marlin specific widget.

Tested on both Windows and Linux (Debian 9, AMD64).

#### Versions

- CNCjs: 1.9.14
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Duet3D/Marlin

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [x] Linux

cheton, 2018-03-07:
I just discovered that Duet3D runs RepRap Firmware, I'm not sure if there is any compatibility issue when running with CNCjs's Marlin controller code. 

For Windows desktop app, you can follow the instruction of troubleshooting guide to dump verbose output (e.g. `-vvv` option), so we might be able to know what's going wrong.
https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows

RaraAvisRO, 2018-03-08:
I have used the debug mode. The following log is showing what was dumped in the console from the moment I have connected to the Duet3D to the moment I have disconnected. I left it running for a while, not touching anything, I have issues a jog command (X+) and saw no reply until I forced issuing the M114 command through the Marlin specific widget.

If you need me to do more specific tests, please let me know!

*********************

2018-03-08T21:56:44.343Z - debug: controller:Marlin Add socket connection: id=rtMgyLmXYQHlLOLvAAAA     at MarlinController.addConnection (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:20602:13)
2018-03-08T21:56:44.348Z - debug: controller:Marlin Connected to serial port "COM22"     at SerialPort.<anonymous> (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:20538:17)
2018-03-08T21:56:44.445Z - silly: controller:Marlin < RepRapFirmware for Duet WiFi and Duet Ethernet Version 1.21RC2 dated 2018-02-15 build 2
2018-03-08T21:56:44.446Z - silly: controller:Marlin < Executing config.g...Error: M557: bad grid definition: Too many grid points; suggest increase spacing to 25.0mm
2018-03-08T21:56:44.446Z - silly: controller:Marlin < HTTP is enabled on port 80
2018-03-08T21:56:44.447Z - silly: controller:Marlin < FTP is disabled
2018-03-08T21:56:44.447Z - silly: controller:Marlin < TELNET is disabled
2018-03-08T21:56:44.448Z - silly: controller:Marlin < Warning: Macro file config-override.g not found.
2018-03-08T21:56:44.448Z - silly: controller:Marlin < Done!
[ 'M115' ]
[ 'M115' ]
2018-03-08T21:56:45.366Z - silly: controller:Marlin > M115
2018-03-08T21:56:45.369Z - silly: controller:Marlin < RFIRMWARE_NAME: RepRapFirmware for Duet WiFi and Duet Ethernet FIRMWARE_VERSION: 1.21RC2 ELECTRONICS: Duet Ethernet 1.02 or later FIRMWARE_DATE: 2018-02-15 build 2
2018-03-08T21:56:45.370Z - silly: controller:Marlin < ok
2018-03-08T21:56:45.370Z - silly: controller:Marlin controller.on('ok'): source=feeder, line="M115", res={"raw":"ok"}
[ 'M114' ]
2018-03-08T21:56:45.613Z - silly: controller:Marlin < X:0.000 Y:0.000 Z:0.000 E0:0.0 E1:0.0 E2:0.0 E3:0.0 E4:0.0 E5:0.0 E6:0.0 E7:0.0  Count 0 0 0 User 0.0 0.0 0.0
2018-03-08T21:56:45.614Z - silly: controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.000","y":"0.000","z":"0.000"},"raw":"X:0.000 Y:0.000 Z:0.000 E0:0.0 E1:0.0 E2:0.0 E3:0.0 E4:0.0 E5:0.0 E6:0.0 E7:0.0  Count 0 0 0 User 0.0 0.0 0.0"}
2018-03-08T21:56:45.614Z - silly: controller:Marlin < ok
2018-03-08T21:56:45.614Z - silly: controller:Marlin controller.on('ok'): source=server, line="M114", res={"raw":"ok"}
[ 'M114' ]
2018-03-08T21:56:46.143Z - silly: controller:Marlin < X:0.000 Y:0.000 Z:0.000 E0:0.0 E1:0.0 E2:0.0 E3:0.0 E4:0.0 E5:0.0 E6:0.0 E7:0.0  Count 0 0 0 User 0.0 0.0 0.0
2018-03-08T21:56:46.143Z - silly: controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.000","y":"0.000","z":"0.000"},"raw":"X:0.000 Y:0.000 Z:0.000 E0:0.0 E1:0.0 E2:0.0 E3:0.0 E4:0.0 E5:0.0 E6:0.0 E7:0.0  Count 0 0 0 User 0.0 0.0 0.0"}
2018-03-08T21:56:46.143Z - silly: controller:Marlin < ok
2018-03-08T21:56:46.143Z - silly: controller:Marlin controller.on('ok'): source=server, line="M114", res={"raw":"ok"}
[ 'M105' ]
2018-03-08T21:56:46.648Z - silly: controller:Marlin < ok T
2018-03-08T21:56:52.318Z - silly: controller:Marlin Reschedule current position query: now=1520546212318ms, timespan=5158ms
2018-03-08T21:56:56.956Z - silly: controller:Marlin Reschedule temperture report query: now=1520546216956ms, timespan=10346ms
2018-03-08T21:56:57.510Z - silly: controller:Marlin Reschedule current position query: now=1520546217510ms, timespan=5192ms
2018-03-08T21:57:02.715Z - silly: controller:Marlin Reschedule current position query: now=1520546222715ms, timespan=5205ms
2018-03-08T21:57:07.336Z - silly: controller:Marlin Reschedule temperture report query: now=1520546227336ms, timespan=10380ms
2018-03-08T21:57:07.873Z - silly: controller:Marlin Reschedule current position query: now=1520546227873ms, timespan=5158ms
2018-03-08T21:57:13.073Z - silly: controller:Marlin Reschedule current position query: now=1520546233073ms, timespan=5200ms
2018-03-08T21:57:17.684Z - silly: controller:Marlin Reschedule temperture report query: now=1520546237684ms, timespan=10348ms
2018-03-08T21:57:18.237Z - silly: controller:Marlin Reschedule current position query: now=1520546238237ms, timespan=5164ms
2018-03-08T21:57:23.389Z - silly: controller:Marlin Reschedule current position query: now=1520546243389ms, timespan=5152ms
2018-03-08T21:57:23.610Z - debug: service:cncengine socket.command("COM22", "gcode"): id=rtMgyLmXYQHlLOLvAAAA     at Socket.<anonymous> (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:16897:21)
2018-03-08T21:57:23.613Z - debug: service:cncengine socket.command("COM22", "gcode"): id=rtMgyLmXYQHlLOLvAAAA     at Socket.<anonymous> (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:16897:21)
2018-03-08T21:57:23.613Z - debug: service:cncengine socket.command("COM22", "gcode"): id=rtMgyLmXYQHlLOLvAAAA     at Socket.<anonymous> (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:16897:21)
2018-03-08T21:57:28.023Z - silly: controller:Marlin Reschedule temperture report query: now=1520546248023ms, timespan=10339ms
[... cut ...]
2018-03-08T21:57:54.703Z - silly: controller:Marlin Reschedule current position query: now=1520546274703ms, timespan=5167ms
2018-03-08T21:57:59.533Z - debug: service:cncengine Disconnected from 127.0.0.1: id=rtMgyLmXYQHlLOLvAAAA, user.id=, user.name=     at Socket.<anonymous> (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:16743:21)
2018-03-08T21:57:59.533Z - debug: controller:Marlin Remove socket connection: id=rtMgyLmXYQHlLOLvAAAA     at MarlinController.removeConnection (c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js:20655:13)

RaraAvisRO, 2018-03-12:
More to this one... RepRap firmware replies to the M105 command with just "ok T" when set to CNC mode and having no extruder defined. I just tested this with a console. I will try to get more info and see if this a bug or an intended behavior.

cheton, 2018-03-13:
The parser code was implemented according to spec of Marlin's [Temperature::print_heaterstates()](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/temperature.cpp#L2189-L2275) function, it can support either one of the following forms:

```
ok T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0
ok T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0 W:?
ok T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0 W:0
 T:293.0 /0.0 B:25.9 /0.0 @:0 B@:0
 T:293.0 /0.0 B:25.9 /0.0 T0:293.0 /0.0 T1:100.0 /0.0 @:0 B@:0 @0:0 @1:0
 T:293.0 /0.0 (0.0) B:25.9 /0.0 T0:293.0 /0.0 (0.0) T1:100.0 /0.0 (0.0) @:0 B@:0 @0:0 @1:0
```

I may need to check how RepRapFirmware replies the M105 command to cover all test cases.

cheton, 2018-03-13:
https://github.com/dc42/RepRapFirmware

RaraAvisRO, 2018-03-13:
Loot for "GCodes::GenerateTemperatureReport" in src/GCodes,cpp. As far as I can tell, if there is no extruder, all you get is "ok T".
And I'm trying to use the Duet3D with a WorkBee CNC router, so I have configured the firmware to go for CNC mode.

cheton, 2018-03-14:
https://github.com/dc42/RepRapFirmware/blob/dev/src/GCodes/GCodes.cpp#L4529-L4560
```cpp
// Store a standard-format temperature report in 'reply'. This doesn't put a newline character at the end.
void GCodes::GenerateTemperatureReport(const StringRef& reply) const
{
	Heat& heat = reprap.GetHeat();

	// Pronterface, Repetier etc. expect the temperatures to be reported for T0, T1 etc.
	// So scan the tool list and report the temperature of the heaters associated with each tool.
	// If the user configures tools T0, T1 etc. with 1 heater each, that will return what these programs expect.
	reply.copy("T");
	char sep = ':';
	for (const Tool *tool = reprap.GetFirstTool(); tool != nullptr; tool = tool->Next())
	{
		for (size_t i = 0; i < tool->HeaterCount(); ++i)
		{
			const int heater = tool->Heater(i);
			reply.catf("%c%.1f /%.1f", sep, (double)heat.GetTemperature(heater), (double)heat.GetTargetTemperature(heater));
			sep = ' ';
		}
	}

	const int bedHeater = (NumBedHeaters > 0) ? heat.GetBedHeater(0) : -1;				// default to first heated bed
	if (bedHeater >= 0)
	{
		reply.catf(" B:%.1f /%.1f", (double)heat.GetTemperature(bedHeater), (double)heat.GetTargetTemperature(bedHeater));
	}

	const int chamberHeater = (NumChamberHeaters > 0) ? heat.GetChamberHeater(0) : -1;	// default to first chamber heater
	if (chamberHeater >= 0)
	{
		reply.catf(" C:%.1f /%.1f", (double)heat.GetTemperature(chamberHeater), (double)heat.GetTargetTemperature(chamberHeater));
	}
}
```

cheton, 2018-03-15:
Before I make the change, could you do the following steps:
1. Open the file `c:\Users\Catalin Ionescu\AppData\Local\Programs\cncjs-app\resources\app\app\index.js` in an editor. Notepad is not recommended as it will not treat `\n` as line breaks on Windows.
2. Comment out the line `_this.queryTemperature();` around line number 20337.
    [image]
3. Save the change, close running program, and restart it again

I need your help to check if it will work after disabling temperature query.

RaraAvisRO, 2018-03-15:
I have done the change and it works. Meanwhile, as there are several other RepRap firmware changes in the making, let me discuss with its main developer. I thunk that the rogue "T" in the reply is just a leftover from the days before adding support for CNC.

Right now the firmware has 3 working modes - FFF, Laser and CNC. M450 returns current mode. For more info you can check here - https://duet3d.dozuki.com/Wiki/Gcode#Section_M450_Report_Printer_Mode.

closed 2019-05-09 by cheton