cncjs/cncjs #289: CNC no connect

nickm324, 2018-03-07:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.15

#### How Do You Install CNCjs?
 NPM
#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Raspberry Pi 3B

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

After executing cnc -m /tablet:/home/pi/cncjs-shopfloor-tablet/src presented with the connect screem clicking open takes you to the main screen and all controls are either disabled or non-functional and says no connect. If I open a regular workspace everthing sees to work.

In the console I am getting this error.
2018-03-07T17:12:48.493Z - **error: uncaughtException: "path"** is not defined: null                                             date=Wed Mar 07 2018 11:12:48 GMT-0600 (CST), pid=24400, uid=1000, gid=1000, cw                                            d=/usr/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/node, version=v8.9                                            .4, argv=[/usr/bin/node, /usr/bin/cncjs, -m, /tablet:/home/pi/cncjs-shopfloor-ta                                            blet/src], rss=163409920, heapTotal=135667712, heapUsed=115035048, external=7083                                            48, loadavg=[2.64013671875, 2.23779296875, 2.02587890625], uptime=9893, trace=[c                                            olumn=19, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index.js, function=new S                                            erialConnection, line=2018, method=null, native=false, column=27, file=/usr/lib/                                            node_modules/cncjs/dist/cnc/app/index.js, function=new GrblController, line=1718                                            1, method=null, native=false, column=34, file=/usr/lib/node_modules/cncjs/dist/c                                            nc/app/index.js, function=null, line=16817, method=null, native=false, column=13                                            , file=events.js, function=emitTwo, line=126, method=null, native=false, column=                                            7, file=events.js, function=Socket.emit, line=214, method=emit, native=false, co                                            lumn=12, file=/usr/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js,                                             function=null, line=503, method=null, native=false, column=7, file=internal/proc                                            ess/next_tick.js, function=_combinedTickCallback, line=131, method=null, native=                                            false, column=9, file=internal/process/next_tick.js, function=process._tickCallb                                            ack, line=180, method=_tickCallback, native=false], stack=[TypeError: "path" is                                             not defined: null,     at new SerialConnection (/usr/lib/node_modules/cncjs/dist                                            /cnc/app/index.js:2018:19),     at new GrblController (/usr/lib/node_modules/cnc                                            js/dist/cnc/app/index.js:17181:27),     at Socket.<anonymous> (/usr/lib/node_mod                                            ules/cncjs/dist/cnc/app/index.js:16817:34),     at emitTwo (events.js:126:13),                                                 at Socket.emit (events.js:214:7),     at /usr/lib/node_modules/cncjs/node_mod                                            ules/socket.io/lib/socket.js:503:12,     at _combinedTickCallback (internal/proc                                            ess/next_tick.js:131:7),     at process._tickCallback (internal/process/next_tic                                            k.js:180:9)]

nickm324, 2018-03-07:
Also I run cncjs normally on the Pi using pm2, how would I get this run under pm2 or in the background?

MitchBradley, 2018-03-07:
As time permits, I will upgrade to 1.9.15 to try and reproduce the problem.  Meanwhile -

I too have seen that uncaught exception problem, going back through many versions.  It has never caused any actual failures for me.  I have tried several times to diagnose it, but I have always gotten lost in twisty little mazes deep in library routines.  It is conceivable that it might be related to your problem, but my best guess is that it is a not.

To start with pm2, you will need to add the phrase "-m /tablet:/home/pi/cncjs-shopfloor-tablet/src" to the "pm2 start" that you used to setup the cncjs autostart.  That might require that you first remove the existing autostart.  I'm not a pm2 expert, so I can't give you exact details for undoing what you currently have going.

Personally, I find pm2 to be a very heavyweight way to do something that can be done much more simply with a crontab entry, thus speeding up startup and avoiding the need to learn pm2 and keep it up to date.  If you have a whole suite of Node applications, then pm2 is probably a good way to manage them, but for autostarting one app, it seems like massive overkill.

https://github.com/cncjs/cncjs/issues/65 tells how to use crontab.  Just add the "-m ..." phrase to the cncjs start line.

-

MitchBradley, 2018-03-07:
I see that you closed the issue.  Did you find a fix?

nickm324, 2018-03-07:
Well I closed it cause I thought I put it in the wrong place, it was an issue using the tablet pendent so I closed it and recreated one at its github.

nickm324, 2018-03-07:
I tried using the -m option with pm2 and while cncjs would work and was accessible via /#/workspace when trying to access /tablet would return a weird error page.
I tried pm2 start cncjs -m /tablet:/home/pi/cncjs-shopfloor-tablet/src and 
pm2 start cnc -m /tablet:/home/pi/cncjs-shopfloor-tablet/src and neither works as far as the tablet part goes but cncjs does run and is accessible.

[image]

MitchBradley, 2018-03-08:
@cheton -

Regarding the uncaughtException error above, can you explain to me how the following code is supposed to work?

SerialConnection.js line 74: the constructor throws an error if !settings.path

GrblController.js does "new SerialConnection({ path: port, ...})

According to the exception dump, apparently some tick callback (maybe startup?) is doing "new GrblController", but it seems that "port" is not set to anything that can be construed as true at that point.

How is the initial value of 'port' supposed to be set?

nickm324, 2018-03-08:
thanks for the info, I ended up using  pm2 start $(which cncjs) -- --port 8000 -m /tablet:/home/pi/cncjs-shopfloor-tablet/src and it appears it is working.
 
Which I kind of pieced together from a couple different articles I read.

 And I see now that the port issues is because for what ever reason its not seeing the other USB port that is avaiable and instead is trying to connect to the AMA0 port which is not the USB port to the CNC, its possible something else has already attached to the CNC port and thats why its not able to try and connect correctly. I am going to wipe the Pi and do a fresh install and try getting it working with out the issues.

nickm324, 2018-03-08:
pi@X-Carve:~ $  ps ax | grep cnc
  947 ?        Rsl    2:11 node /usr/bin/cncjs
 1386 pts/0    S+     0:00 grep --color=auto cnc

Just noticed another issue that might be cause part of these issues, ever since I updated to 1.9.15 and run cncjs the cpu stays at 100% usages and the web interface is slow to respond.

 PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
  947 pi        20   0  849780 758944  12328 R 100.0 79.9   4:02.42 node /usr/bin/c
 1623 pi        20   0    8212   3272   2732 R   3.9  0.3   0:00.12 top

MitchBradley, 2018-03-08:
I have no idea about that 100% problem.  The churn rate of cncjs is too fast for me to keep up.

nickm324, 2018-03-08:
You know how I can roll back to the previous version?
Just to see if it makes a difference.

MitchBradley, 2018-03-08:
There is probably some nvm command that will change the version, but I don't know the details.

MitchBradley, 2018-03-08:
npm, not nvm, I think,

nickm324, 2018-03-08:
ok I will figure it out, thanks.

nickm324, 2018-03-08:
Still have a high cpu usage issue but everything is working.

closed 2018-03-08 by nickm324

cheton, 2018-03-08:
@nickm324 Did you find anything weird things from the output logs? Will cpu usage stay at 100% if you roll back to v1.9.14?

nickm324, 2018-03-08:
Rolling back did not help I am still experiencing 100% or higher CPU usage. Still trying to track down the problem.

nickm324, 2018-03-08:
Did find this in the log.
FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory

cheton, 2018-03-08:
What version of Node.js you're currently using? You can check the version with `node -v`.

BTW, I found some people workaround this issue with the `--max_old_space_size` option:
* https://github.com/angular/angular-cli/issues/5618#issuecomment-357839039
* https://github.com/ionic-team/ionic-cli/issues/1453

You can open the executable file in the editor (e.g. `vim $(which cnc)`), and append the `--max-old-space-size` flag, as shown below:

```bash
#!/usr/bin/env node --max-old-space-size=8192
```

or add an alias

```bash
alias cncjs="node --max-old-space-size=8192 $(which cnc)"
```

nickm324, 2018-03-08:
I was running version 8 of node, I  tried down grading to 6 and 5 and nothing changed still had the CPU usage issue. I am in the process of wiping the Pi and doing a fresh install from the ground up.