cncjs/cncjs #290: Expand Z-Probe widget for x and y axis

ENGR-mike, 2018-03-07:
Cheton,
I would love to see the Z-Probe widget expanded to include X-Probe and Y-Probe.  I copied the G code from the Z-Probe widget and created macros that work just fine.  Unfortunately, whenever I want to change a parameter I have to manually edit the macros.  I like how I can just enter values in the fields for Z-Probe and wish I could do the same for the x and y axis.

cheton, 2018-03-08:
May I know if there is anything different (excluding the Axis word) in your macros between X/Y probe and Z probe?

ENGR-mike, 2018-03-09:
No, nothing is different.

cheton, 2018-03-09:
ok. I will find a way to add X/Y probe support in addition to Z probe.

GeerGuy, 2018-03-09:
ENGR-mike - can you share your macros for X and Y probe?

ENGR-mike, 2018-03-11:
The following is for the X axis using a 1/4" bit:
`; X-Probe
G91
G38.2 X0.5 F1
G90
; Set the active WCS X0
G10 L20 P1 X-0.125
; Retract from the material
G91
G0 X-0.1
G90`

The following is for the Y axis using a 1/4" bit:
`; Y-Probe
G91
G38.2 Y0.5 F1
G90
; Set the active WCS Y0
G10 L20 P1 Y-0.125
; Retract from the material
G91
G0 Y-0.1
G90`

cheton, 2018-03-27:
@ENGR-mike  I'm really busy with my day work over these two months. I will get back to it once I'm free. Sorry for the late.

jakeface1, 2019-05-31:
I know it’s been a while, but any updates on this feature? Cncjs is by far the best app I’ve used. This would complete it for me.

cheton, 2020-03-07:
This feature is available in [1.9.21](https://github.com/cncjs/cncjs/releases/tag/v1.9.21)

closed 2020-03-07 by cheton

ENGR-mike, 2020-03-07:
Thank you so much for implementing this feature.  I greatly appreciate it.  I have been using my macros successfully, but it is nice to have this as a built-in feature. Sent from Mail for Windows 10 From: Cheton WuSent: Friday, March 6, 2020 8:52 PMTo: cncjs/cncjsCc: ENGR-mike; MentionSubject: Re: [cncjs/cncjs] Expand Z-Probe widget for x and y axis (#290) Closed #290.

ENGR-mike, 2020-03-08:
It doesn’t work correctly.  I will explain later. Sent from Mail for Windows 10 From: Cheton WuSent: Friday, March 6, 2020 8:52 PMTo: cncjs/cncjsCc: ENGR-mike; MentionSubject: Re: [cncjs/cncjs] Expand Z-Probe widget for x and y axis (#290) Closed #290.