cncjs/cncjs #291: A and B axis control key in 1.9.15

nesquik011, 2018-03-08:
#### Description

how to use the short cuts ? i did try  ctrl + alt  + b   and  ctrl + alt + [ or b        nothing happen i searched for the commant key on windows some sais its ctrl and some said its the start key , i even try that , nothing happened i mean no controlling key appeared , so how to use it ? [https://ibb.co/ecvnDS](url)

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

nesquik011, 2018-03-08:
NOTE i dont have any connection to any board  ,  i am just seeing the functions on windows 

i just want to see  A and B  controlling keys  for rotation left and right etc so someone help me please !

nesquik011, 2018-03-08:
i think the  jog control is not a displayed one ?  just a short cut act in real    i will upload picture , if this option is not there please i request it https://ibb.co/hOzEA7

cheton, 2018-03-08:
For using keyboard shortcuts, see https://github.com/cncjs/cncjs/wiki/User-Guide#keyboard-shortcuts

You have to select an axis at first

* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>x</kbd> - Select/Deselect X Axis
* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>y</kbd> - Select/Deselect Y Axis
* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>z</kbd> - Select/Deselect Z Axis
* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>a</kbd> - Select/Deselect A Axis
* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>b</kbd> - Select/Deselect B Axis (_Supported in v1.9.15_)
* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>c</kbd> - Select/Deselect C Axis (_Supported in v1.9.15_)

and jog forward / backward with

* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>[</kbd> or <kbd>b</kbd> - Jog Backward
* <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>]</kbd> or <kbd>f</kbd> - Jog Forward

A/B/C axis control buttons are already there for moving forward (+) or moving backward (-).

[image]

nesquik011, 2018-03-08:
if want to set them back to home ? i press on the house key right ?

nesquik011, 2018-03-08:
and where to write down the ctrl + alt + command + [ or b - Jog Backward    ?  in console ( black window )  
and   where command is in mac right ?  so in windows i only  press ctrl + alt  + [ or b - Jog Backward  am i right ?

cheton, 2018-03-08:
Just press combo keys on the UI, not in the console widget.

cheton, 2018-03-08:
I suggest you connect to a board for testing, you can flash Grbl to an Arduino Nano or Uno. All buttons will be enabled only when a serial connection is successfully established.

closed 2018-03-08 by cheton