cncjs/cncjs #293: Request Auto tool change

nesquik011, 2018-03-08:
A Request Auto tool change something more easy than macro  ,  on and off option relay like , for the spindle to release the tool 👍

nesquik011, 2018-03-09:
Watch the video please https://youtu.be/fvYWyE2GBsg

See how smooth it is make some option like this please 😍

nesquik011, 2018-03-09:
is it possible ?  @cheton

cheton, 2018-03-09:
Also see https://discuss.inventables.com/t/auto-tool-changer-works-great/22518

Note that you still have to prepare [tool change macros](https://github.com/cncjs/cncjs/wiki/Tool-Change) by yourself because there is no standarized way to know how users setup their auto tool changer (ATC). 

The way that a G-code sender can do is to replace the M6 tool change command (e.g. `M6 T1` or `T1 M6`) with a tool change macro created by the user. I think it is possible to do this in a future version, user can assign a tool number for a tool change macro, then the sender will replace `M6 Tn` with a specific macro to do the tool change.

nesquik011, 2018-03-10:
@cheton   NOTE the spindle will auto zero when it arrive at tools area to know its length , and after picking the tool it will auto zero again to know the length of the tool then go back to work

nesquik011, 2018-03-10:
**## make me happy and tell me you will do it :(   @cheton**

ahedderich, 2018-03-13:
I've build a prototype of an automatic tool changer for an er11 spindel. the first test where pretty good.
I'm using an esp11 as the controller of the tool changer. With HTTP requests the tool can be selected and it screws the old one from the spindel and the new one in. Since the toolholder cups are specificly 3d printed for each tool the z-offset per tool is already known to tool changer and the guide makes sure the tool is screwed in perfectly strait. (it does it already better then me by hand ;))
I'm currently waiting for a few magnets from china to finish the build and over the next weeks i will start building a new widget for cncjs which lets you configure the tool list and trigger the tool change.
When that works i will try to integrate it with the M6 command and upload the entire thing.
@cheton in my opinion your time is better spend on the auto leveling feature ;)

nesquik011, 2018-03-16:
@ahedderich   thats great ! are you uploading it to cncjs ?  should i follow you or something to know when you make it ? by the way i want to see your auto tool project , did you make it or buy it ? i mean the ACT spindle

ahedderich, 2018-03-18:
yes. i will open a pull request as soon as i have build something. It might take a few weeks though. I can only work on it in my very limited free time.

For the tool changer no special spindle will be needed. An atc spindle like bt30/40 is way to expensive for my hobby cnc, so i thought about a way to automate my manual process of changing tools on my china spindel like this one: https://www.banggood.com/ER11-Chuck-CNC-500W-Spindle-Motor-with-52mm-Clamps-and-Power-Supply-Speed-Governor-p-1027937.html
Of course it wont compare in speed and efficiency with a real atc spindle but good enough for my usage.

I will also release the stl files and software bits for that. There will be a link in this threat then. 
The widget for cncjs should be usable with other tool changing methods as well with some modifications on how to interact with different tool changers.

nesquik011, 2018-03-18:
wow ! that would be amazing , i have been working on something like that , change normal cnc spindle air cooling or water to  ATC  , just a part you plug in the spindle then turns to ATC   like cnc cat if you them , and if you need anything ! please tell me any graphics 3D 2D whatever i can make them easy as 1 2 3

nesquik011, 2018-04-01:
Any news ? @ahedderich

ahedderich, 2018-04-03:
I'm testing currently if i can make some triamic drivers work with the changer so that i can use the drivers current sensing to check when the screw is tight enough.
Next weekend i will have a look implementing the cncjs widget.

ahedderich, 2018-04-07:
@cheton i started building the general interface for the toolchanger widget. i tried to stay close to the projects source structure.
If you got time have a look at the code so far and give me a feedback if it's in your style.
https://github.com/ahedderich/cncjs/commit/49566ee6f87f6e417505c2b7502e5b766ccb7b0f

While the tool changer is deactivated, the current functionallity will be unchanged:
[image]

Active:
[image]
While changing to a new tool:
[image]

Tool library:
[image]

If the widget is enabled, the M6 notification is changed to a new M6 auto change message.
I hooked in the controller status event which seems to work. The next steps would be to select two macros in the settings. One to change a tool and one to request the current tool form the cnc maschine.
The toolnumber and the new z offset (if selected) could be used as variables there.
Not sure thou how it normaly works with automatic tool changers. Espessially what kind of response is to be expected.
For my personal case i will define an additionall option for my device specific external tool changer. Basically a wrapper class that will take the state, controller and action instances and provide the two methods of status-checking and tool-changing. I need to mix xhr requests and gcode commands to make it work. maybe other tool changer specific "driver"-classes could be added later.

PS: I added the new widget to the widget manager although currently it's only possible to activate by editing the config file manually. i guess there is still a spot missing where i have to register the option. ;)

nesquik011, 2018-04-07:
@ahedderich   there should be coordinates of the took poisons   X Y Z  location , and the spindle should test the length of tool before and after doing the job , for example we have 5 tools  now , the spindle will go check its length first , then pick a tool number 1  then check the length once more , then do the milling and before change to tool number 2 it will re check the length then go to tool number 1 place then release the tool and go back check length then pick the new tool number 2  then go back check length  then go to milling     ETC   , and you should add the speed of the Z going down and up , and you should add how long does the spindle air compressor push the air  to spindle to release because it make stop the air before it compliantly leaves the tool  so it will pick again the same tool   just check my design   

https://ibb.co/kSJq3S

ahedderich, 2018-04-07:
you can define those thing in a custom macro for the tool change. since the operation is too individual, i don't think i would make sense to predefine everything by cncjs. i suspect you already use a macro for your manual toolchange now. with the new variables for z-offset and toolnumber you only need to modify slightly what you already have.

nesquik011, 2018-04-07:
well how your auto tool change works ? may you show me picture of it , and how is your general interface for the toolchanger widge  works ?  maybe i dont understand well

ahedderich, 2018-04-07:
the tool change you were talking about in the early post works by driving over a specific position drop the current tool, go to a different position and picking up a new tool.
so in your case basically your first part of a the macro would be like:
- ...some pre stuff...(probing maybe)
- go to drop position based on the tool you want to drop
- release tool
- go to a pick up position based on the tool you want
- secure tool
- ...some post stuff...(probing, testing etc.)

In my case it's a bit different
- i will move the tool head to a specific position.
- orient the tool head with an optical sensor in the a consistent orientation.
- move the toolhead into the changer
- have the toolchager detach the tool and attach a new one
- move the spindle back to the stored position

most of the sensors for detecting a failure will be handled by the tool changer itself. Independent from the cnc maschine.
the tools themself are located in a magazine underneath the maschine and will be loaded and unloaded through a hole.
[image]
the image shows an earlier version. i haven't changed it yet since i'm currently experimenting more with the electronics needed.

that's what i mean with "the process of changing a tool is very individual". i don't know your maschine and what gcode would be needed to release and secure the tool. so everyone should be able to define their maschine specific macro to perform the actual tool swap.