cncjs/cncjs #294: Request auto save Gcode when the power (light) goes off and on

nesquik011, 2018-03-08:
Request auto save for the Gcode so when the power(Light) go off , and go back on , so it start automatically resume the work ( or ask you for resume the work as an option you can set )

cheton, 2018-03-08:
Duplicate of #277.

nesquik011, 2018-03-09:
so i didnt get it ?  did you make an update or something resuming the gcode when the power comes back ?

MitchBradley, 2018-03-09:
I think this is a bad idea.  The number of things that potentially have to be reestablished after a power outage is almost impossible to enumerate. It would be an endless source of bug reports. The chances of it working reliably in any given setup is almost 0.

nesquik011, 2018-03-09:
I saw mach3 board has built in memory with a small battery so it resume the gcode after power is off

MitchBradley, 2018-03-09:
If you know lots of details about the system, things like this are do-able.  But cncjs works with so many different setups - different firmware like GRBL, Smoothie, TinyG, G2Core, Marlin (and different versions of those) - different machines like 3D printers, laser engravers, milling machines, lathes - different setups like host/client vs. all on one computer - different browsers - different pendant UIs - different network connectivity between the host and client.  The matrix is enormous.  Every different combination of those factors would have different issues with automatic restart.

cheton, 2018-03-09:
CNCjs is a general purpose G-code sender software, the one thing it can do in a future version is to remember the last line number of G-code sent to the controller, so you can restart a G-code program from a specific line after restoring the modal state.

Something like auto save is beyond the scope of current development roadmap.

MitchBradley, 2018-03-09:
Note that, in order to restart safely, you need to know the last G-code line that the controller actually finished executing, not the last one that was sent.  If you try to return to the position after the last one that was sent, you risk crashing the tool into uncut material.

Even if you know where to start, there can be problems depending on the type of machine.  For a laser engraver, restarting in the middle will often result in an over-burned spot or line.  For a mill, you can get a tool dimple if you return to a position by a different path than the original cut path.  For 3d printer, restarting can cause bad adhesion or blobs.  This is a very difficult problem to solve well.

I used Mach3, which supports resume-from-given-line, for many years.  I tried to use that feature several time and don't recall ever succeeding.  I made a lot of messes, but never got a usable part.

nesquik011, 2018-03-09:
ok make it like  * remember the last line number of G-code sent to the controller , then the machine XYZ ABC all go to home then ask should i continue ? yes then it goes back to the last G-code thats it was sent to the controller *    okay ?

dafyddhirst, 2018-04-25:
@nesquik011 Auto resume is possibly the most dangerous feature of a CNC you could add. even go back to home after power outage may mean your tool path smashes through your work piece while returning home. On a 3d printer this is easier to achieve and with a lot less risks associated with it. If they coded it and your router head smashed your machine or work piece up im sure the first port of call would be to complain in here. So i say id rather have a machine that just stops safely. and cant destroy its self or your workshop a much better idea . Id rather spend the starting again and spend the money on a big ups so i could safely park the machine should there be a power problem.

closed 2020-03-04 by cheton