cncjs/cncjs #295: Versions 1.9.14 and 1.9.15 High CPU Usage

nickm324, 2018-03-09:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.14
- CNCjs: 1.9.14

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Raspberry Pi

#### Operating System

- [x] Not Applicable

When using version 1.9.14 or 1.9.15 CPU usages is always above 100%. Roll back to version 1.9.13 and CPU usages is normal around 1% unless running a project.

Name  │ mode │ status │ ↺ │ cpu  │ memory    │
├───────┼──────┼────────┼───┼──────┼───────────┤
│ cncjs │ fork │ online │ 0 │ 149% │ 72.2 MB   │

nickm324, 2018-03-09:
this is with .13 running
 Name  │ mode │ status │ ↺ │ cpu │ memory    │
├───────┼──────┼────────┼───┼─────┼───────────┤
│ cncjs │ fork │ online │ 3 │ 0%  │ 78.0 MB   │

cheton, 2018-03-09:
I would like to know if there is version difference of the serialport module between your v1.9.13 and v1.9.14 installation. You can check the version for installed serialport module as below:

```bash
cd $(dirname `which cnc`)/../lib/node_modules/cncjs/node_modules/serialport
grep -r '"version"' package.json
```

It will output something like this:
```
"version": "6.0.5"
```

nickm324, 2018-03-09:
Says  "version": "6.0.5"

nickm324, 2018-05-04:
This is still an issue, its gets to the point where the server quits responding and I cannot even connect via ssh shell. Going to downgrade again to 1.9.13.

nickm324, 2018-05-04:
Definitely has something to do with mjpeg because if I unplug my camera the server seems to be fine. As I stated before mjpeg maxes out the CPU after a little while.

cheton, 2018-05-05:
So was it probably an issue where MJPEG consumes most of the CPU, no matter what CNCjs version (1.9.13 ~ 1.9.15) you're running? What's the generation of your Raspberry Pi?

nickm324, 2018-05-05:
No if I downgrade to 1.9.13 I dont have the issue at all. Only when I run 14 or 15 cpu process mjpeg is +100% cpu usage.

nickm324, 2018-05-20:
Having all kinds of issue with .15 version. Web interface stops responding. If you close the browser and reopen then the job stops cutting and the browser is hanging trying to load the web interface. If you then close the browser while its struggling to display the cncjs web interface the job will finally continue if you wait for it long enough to start responding. From that point forward any access attempts to view the web interface stalls the job from cutting.

lspiker, 2018-06-21:
Has anyone been able to workout the high cpu / non-responsiveness with the .15 version. I did a complete wipe and install on my Raspberry PI 3 and am still having the same issues.

AustinSaintAubin, 2018-06-21:
Use `htop` to look at the process that is using the CPU.

-- Austin Saint Aubin

lspiker, 2018-06-21:
node /usr/bin/cncjs is the process using the cpu and pm2 status provides the following:

 App name │ id │ mode │ pid   │ status │ restart │ uptime │ cpu │ mem        │ user │ watching │
├──────────┼────┼──────┼───────┼────────┼─────────┼────────┼─────
│ cncjs    │ 0  │ fork │ 20052 │ online │ 0       │ 45s    │ 99% │ 101.9 MB   │ pi   │ disabled

AustinSaintAubin, 2018-06-21:
That is odd. Looks like the process has only been alive for 45 sec. You check to see if its being restarted by PM2... or if processor time lowers after a few min?

lspiker, 2018-06-21:
I do not believe it is being restarted by PM2. The memory used will continue to climb and the cpu will remain at approximately 100%. However, if the serial port is not connected it does not have high cpu usage.

nickm324, 2018-06-21:
This is exactly what I have been experiencing. I have gone back to .13 because of it and have not seen the issue with .13. In my case my CPU in top shows %100+ sometimes showing %149. Eventually it stops responding altogether and you have to reboot it. .13 does not seems to have that issue. I also had the same issue with .14.

lspiker, 2018-06-22:
I am getting the same result with .13. What version of serialport do you
have installed?

AustinSaintAubin, 2018-06-22:
What distro are you running?
`cat /etc/*-release`

lspiker, 2018-06-22:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

AustinSaintAubin, 2018-06-22:
I think that there might be an issue with 'Raspbian GNU/Linux 9 (stretch)' and node.
You might try fresh install on `Raspbian GNU/Linux 8 (jessie)`, just as a test.

I will do a fresh install of CNCjs on 'Raspbian GNU/Linux 9 (stretch)' this weekend to test on my end.

lspiker, 2018-06-23:
I installed Jessie Lite. There is no longer an issue with the Pi using 100%
of cpu. However, the x and y machine and work positions do not change from
0 when the router is moved. I previously had a much earlier version of
cncjs installed and did not have any issues. Unfortunately, I do not know
now which version I previously had installed.

AustinSaintAubin, 2018-06-25:
I just published [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian), per-built Raspbian Pi image. Give it a try and let me know if you still have issues.

AustinSaintAubin, 2018-06-26:
Tested the image over the weekend and today. No issues with high CPU usage on:
```
Raspbian GNU/Linux 9 (stretch)
CNCjs v1.9.15
PM2: v2.10.4
NPM: v6.1.0
NODE: v10.5.0
```

nickm324, 2018-06-26:
Are you referring to the image you built or the normal distribution of CNCJs?

lspiker, 2018-06-26:
I initially had issues with the image, but they resolved when I updated to
Grbl 1.1 on my Shapeoko XL. I would note that the pendant process in pm2 on
the image results in 100% cpu when you do not have a pendant.

AustinSaintAubin, 2018-06-27:
I thought I had fixed that.... 
Just kill that pm2 process `cncjs-pendant-raspi-gpio`, or if you plan to use that pi-gpio pendant, make sure that its connecting to the correct serial port.
The pendant is trying to connect to a serial port, if it fails it gets restarted. I might need to add a max retries to cncjs-pendant-raspi-gpio, so pm2 will give up after a min or two.

Change: `/dev/ttyUSB0` as needed.
`pm2 start ~/.cncjs/cncjs-pendant-raspi-gpio/bin/cncjs-pendant-raspi-gpio -- --port /dev/ttyUSB0`

cheton, 2018-07-01:
> Are you referring to the image you built or the normal distribution of CNCJs?

@nickm324 
You can find the distribution archive at https://github.com/cncjs/cncjs-pi-raspbian/releases