cncjs/cncjs #297: Is there a way to use CNC JS with a servo for a z axis?

joepuzzo, 2018-03-09:


neilferreri, 2018-03-09:
CNCjs is a gcode sender. As long as your setup and firmware are capable, you can run a servo for a Z-axis. I know there is a grbl fork with this capability.

nesquik011, 2018-03-10:
you can control anything with cncjs stepper servo whatever , its like smart phone you can call dial up or call via internet just connect your main board with cncjs and make sure you have made the setup right

joepuzzo, 2018-03-12:
Do you have an example of how i would set that up? I have a cnc i have built and it uses grbl on an arduino but grbl does not support substituting a servo in for a stepper as one of the main axis

neilferreri, 2018-03-12:
https://github.com/cprezzi/grbl-servo

MitchBradley, 2018-03-12:
You could try this version of GRBL - https://github.com/robottini/grbl-servo

I found it by searching for "grbl servo".  I have no idea how well it works.

joepuzzo, 2018-03-12:
Yeah unfortunately all that does is let you thurn servo with an M3 gcode command :( . Cant use Z1 or somthing

neilferreri, 2018-03-12:
@joepuzzo Maybe if you provide more detail on your setup/machine, it would be easier to help.

MitchBradley, 2018-03-12:
Now that I think about it in more detail, I suspect that GRBL won't be able to handle servos.  GRBL running on AVR Arduino is already pushing hard against the limits of the hardware platform.  Between GRBL v0.9 and v1.1, they had to remove some features (text strings describing parameters) in order to make room for new features.  When you get to that point, new development gets harder and harder.

G2core (descendent of TinyG) is in a better position because it runs on ARM which has more memory - but I don't think it has great support for cnc-grade servo motors.  I searched for "servo" in the g2core source and found only support for "hobby servos" - those little servo motors that are used for remote control toys and stuff like that.  Maybe that could be adapted for a bigger servo motor, but I would expect that you would have to solve some gnarly problems.  Closed-loop motion control can be tricky.

If you want to use CNCjs, you might have to use "hybrid stepper servo" hardware, where special driver hardware implements the closed-loop control function, but from the standpoint of GRBL or TinyG or whatever, the hardware presents itself as a stepper, with just STEP and DIR lines to control the motion.  Here's an example:  https://www.amazon.com/dp/B07411XP67/ref=sspa_dk_detail_3?psc=1

Another example of a servo system that looks like a stepper is the ClearPath-SD product line.  Google it.  I don't have any experience with either of the above, I just found them by searching.

joepuzzo, 2018-03-12:
Thanks! I will take a look

joepuzzo, 2018-03-13:
Hmm looks liketheese are closed loop stepper drivers. Not for servos :( the issue is we are building a two axis grinder and one axis uses a linear actuator with servo. Our goal was to control the two axis using gcode with cncjs as interface.. i wonder if anyone has created a coverter from stepper to pwm. ( obviously with some loss in accuracy )

MitchBradley, 2018-03-13:
Google is your friend.  This is not the best place to ask, because CNCjs is agnostic about the details of the interface between the motor and the control firmware.   If you ask on GRBL, TinyG, Smoothie, etc forums, you might get useful answers from people who have solved the problem but who use non-CNCjs UIs.  If the problem is solved at the GRBL/TinyG/Smoothie level, the same solution will work with CNCjs.  Continuing to ask here is unlikely to get you very far.

joepuzzo, 2018-03-13:
Thanks will do 👍🏼

closed 2018-03-13 by joepuzzo