cncjs/cncjs #298: Running Local Development Server - error

hublerdesignworks, 2018-03-10:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.15
- Node.js: 9.7.1
- NPM: 5.7.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows 10
- [ ] Mac
- [ ] Linux

Followed the instructions below for Production Build and works fine :

Running Production Build

$ npm install  # Ensure that packages are installed
$ npm run prepare
$ ./bin/cnc -vv
  :  :  :
Server is listening on 0.0.0.0:8000

Followed the instructions below for Local Dev. Server with error (bottom of post) - why? :

Running Local Development Server

$ npm run dev

> cncjs@1.9.15 dev C:\cncjs-master
> npm run build-dev && npm run start-dev

> cncjs@1.9.15 prebuild-dev C:\cncjs-master
> bash scripts/prebuild-dev.sh

> cncjs@1.9.15 package-sync C:\cncjs-master
> node scripts/package-sync.js

/c/cncjs-master/src /c/cncjs-master
cnc.js -> ..\output\cnc.js
main.js -> ..\output\main.js
electron-app/AutoUpdater.js -> ..\output\electron-app\AutoUpdater.js
electron-app/log.js -> ..\output\electron-app\log.js
electron-app/menu-template.darwin.js -> ..\output\electron-app\menu-template.darwin.js
electron-app/menu-template.default.js -> ..\output\electron-app\menu-template.default.js
electron-app/menu-template.js -> ..\output\electron-app\menu-template.js
electron-app/WindowManager.js -> ..\output\electron-app\WindowManager.js
electron-app/WinSquirrelStartupEventHandler.js -> ..\output\electron-app\WinSquirrelStartupEventHandler.js
/c/cncjs-master
[... 149 lines cut ...]
> cncjs@1.9.15 start-dev C:\cncjs-master
> NODE_ENV=development ./bin/cnc -vv -p 8000 -m /widget:https://cncjs.github.io/cncjs-widget-boilerplate/v1/

'NODE_ENV' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.15 start-dev: `NODE_ENV=development ./bin/cnc -vv -p 8000 -m /widget:https://cncjs.github.io/cncjs-widget-boilerplate/v1/`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.15 start-dev script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-03-10T18_17_49_786Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cncjs@1.9.15 dev: `npm run build-dev && npm run start-dev`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the cncjs@1.9.15 dev script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\huble\AppData\Roaming\npm-cache\_logs\2018-03-10T18_17_49_834Z-debug.log

cheton, 2018-03-11:
I just made an update to the "start-dev" script that allows to set environment variables on Windows platform using cross-env. You can update the line in your package.json file as below:
https://github.com/cncjs/cncjs/blob/master/package.json#L62

```bash
"start-dev": "cross-env NODE_ENV=development ./bin/cnc -vv -p 8000 -m /widget:https://cncjs.github.io/cncjs-widget-boilerplate/v1/",
```

hublerdesignworks, 2018-03-13:
Perfect! That did the job - thanks.