cncjs/cncjs #299: Z probe after tool change

MeDerek, 2018-03-14:
#### Description
when running a job that has a tool change (M6) the new too can not be zeroed bu using the "Z-Probe" button. the job must be stopped before the "Z-Probe" button is enabled.  the macros appear to be enabled, so this could be a quick work around (if I can find the proper macro). a minor inconvenience, but I have become so dependent on the z probe zero out.

#### Versions

- CNCjs: 1.9.15
- Node.js: 7.10.1
- NPM: 5.5.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [X] Linux

MeDerek, 2018-03-15:
I forgot to mention that tool change with Z probing was possible in version 1.9.14

cheton, 2018-03-15:
I need to change the code in the [onClick](https://github.com/cncjs/cncjs/blob/v1.9.15/src/web/widgets/Probe/index.jsx#L403-L405) function from 

```js
canClick() {
    :    :    :
    if (workflow.state !== WORKFLOW_STATE_IDLE) {
        return false;
    }
   :    :    :
```

to 

```js
canClick() {
    :    :    :
    if (workflow.state === WORKFLOW_STATE_RUNNING) {
        return false;
    }
   :    :    :
}
```

That is, the above change will disable "Z-probe" button only when workflow is in the running state.

cheton, 2018-03-15:
BTW, I just compared the code between v1.9.14 and v1.9.15 and cannot found any differences:
* v1.9.14
  https://github.com/cncjs/cncjs/blob/v1.9.14/src/web/widgets/Probe/index.jsx#L403-L417
* v1.9.15
  https://github.com/cncjs/cncjs/blob/v1.9.15/src/web/widgets/Probe/index.jsx#L403-L417

I don't think you can run Z-probe with v1.9.14 during a pause of M6 tool change when running a job, maybe you are in a condition where Grbl's active state is idle and workflow state is idle as well.

closed 2026-03-13 by cheton