cncjs/cncjs #301: Gcode and Javascript for circle hole center finding

bostonbladeco, 2018-03-21:
#### Description

Here is Gcode and javascript for finding the center of a hole using a digital probe that is connected to the probe pin of GRBL. I was not sure the best way to contribute this but had some related questions. I do not take responsibility if this crashes your machine or probe, beware. I have been using this on a converted mid-size CNC mill running GRBL

I was hoping to receive feedback as to the best way to turn this into a macro or widget. It would take input of feedrate, distance and auto move to work coordinates X,Y as center at end of routine. I can't figure out how to take the PRB results from the gcode, access those in JS then generate the needed gcode and execute to move to center. Seems like macros were setup to be gcode only?

Gcode: (code moves back 20mm then centers, then forward 20mm, then centers, then right 20mm then centers at 30mm feedrate, ideally it should move to three points at 60 degrees from center for better accuracy but my current probe is picky in non-cardinal directions) The results of the probing get output as PRB in console

%X0=posx,Y0=posy
G91 G38.2 Y20 F30
%YBACK = posy
G1 X[X0] Y[Y0]
G91 G38.2 Y-20 F30
%YFRONT = posy
G1 X[X0] Y[Y0]
G91 G38.2 X20 F30
%XRIGHT = posx
G1 X[X0] Y[Y0]

-----

Javascript function to compute center coordinates:
//Take the respective coordinates from PRB output and in put to this function. Run in browser console and put the end coordinates it returns into you cncjs
//This function takes 3 points from the arc of a circle in X,Y and returns the center of that circle in X,Y. It may not work correctly in all environments if you send it negative coordinates and can probably be improved in relation to handling that.

var circ = function (x1,y1,x2,y2,x3,y3) {

var p1 = {x:x1,y:y2};
var p2 = {x:x2,y:y2};
var p3 = {x:x3,y:y3};

//this is necessary to prevent div/0 problems
if (p2.x-p1.x == 0) {
	p2.x+=0.0001;
}

if (p3.x-p2.x == 0) {
	p3.x+=0.0001;
}

var t1=(p2.y-p1.y)/(p2.x-p1.x); 
var t2=(p3.y-p2.y)/(p3.x-p2.x);
var t0=((t1*t2)*(p1.y-p3.y)+(t2*(p1.x+p2.x))-(t1*(p2.x+p3.x)))/(2*(t2-t1));
var t3=(-1/t1)*(t0-(p1.x+p2.x)/2)+(p1.y+p2.y)/2;
var result=(Math.sqrt(((p2.x-t0)*(p2.x-t0))+((p2.y-t3)*(p2.y-t3))))*2;

return [t0,t3];

}

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- Yarn but it is a PITA
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

abdulasis12, 2018-05-23:
hi bro how to download this script ?

bostonbladeco, 2018-05-25:
I was literally asking how best to contribute this to be useable for everyone and got no response. I didn't see enough documentation on how widgets work to permit data exchange between JS and gcode to, simply, build a widget to contribute this. Manually speaking you give the javascript call the position information from the MDI of cncjs and then re-input the result into the gcode and run it. I run it in another browser tab but it's a huge pain.