cncjs/cncjs #302: Cannot seem to get tool change macro to work with TLO

bostonbladeco, 2018-03-21:
#### Description

I am attempting to use the tool change macro from (2.2 TLO): https://github.com/cncjs/cncjs/wiki/Tool-Change

Despite my best efforts and a comprehensive review of the other bug reports I cannot seem to get it to work nor understand how it was supposed to work to begin with. It is possible the issue is more GRBL related but needed to start with cncjs because of the errors I'm seeing.

I have a z setter probe that triggers the probe pin on GRBL when pressure is made. It is in a fixed location in the machine space of a cnc mill. It's quite possible I'm just being dumb.

Problem #1: The tool setter is in a specific location in machine coordinates. The macro does not use G53 so seems to depend on WCS being in the same position related to the tool setter.'

Problem #2: GRBL will not accept G43.1 as it appears in the macro. Returns "unsupported command". GRBL seems to only accept G43.1 if is immediately follows something like: T1 M6 G43 .

Problem #3: My tools vary in length quite a bit so to avoid crashes I need to be able to have a pause to manually move the tool end to within probe distance of the probe

I am very willing to reprogram this but I would appreciate input on the best means to so either in terms of a widget or  macro or combination thereof. I am confused on cncjs capability to use javascript to drive a gcode action, take results from that action perform calculations, permit user manual movement (jog type), return more gcode then handoff back to the currently executing gcode program during M06 tool change.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

neilferreri, 2018-03-29:
> The macro does not use G53

I changed the commands in the macro to G53

> GRBL will not accept G43.1 as it appears in the macro. Returns "unsupported command"

There is an error in the posted macro:
`G43.1 [posz - TOUCH_PLATE_HEIGHT]` is missing the axis word. Should read `G43.1 Z[posz - TOUCH_PLATE_HEIGHT]`

As far as I can tell, I'd have to know the difference from my work zero to the top of my TLO touch plate. TOUCH_PLATE_HEIGHT shouldn't matter when I'm using a fixed location.

I got the following Macro to work.  Basically, it:

1. Goes to probe location (machine coordinates) and waits for user to continue (allows for wiring or placing of plate/switch).

1. Probes and stores posz at probe plate surface

1. Retracts and goes to tool change location (machine coordinates)...waits for user input to continue

1.  Goes back to probe location and waits for user input to continue

1.  Probes again, applies dynamic tool length offset based on difference of two probes, goes to tool change location (to remove wires, plates, alligator clips), and waits for user input to continue

1. Goes to previous work location and restores all modal states.

```
; Wait until the planner queue is empty
%wait

; Set user-defined variables
%CLEARANCE_HEIGHT = -10
%TOOL_CHANGE_X = -120
%TOOL_CHANGE_Y = -180
%TOOL_CHANGE_Z = -10
%TOOL_PROBE_X = -207
%TOOL_PROBE_Y = -177
%TOOL_PROBE_Z = -35
%PROBE_DISTANCE = 100
%PROBE_FEEDRATE = 100
%RETRACTION_DISTANCE = 10
; I'm not really sure about the touch plate height with a G43.1
%TOUCH_PLATE_HEIGHT = 0

; Keep a backup of current work position
%X0=posx, Y0=posy, Z0=posz

; Save modal state
; * Work Coordinate System: G54, G55, G56, G57, G58, G59
; * Plane: G17, G18, G19
; * Units: G20, G21
; * Distance Mode: G90, G91
; * Feed Rate Mode: G93, G94
; * Spindle State: M3, M4, M5
; * Coolant State: M7, M8, M9
%WCS = modal.wcs
%PLANE = modal.plane
%UNITS = modal.units
%DISTANCE = modal.distance
%FEEDRATE = modal.feedrate
%SPINDLE = modal.spindle
%COOLANT = modal.coolant

; Stop spindle
M5
; Absolute positioning
G90

; Go to Clearance Height
G53 Z[CLEARANCE_HEIGHT]
; Go to tool probe X,Y
G53 X[TOOL_PROBE_X]Y[TOOL_PROBE_Y]
; Go to probe Z - I set at WCS Z0 because my probe is below any work surface
G53 Z[TOOL_PROBE_Z]

; Wait until the planner queue is empty
%wait

; Pause the program before probing
M1

; Probe toward workpiece with a maximum probe distance
G91 ; Relative positioning
G38.2 Z-[PROBE_DISTANCE] F[PROBE_FEEDRATE]
G90 ; Absolute positioning
%ORIGINAL_TOOL = [posz]
; A dwell time of one second to make sure the planner queue is empty
G4 P1

; Retract from the touch plate
G91 ; Relative positioning
G0 Z[RETRACTION_DISTANCE]
G90 ; Absolute positioning

; Raise to tool change Z
G53 Z[TOOL_CHANGE_Z]
; Go to tool change X,Y
G53 X[TOOL_CHANGE_X] Y[TOOL_CHANGE_Y]
; Wait until the planner queue is empty
%wait
; Pause the program for a manual tool change
M0

; Go to Clearance Height
G53 Z[CLEARANCE_HEIGHT]
; Go to tool probe X,Y
G53 X[TOOL_PROBE_X]Y[TOOL_PROBE_Y]
; Go to probe Z - I set at WCS Z0 because my probe is below any work surface
G53 Z[TOOL_PROBE_Z]

; Wait until the planner queue is empty
%wait

; Pause the program before probing
M1

; Cancel tool length offset
G49

; Probe toward workpiece with a maximum probe distance
G91 ; Relative positioning
G38.2 Z-[PROBE_DISTANCE] F[PROBE_FEEDRATE]
G90 ; Absolute positioning

; A dwell time of one second to make sure the planner queue is empty
G4 P1

; Update the tool length offset
G43.1 Z[posz-ORIGINAL_TOOL]

; Retract from the touch plate
G91 ; Relative positioning
G0 Z[RETRACTION_DISTANCE]
G90 ; Absolute positioning

; Raise to tool change Z
G53 Z[TOOL_CHANGE_Z]
; Go to tool change X,Y
G53 X[TOOL_CHANGE_X] Y[TOOL_CHANGE_Y]
; Wait until the planner queue is empty
%wait

; Pause the program for cleanup (e.g. remove touch plate, wires, etc)
M0

; Go to previous work position
G0 X[X0] Y[Y0]
G0 Z[Z0]

; Restore modal state
[WCS] [PLANE] [UNITS] [DISTANCE] [FEEDRATE] [SPINDLE] [COOLANT]
```

justinclift, 2018-08-26:
@cheton ping? :smile: