cncjs/cncjs #303: Shuttle express shortcut issue

eglrd, 2018-03-21:
#### Description

'had to revert back to 1.9.14.3062 on MacOS high sierra using electron app: 
Changes in shortcuts created a conflict between switching axis to A  and jogging back. 

'Tried to get shuttle express configured with [ and] characters to resolve that, but it seems there is no way to enter those characters into limited(to not use another word) ShuttleExpress interface.

same trouble is generating headaches with + (plus) character that this interface is not accessing for some reason.

is there a way to use ( instead of [ for shortcuts
... or is there a way to edit shortcut file for electron app to sort this without impacting your devs ?

#### Versions

- CNCjs:  1.9.14.3062
- Node.js: 8.9.3
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [x] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [ x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [x ] Mac
- [ ] Linux

eglrd, 2018-03-22:
Hi Cheton, 
I did find a fix for this, so you can close it for now.
To solve it I had to install Unicode Hex software keyboard on my mac, and I could access to the special characters that were not accessible in ShuttleExpress with my standard keyboard. Hence I could configure to match the shortcut you had chosen: don't waste your time here.
... many thx for this awsome work. 'Looking forward to grow my knowledge to be able to contribute at some point.
e

eglrd, 2018-03-22:
... although, "+" remains a special character for ShuttleExpress, that is not allowed it seems... maybe could you setup another character instead of it ?
best.
e

cheton, 2018-03-22:
I found a shortcut conflict in 1.9.15:

<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>b</kbd> - Select/Deselect B Axis (_Supported in v1.9.15_)
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>[</kbd> or <kbd>b</kbd> (alias) - Jog Backward

The shortcut "b" will toggle B axis and then jog backward. I will remove the alias key "b" in the next version to prevent jogging backward.

Otherwise, some shortcuts, like "+" and "-", were added since 1.9.15. Did you try it with 1.9.14?
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>=</kbd> - Toggle Jog Distance
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>+</kbd> - Increase Jog Distance (_Supported in v1.9.15_)
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>-</kbd> - Decrease Jog Distance (_Supported in v1.9.15_)

eglrd, 2018-03-22:
Reason I did revert to 1.9.14 was this conflict you pointed that forced to use ctrl + alt + command + [ to jog forward instead of ctrl + alt + command + b in 1.9.15.
On top I think "+" is a reserved character in ShuttleExpress so it is impossible to increase jog distance using the shuttle express. 
Now back in 1.9.15, 
ctrl + alt + command + [ works perfect
ctrl + alt + command + ] works
ctrl + alt + command + = works fine
ctrl + alt + command + - works too
ctrl + alt + command + + cannot be configured (+ is a forbiden value for shuttle express ?)

eglrd, 2018-03-22:
I found a workaround to solve the issue: I hooked a keyboard with a numeric keypad and could enter "num +" in shuttle software. I'm closing this issue as with the proper config one can get it to work.

happy times.

closed 2018-03-22 by eglrd