cncjs/cncjs #305: Unable to connect to UI in any browser other than raspberry pi

labmat18, 2018-03-28:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.
Can't see UI in Chrome or Firefox on my PC. Did I miss something? I go to http://127.0.1.1:8888/ on raspberry pi can I get the cncjs UI but on Chome, on my pc, all I see is 127 8888 refused to connect.   
#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2018-03-28:
127.0.0.1 is a special IP address that means "this machine".  On the Pi, that address lets programs running on the Pi talk only to other programs running on the same Pi.  On the PC, that address lets programs on the PC talk only to other programs on the PC.

If you want to use a browser on the PC to connect to CNCjs on the Pi, you must use the Pi's external IP address.  To find it, you can type "uname -a" in a terminal window on the Pi.  If you have a typical WiFi setup, the address might be something like "192.168.0.xxx".  The prefix "192.168." is another special form of IP address that means "this local network".  By using that 192.168. prefix, machines on a local network can talk to one another, but machines from the outside cannot "reach in" and find one of those machines.

labmat18, 2018-03-28:
Thank you so much! I feel really dumb now, but thank you so much for helping!!!

closed 2018-03-28 by labmat18