cncjs/cncjs #306: 4th axis data not updating

eglrd, 2018-03-29:
#### Description
Hooked a 4th axis attached to last version firmware smoothieboard (build version: edge-4f622ed). Got A axis to work: either by entering direct commands or by using Axes interface pressing + and -. Awsome.

Data fields for A axis are not updating though, neither WCS or MCS. 
M114 is not showing A value: might be related ?

#### Versions

- CNCjs: Version 1.9.15 (1.9.15.3078)
- Node.js: v8.9.3
- NPM: 5.6.0

#### How Do You Install CNCjs?

- [x] NPM
- [ x] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [x] Mac
- [ ] Linux

cheton, 2018-03-29:
The data fields rely on the result of M114. Could you type the following commands on the console widget and dump the output?

M-code | Description
---- | ----
M114 | Show current position of all axes, XYZ will be the last requested position
M114.1 | Show current real time position of all axes 
M114.2 | Show current real time machine position of all axes
M114.3 | Show current real time actuator position of all actuators
M114.4 | Show last milestone
M114.5 | Show last machine position

eglrd, 2018-03-29:
'Will do a dump but I'm positive M114 / M114.1 display only X Y Z values (no A).

I'm using latest build on edge branch of Smoothieware that is now 5 axis, BUT it is documented not to provide A coordinates unless compiled with PAXIS=4 (Make PAXIS=4 CNC=1). I'm struggling to get such a setup compiled on my side and had to revert to latest official build from smoothieware to get everything to move. I suspect this build is compiled without PAXIS=4 option ?

+I'm writing a processor for fusion 360 to output gcode for a smoothie with 4th axis on.

nesquik011, 2018-03-29:
@eglrd  and can i set auto tool + 5 axis steppers ?

eglrd, 2018-03-30:
Hi. you're lucky I looked at this closed item... Smoothie is in the process of moving to 5 axis. Starting to work now. I'm focused on 4 axis on my side.
> yes you can control 5 steppers (withoutside drivers)
> not sure about auto tool, but I can't see why (providing the tool dispenser is having his own controller). There are plenty of pin available on the smoothie.

closed 2018-03-30 by eglrd