cncjs/cncjs #307: Feed Rate Override - tinyG

swpaxton66, 2018-03-31:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.14

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [ x] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ x] Windows
- [ ] Mac
- [ ] Linux

Hello Cheton,

I use cncjs with tinyG.  Do you have any plans to implement a feed rate override?  Currently I don't see any options to override the feed rate (unless I missed it).  As you know I can change the feed rate in the console window but as soon as the program sends a new feed rate command the feed rate reverts back to the original program state.  I find it extremely helpful and much more productive to be able to adjust the feedrate while the cutting job is in process.

Thanks,

Scott

cheton, 2018-04-04:
The override control buttons will be visible when it's supported by the firmware. You can type `{sso:n}`, `{mfo:n}`, and `{mto:n}` in the console to check the result.

[image]

cheton, 2018-04-04:
The firmware version and build version of mine:
```
> {fb:n}
{"r":{"fb":100.19},"f":[1,0,8]}
> {fv:n}
{"r":{"fv":0.99},"f":[1,0,8]}
```

swpaxton66, 2018-04-05:
Hi Cheton,

I don't recognize the tinyG version you have.  I have a V8 @ 440.2 which is the latest build.  I tried the unstable edge release but it doesn't play nice with my board.  What version of the tinyG do you have?

Thanks.

cheton, 2018-06-30:
Feed rate override was supported since edge-0.98:
https://github.com/synthetos/TinyG/blob/edge-0.98/firmware/tinyg/spindle.h#L81

You can ask for support there to see if there is any chance to update your firmware to this version.

closed 2018-06-30 by cheton