cncjs/cncjs #309: PS3 Controller / Desktop CNCjs

spankurmonkey, 2018-04-05:
#### Description

Want to use PS3 controller with desktop CNCjs.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [X ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [X] Windows
- [ ] Mac
- [ ] Linux

Fusseldieb, 2020-02-20:
I currently use [AntiMicro ](https://github.com/AntiMicro/antimicro)to map the PS4 controls to normal keyboard keys, then I enable the keyboard jogging in cncJS and voilá!

[image]

AustinSaintAubin, 2020-02-27:
There is also https://github.com/cncjs/cncjs-pendant-ps3