cncjs/cncjs #311: Greater Axis Control

JorgenCollinson, 2018-04-09:
#### Description

Am new'ish to CNC, really like the CNCJS GUI.
Is there a way to invert the X/Y/Z Axis default directions.

Found that , Easel default for Z was different to Z Axis CNCJS direction.
In the Axis Widget, can we expand the options to allow the setting of default direction per Axis.

#### Versions
- CNCjs: 1.9.15
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2018-04-11:
This likly means your GRBL motion controller has not been configured.
If you have not yet, you need to [setup / configure your X-Carve](http://x-carve-instructions.inventables.com/1000mm/step10/).
You might also read up on [GRBL: direction port invert mask](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask) (look at $2 for mask setting and values chart).

You might also watch [my video about setting up X-Carve for laser work](https://youtu.be/DOdkc9_lqtA). I talk about GRBL setup.

Fusseldieb, 2018-06-14:
I think "issues" like these can be closed. There are a lot around here which even aren't.
When I first looked at the "Issues" tab it almost shocked me, but then upon opening I found that 90% aren't.

closed 2018-06-15 by cheton