cncjs/cncjs #312: Marlin not sending commands

jeffeb3, 2018-04-10:
#### Description

I have a flashed Marlin machine (version 1.1.6, I configured). When I connect with Marlin, I see the output of the M115 command, and for whatever reason, it doesn't end in 'ok'.

This causes the queries to not send, and even if I push the movement buttons, it queues up the commands.

If I just type something like G0 X0 in the command line, it will receive an 'ok' message, which then "unfreezes" cnc.js. Then the queries start sending, and any home/movement/etc. commands get sent immediately.

I traced it down to the MarlinController.prototype.command function, and the feeder doesn't get the next() called unless:

```
                { // The following criteria must be met to trigger the feeder
                    const notBusy = !(this.history.writeSource);
                    const senderIdle = (this.sender.state.sent === this.sender.state.received);
                    const feederIdle = !(this.feeder.isPending());

                    if (notBusy && senderIdle && feederIdle) {
                        this.feeder.next();
                    }
                }
```

I put a print in an else on that if, and notBusy is false, and feederIdle is false.

```
                    if (notBusy && senderIdle && feederIdle) {
                        _this4.feeder.next();
                    }
                    else {
                        log.silly(`not feeding: notBusy: ${notBusy} senderIdle: ${senderIdle} feederIdle: ${feederIdle}`);
                    }

```

```
2018-04-10T02:57:56.363Z - debug: service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Marlin","baudrate":115200,"rtscts":false}): id=DZ4fT--RxL1gJI04AAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16793:21)
2018-04-10T02:57:56.452Z - debug: controller:Marlin Add socket connection: id=DZ4fT--RxL1gJI04AAAA     at MarlinController.addConnection (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20602:13)
2018-04-10T02:57:56.683Z - debug: controller:Marlin Connected to serial port "/dev/ttyUSB0"     at SerialPort.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20538:17)
2018-04-10T02:57:57.828Z - silly: controller:Marlin > M115
2018-04-10T02:57:58.154Z - silly: controller:Marlin < start
2018-04-10T02:57:58.177Z - silly: controller:Marlin < echo:Marlin 1.1.6
2018-04-10T02:57:58.195Z - silly: controller:Marlin < echo: Last Updated: 2017-10-10 12:00 | Author: Jeffeb3
2018-04-10T02:57:58.200Z - silly: controller:Marlin < echo:Compiled: Feb  8 2018
2018-04-10T02:57:58.206Z - silly: controller:Marlin < echo: Free Memory: 12675  PlannerBufferBytes: 1312
2018-04-10T02:57:58.212Z - silly: controller:Marlin < echo:EEPROM version mismatch (EEPROM=V38 Marlin=V41)
2018-04-10T02:57:58.217Z - silly: controller:Marlin < echo:Hardcoded Default Settings Loaded
2018-04-10T02:57:58.223Z - silly: controller:Marlin < echo:  G21    ; Units in mm
2018-04-10T02:57:58.229Z - silly: controller:Marlin < echo:  M149 C ; Units in Celsius
2018-04-10T02:57:58.233Z - silly: controller:Marlin < echo:Filament settings: Disabled
2018-04-10T02:57:58.238Z - silly: controller:Marlin < echo:  M200 D3.00
2018-04-10T02:57:58.242Z - silly: controller:Marlin < echo:  M200 D0
2018-04-10T02:57:58.247Z - silly: controller:Marlin < echo:Steps per unit:
2018-04-10T02:57:58.253Z - silly: controller:Marlin < echo:  M92 X80.00 Y80.00 Z400.00 E100.00
2018-04-10T02:57:58.259Z - silly: controller:Marlin < echo:Maximum feedrates (units/s):
2018-04-10T02:57:58.291Z - silly: controller:Marlin < echo:  M203 X300.00 Y300.00 Z15.00 E25.00
2018-04-10T02:57:58.297Z - silly: controller:Marlin < echo:Maximum Acceleration (units/s2):
2018-04-10T02:57:58.302Z - silly: controller:Marlin < echo:  M201 X700 Y700 Z100 E10000
2018-04-10T02:57:58.306Z - silly: controller:Marlin < echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2018-04-10T02:57:58.311Z - silly: controller:Marlin < echo:  M204 P700.00 R3000.00 T700.00
2018-04-10T02:57:58.317Z - silly: controller:Marlin < echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
2018-04-10T02:57:58.322Z - silly: controller:Marlin < echo:  M205 S0.00 T0.00 B20000 X4.00 Y4.00 Z0.40 E1.00
2018-04-10T02:57:58.328Z - silly: controller:Marlin < echo:Home offset:
2018-04-10T02:57:58.333Z - silly: controller:Marlin < echo:  M206 X0.00 Y0.00 Z0.00
2018-04-10T02:57:58.337Z - silly: controller:Marlin < echo:Auto Bed Leveling:
2018-04-10T02:57:58.342Z - silly: controller:Marlin < echo:  M420 S0 Z0.00
2018-04-10T02:57:58.347Z - silly: controller:Marlin < echo:Material heatup parameters:
2018-04-10T02:57:58.351Z - silly: controller:Marlin < echo:  M145 S0 H180 B60 F0
2018-04-10T02:57:58.356Z - silly: controller:Marlin <   M145 S1 H240 B100 F0
2018-04-10T02:57:58.370Z - silly: controller:Marlin < echo:PID settings:
2018-04-10T02:57:58.374Z - silly: controller:Marlin < echo:  M301 P10.17 I0.45 D57.88
2018-04-10T02:57:58.380Z - silly: controller:Marlin < echo:Z-Probe Offset (mm):
2018-04-10T02:57:58.385Z - silly: controller:Marlin < echo:  M851 Z-0.40
2018-04-10T02:57:58.390Z - silly: controller:Marlin < echo:Linear Advance:
2018-04-10T02:57:58.396Z - silly: controller:Marlin < echo:  M900 K75.00 R0.00
2018-04-10T02:58:00.976Z - debug: service:cncengine socket.command("/dev/ttyUSB0", "gcode"): id=DZ4fT--RxL1gJI04AAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16897:21)
2018-04-10T02:58:00.994Z - silly: controller:Marlin not feeding: notBusy: false senderIdle: true feederIdle: false
2018-04-10T02:58:01.008Z - debug: service:cncengine socket.command("/dev/ttyUSB0", "gcode"): id=DZ4fT--RxL1gJI04AAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16897:21)
2018-04-10T02:58:01.014Z - silly: controller:Marlin not feeding: notBusy: false senderIdle: true feederIdle: false
2018-04-10T02:58:01.028Z - debug: service:cncengine socket.command("/dev/ttyUSB0", "gcode"): id=DZ4fT--RxL1gJI04AAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16897:21)
2018-04-10T02:58:01.034Z - silly: controller:Marlin not feeding: notBusy: false senderIdle: true feederIdle: false
```

Maybe there needs to be a timer somewhere, in case the 'ok' gets lost. The M115 isn't printing it (which I think is a mistake, really), but if the 'ok' got garbled during a gcode carving, that could shut the whole thing down.

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.2.1
- NPM: 3.9.3

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Marlin
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

cheton, 2018-04-14:
Hi @jeffeb3 

According to your attached log, it seems the **M115** command was sent before receiving the **start**:

```
2018-04-10T02:57:56.363Z - debug: service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Marlin","baudrate":115200,"rtscts":false}): id=DZ4fT--RxL1gJI04AAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16793:21)
2018-04-10T02:57:56.452Z - debug: controller:Marlin Add socket connection: id=DZ4fT--RxL1gJI04AAAA     at MarlinController.addConnection (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20602:13)
2018-04-10T02:57:56.683Z - debug: controller:Marlin Connected to serial port "/dev/ttyUSB0"     at SerialPort.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20538:17)
2018-04-10T02:57:57.828Z - silly: controller:Marlin > M115
2018-04-10T02:57:58.154Z - silly: controller:Marlin < start
2018-04-10T02:57:58.177Z - silly: controller:Marlin < echo:Marlin 1.1.6
```

Could you try increasing the delay `await delay(1000);` to a higher value or comment out the line of `this.command('gcode', 'M115');` to check if it works from your side? 

https://github.com/cncjs/cncjs/blob/master/src/app/controllers/Marlin/MarlinController.js#L826-L834
```jsx
async initController() {
    // Wait for the bootloader to complete before sending commands
    await delay(1000);

    // M115: Get Firmware Version and Capabilities
    this.command('gcode', 'M115');

    this.ready = true;
}
```

jeffeb3, 2018-04-15:
Thanks for taking a look.

So commenting out the M115 did not fix the problem:

```2018-04-15T03:48:11.678Z - debug: service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Marlin","baudrate":115200,"rtscts":false}): id=McCDg9gcP61oSE1ZAAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16793:21)
2018-04-15T03:48:11.784Z - debug: controller:Marlin Add socket connection: id=McCDg9gcP61oSE1ZAAAA     at MarlinController.addConnection (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20602:13)
2018-04-15T03:48:11.933Z - debug: controller:Marlin Connected to serial port "/dev/ttyUSB0"     at SerialPort.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20538:17)
2018-04-15T03:48:13.442Z - silly: controller:Marlin < start
2018-04-15T03:48:13.480Z - silly: controller:Marlin < echo:Marlin 1.1.6
2018-04-15T03:48:13.494Z - silly: controller:Marlin < echo: Last Updated: 2017-10-10 12:00 | Author: Jeffeb3
2018-04-15T03:48:13.501Z - silly: controller:Marlin < echo:Compiled: Feb  8 2018
2018-04-15T03:48:13.507Z - silly: controller:Marlin < echo: Free Memory: 12675  PlannerBufferBytes: 1312
2018-04-15T03:48:13.512Z - silly: controller:Marlin < echo:EEPROM version mismatch (EEPROM=V38 Marlin=V41)
2018-04-15T03:48:13.518Z - silly: controller:Marlin < echo:Hardcoded Default Settings Loaded
2018-04-15T03:48:13.523Z - silly: controller:Marlin < echo:  G21    ; Units in mm
2018-04-15T03:48:13.528Z - silly: controller:Marlin < echo:  M149 C ; Units in Celsius
2018-04-15T03:48:13.532Z - silly: controller:Marlin < echo:Filament settings: Disabled
2018-04-15T03:48:13.537Z - silly: controller:Marlin < echo:  M200 D3.00
2018-04-15T03:48:13.541Z - silly: controller:Marlin < echo:  M200 D0
2018-04-15T03:48:13.546Z - silly: controller:Marlin < echo:Steps per unit:
2018-04-15T03:48:13.550Z - silly: controller:Marlin < echo:  M92 X80.00 Y80.00 Z400.00 E100.00
2018-04-15T03:48:13.555Z - silly: controller:Marlin < echo:Maximum feedrates (units/s):
2018-04-15T03:48:13.559Z - silly: controller:Marlin < echo:  M203 X300.00 Y300.00 Z15.00 E25.00
2018-04-15T03:48:13.590Z - silly: controller:Marlin < echo:Maximum Acceleration (units/s2):
2018-04-15T03:48:13.596Z - silly: controller:Marlin < echo:  M201 X700 Y700 Z100 E10000
2018-04-15T03:48:13.605Z - silly: controller:Marlin < echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2018-04-15T03:48:13.610Z - silly: controller:Marlin < echo:  M204 P700.00 R3000.00 T700.00
2018-04-15T03:48:13.615Z - silly: controller:Marlin < echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
2018-04-15T03:48:13.621Z - silly: controller:Marlin < echo:  M205 S0.00 T0.00 B20000 X4.00 Y4.00 Z0.40 E1.00
2018-04-15T03:48:13.626Z - silly: controller:Marlin < echo:Home offset:
2018-04-15T03:48:13.631Z - silly: controller:Marlin < echo:  M206 X0.00 Y0.00 Z0.00
2018-04-15T03:48:13.636Z - silly: controller:Marlin < echo:Auto Bed Leveling:
2018-04-15T03:48:13.641Z - silly: controller:Marlin < echo:  M420 S0 Z0.00
2018-04-15T03:48:13.645Z - silly: controller:Marlin < echo:Material heatup parameters:
2018-04-15T03:48:13.650Z - silly: controller:Marlin < echo:  M145 S0 H180 B60 F0
2018-04-15T03:48:13.654Z - silly: controller:Marlin <   M145 S1 H240 B100 F0
2018-04-15T03:48:13.659Z - silly: controller:Marlin < echo:PID settings:
2018-04-15T03:48:13.664Z - silly: controller:Marlin < echo:  M301 P10.17 I0.45 D57.88
2018-04-15T03:48:13.669Z - silly: controller:Marlin < echo:Z-Probe Offset (mm):
2018-04-15T03:48:13.674Z - silly: controller:Marlin < echo:  M851 Z-0.40
2018-04-15T03:48:13.680Z - silly: controller:Marlin < echo:Linear Advance:
2018-04-15T03:48:13.685Z - silly: controller:Marlin < echo:  M900 K75.00 R0.00
2018-04-15T03:48:18.882Z - silly: controller:Marlin Reschedule current position query: now=1523764098880ms, timespan=5154ms
2018-04-15T03:48:20.860Z - debug: service:cncengine socket.command("/dev/ttyUSB0", "gcode"): id=McCDg9gcP61oSE1ZAAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16897:21)
2018-04-15T03:48:20.900Z - silly: controller:Marlin not feeding: notBusy: false senderIdle: true feederIdle: true
```

But changing the delay to 5000 did seem to fix it:

```
2018-04-15T03:50:51.532Z - debug: service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Marlin","baudrate":115200,"rtscts":false}): id=aeSkcU__qQBJjHWSAAAA     at Socket.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:16793:21)
2018-04-15T03:50:51.639Z - debug: controller:Marlin Add socket connection: id=aeSkcU__qQBJjHWSAAAA     at MarlinController.addConnection (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20602:13)
2018-04-15T03:50:51.798Z - debug: controller:Marlin Connected to serial port "/dev/ttyUSB0"     at SerialPort.<anonymous> (/usr/local/lib/node_modules/cncjs/dist/cnc/app/index.js:20538:17)
_Nhoy8VeLL6VJl3YYQOIV73EBNcQ5FgV GET /api/version/latest?_=1523764249602 200 1791.295 ms
2018-04-15T03:50:53.525Z - silly: controller:Marlin < start
2018-04-15T03:50:53.548Z - silly: controller:Marlin < echo:Marlin 1.1.6
2018-04-15T03:50:53.567Z - silly: controller:Marlin < echo: Last Updated: 2017-10-10 12:00 | Author: Jeffeb3
2018-04-15T03:50:53.573Z - silly: controller:Marlin < echo:Compiled: Feb  8 2018
2018-04-15T03:50:53.578Z - silly: controller:Marlin < echo: Free Memory: 12675  PlannerBufferBytes: 1312
[... cut ...]
2018-04-15T03:50:57.732Z - silly: controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.00","y":"0.00","z":"0.00","e":"0.00"},"raw":"X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0"}
2018-04-15T03:50:57.737Z - silly: controller:Marlin < ok
```

A big difference is the M114 commands that start coming through right away, even before the M115. I'm wondering if the longer delay in initController is really slowing something other than the M115 down, and that's really what's helping here. Between the `start` and the `M115`, there are 5 seconds, but in between, then M114 commands start firing.

If my guesses aren't helping, feel free ignoring them :)

jeffeb3, 2018-04-28:
Do you have any thoughts on the right solution for this? Is the longer pause the right choice?

ekstrand, 2018-05-06:
I am having the same issue.

cheton, 2018-06-30:
I compiled and flashed Marlin 1.8 to an MKS Gen v1.4 board, connected it to my RPi 2 via a USB cable. The baud rate is 250000bps, and hardware flow control is off for making a connection.

It looks good without issues since the M115 was sent after the startup message for my case. However, if I send M115 before receiving **start**, it will get stuck.

[image]

```
2018-06-29T23:56:35.858Z - debug: service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Marlin","baudrate":250000,"rtscts":false}): id=tYAG_In_JklptSC7AAAA     at Socket.<anonymous> (/home/cheton/cncjs/dist/cnc/app/index.js:16831:21)
2018-06-29T23:56:35.887Z - debug: controller:Marlin Add socket connection: id=tYAG_In_JklptSC7AAAA     at MarlinController.addConnection (/home/cheton/cncjs/dist/cnc/app/index.js:20639:13)
2018-06-29T23:56:35.925Z - debug: controller:Marlin Connected to serial port "/dev/ttyUSB0"     at SerialPort.<anonymous> (/home/cheton/cncjs/dist/cnc/app/index.js:20576:17)
2018-06-29T23:56:36.817Z - silly: controller:Marlin < start
2018-06-29T23:56:36.824Z - silly: controller:Marlin < echo:Marlin 1.1.8
2018-06-29T23:56:36.831Z - silly: controller:Marlin < echo: Last Updated: 2017-12-25 12:00 | Author: (none, default config)
2018-06-29T23:56:36.833Z - silly: controller:Marlin < echo:Compiled: Jun 29 2018
2018-06-29T23:56:36.835Z - silly: controller:Marlin < echo: Free Memory: 5539  PlannerBufferBytes: 1264
2018-06-29T23:56:36.836Z - silly: controller:Marlin < echo:Hardcoded Default Settings Loaded
2018-06-29T23:56:36.838Z - silly: controller:Marlin < echo:  G21    ; Units in mm
2018-06-29T23:56:36.839Z - silly: controller:Marlin < echo:Filament settings: Disabled
2018-06-29T23:56:36.840Z - silly: controller:Marlin < echo:  M200 D3.00
2018-06-29T23:56:36.841Z - silly: controller:Marlin < echo:  M200 D0
2018-06-29T23:56:36.843Z - silly: controller:Marlin < echo:Steps per unit:
2018-06-29T23:56:36.844Z - silly: controller:Marlin < echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
2018-06-29T23:56:36.850Z - silly: controller:Marlin < echo:Maximum feedrates (units/s):
2018-06-29T23:56:36.852Z - silly: controller:Marlin < echo:  M203 X300.00 Y300.00 Z5.00 E25.00
2018-06-29T23:56:36.853Z - silly: controller:Marlin < echo:Maximum Acceleration (units/s2):
2018-06-29T23:56:36.854Z - silly: controller:Marlin < echo:  M201 X3000 Y3000 Z100 E10000
2018-06-29T23:56:36.856Z - silly: controller:Marlin < echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2018-06-29T23:56:36.857Z - silly: controller:Marlin < echo:  M204 P3000.00 R3000.00 T3000.00
2018-06-29T23:56:36.861Z - silly: controller:Marlin < echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
2018-06-29T23:56:36.863Z - silly: controller:Marlin < echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
2018-06-29T23:56:36.865Z - silly: controller:Marlin < echo:Home offset:
2018-06-29T23:56:36.866Z - silly: controller:Marlin < echo:  M206 X0.00 Y0.00 Z0.00
2018-06-29T23:56:36.867Z - silly: controller:Marlin < echo:PID settings:
2018-06-29T23:56:36.869Z - silly: controller:Marlin < echo:  M301 P22.20 I1.08 D114.00
2018-06-29T23:56:37.112Z - silly: controller:Marlin < X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
2018-06-29T23:56:37.119Z - silly: controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.00","y":"0.00","z":"0.00","e":"0.00"},"raw":"X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0"}
2018-06-29T23:56:37.121Z - silly: controller:Marlin < ok
2018-06-29T23:56:37.125Z - silly: controller:Marlin controller.on('ok'): source=server, line="M114", res={"raw":"ok"}
2018-06-29T23:56:37.145Z - silly: controller:Marlin > M115
2018-06-29T23:56:37.158Z - silly: controller:Marlin < FIRMWARE_NAME:Marlin 1.1.8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2018-06-29T23:56:37.166Z - silly: controller:Marlin < Cap:SERIAL_XON_XOFF:0
2018-06-29T23:56:37.182Z - silly: controller:Marlin < Cap:EEPROM:0
2018-06-29T23:56:37.186Z - silly: controller:Marlin < Cap:VOLUMETRIC:1
2018-06-29T23:56:37.188Z - silly: controller:Marlin < Cap:AUTOREPORT_TEMP:1
2018-06-29T23:56:37.190Z - silly: controller:Marlin < Cap:PROGRESS:0
2018-06-29T23:56:37.191Z - silly: controller:Marlin < Cap:PRINT_JOB:1
2018-06-29T23:56:37.193Z - silly: controller:Marlin < Cap:AUTOLEVEL:0
2018-06-29T23:56:37.195Z - silly: controller:Marlin < Cap:Z_PROBE:0
2018-06-29T23:56:37.197Z - silly: controller:Marlin < Cap:LEVELING_DATA:0
2018-06-29T23:56:37.199Z - silly: controller:Marlin < Cap:BUILD_PERCENT:0
2018-06-29T23:56:37.201Z - silly: controller:Marlin < Cap:SOFTWARE_POWER:0
2018-06-29T23:56:37.203Z - silly: controller:Marlin < Cap:TOGGLE_LIGHTS:0
2018-06-29T23:56:37.205Z - silly: controller:Marlin < Cap:CASE_LIGHT_BRIGHTNESS:0
2018-06-29T23:56:37.207Z - silly: controller:Marlin < Cap:EMERGENCY_PARSER:0
2018-06-29T23:56:37.209Z - silly: controller:Marlin < ok
2018-06-29T23:56:37.211Z - silly: controller:Marlin controller.on('ok'): source=feeder, line="M115", res={"raw":"ok"}
2018-06-29T23:56:37.435Z - silly: controller:Marlin < X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
2018-06-29T23:56:37.437Z - silly: controller:Marlin controller.on('pos'): source=server, line="M114", res={"pos":{"x":"0.00","y":"0.00","z":"0.00","e":"0.00"},"raw":"X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0"}
[... cut ...]
2018-06-29T23:56:38.443Z - silly: controller:Marlin controller.on('ok'): source=server, line="M114", res={"raw":"ok"}
2018-06-29T23:56:38.940Z - silly: controller:Marlin < ok T:4.38 /0.00 @:0
2018-06-29T23:56:38.942Z - silly: controller:Marlin controller.on('temperature'): source=server, line="M105", res={"ok":true,"extruder":{"deg":"4.38","degTarget":"0.00","power":"0"},"heatedBed":{},"raw":"ok T:4.38 /0.00 @:0"}
2018-06-29T23:56:38.943Z - silly: controller:Marlin controller.on('ok'): source=server, line="M105", res=undefined
```

cheton, 2018-06-30:
This can be fixed without depending on the delay, just simply removing all the code in initController:

```js

async initController() {
    // Wait for the bootloader to complete before sending commands
    //await delay(1000);

    // M115: Get Firmware Version and Capabilities
    //this.command('gcode', 'M115');

    //this.ready = true;
}
```

and send M115 after receiving a `start`:
```js
this.controller.on('start', (res) => {
    this.emit('serialport:read', res.raw);

    // M115: Get Firmware Version and Capabilities
    this.command('gcode', 'M115');

    // Set ready flag to true when receiving a start message
    // Note: It might have chance of receiving garbage characters on startup due to electronic noise.
    this.ready = true;
});
```

johnboiles, 2019-04-11:
@cheton any major concerns with your solution or should I turn it into a PR?

johnboiles, 2019-04-13:
@cheton I put your fix in a PR #449 and will try it this weekend

cheton, 2019-04-13:
@johnboiles Thank you for reminding me. I will merge it to the master branch soon.

johnboiles, 2019-04-13:
Great! And I should get a chance to test today. Seems like a safe enough change though

johnboiles, 2019-04-14:
After about a day of testing I _think_ this is fixed

cheton, 2019-04-15:
@johnboiles Thank you for the verification. I will put it into CNCjs 1.9.19 release.

cheton, 2019-04-22:
Fixed in [1.9.19](https://github.com/cncjs/cncjs/releases/tag/v1.9.19)

closed 2019-04-22 by cheton