cncjs/cncjs #314: Gcode files greater than 1 mb never load

eciramella, 2018-04-19:
#### Description

Hello -

It seems when there's a CarbideCreate generated gcode file over 1 MB, it never loads.  To reproduce, I can attach the project file or gcode file if you want, but when I load it, CNCjs just spins and spins.
I've tailed the logs when trying to upload this way, but they're clean.  Everything else seems to be working fine and when the file is slightly smaller, it seems great.
[image]
[image]

#### Versions

- CNCjs: 1.9.15
- Node.js: v7.10.1
- NPM: 5.8.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

eciramella, 2018-04-19:
Huh - I'm struggling with files as small as 414kb actually.

eciramella, 2018-04-19:
After a LONG time, the logs spat this out:

```
2018-04-19T02:11:19.852Z - error: evaluateExpression evaluateExpression: src="", context=[object Object]
2018-04-19T02:11:19.876Z - error: evaluateExpression TypeError: Cannot read property 'expression' of undefined
    at evaluateExpression (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:2931:51)
    at Sender.dataFilter (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:17327:54)
    at SPCharCounting.callback (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:2512:41)
    at SPCharCounting.process (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:2396:31)
    at Sender.next (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:2704:21)
    at gcodeStart (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:18124:31)
    at GrblController.command (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:18402:9)
    at Socket.<anonymous> (/home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/dist/cnc/app/index.js:16905:36)
    at emitTwo (events.js:106:13)
    at Socket.emit (events.js:194:7)
    at /home/pi/.nvm/versions/node/v7.10.1/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js:503:12
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickDomainCallback (internal/process/next_tick.js:128:9)
```

cheton, 2018-04-20:
There might be something wrong with the parser. Could you please attach a G-code file? Thank you.

eciramella, 2018-04-21:
This is both the Carbide Create file as well as the generated g-code.

For what it's worth, when I upload the file to a "watch" folder, it takes a moment, but it loads.  Doing the upload via the web ui never succeeds.

[PaintSpinner.zip](https://github.com/cncjs/cncjs/files/1934118/PaintSpinner.zip)

cheton, 2018-05-05:
This error is occurred when parsing the line started with %. You can remove the first line "%" in your G-code file to mitigate this issue. I will fix this bug in the next release.

```
2018-04-19T02:11:19.852Z - error: evaluateExpression evaluateExpression: src="", context=[object Object]
2018-04-19T02:11:19.876Z - error: evaluateExpression TypeError: Cannot read property 'expression' of undefined
```

cheton, 2018-05-05:
Except for this error, I can successfully load the attached G-code file from the UI (or even the watch folder) within 2~3 seconds. May I know the generation of your Raspberry Pi?

closed 2019-05-09 by cheton