cncjs/cncjs #317: autozoom zooms in too far

rkoe, 2018-04-24:
#### Description

The auto-zoom in cnc.js does not show the complete drawing.
Tested on Linux with Chromium and (older) Firefox.

Example:
```
G0 X0 Y0
F500
G2 X100 Y0 I50 J0
```
only shows ~77% of the drawing; but autozoom should show the whole drawing plus some border.

[image]

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.11.5
- NPM: 3.10.10

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

MitchBradley, 2018-04-24:
It turns out that computing an accurate bounding box for a G2/G3 arc is not entirely trivial.  I'm not sure how the normal cncjs UI does it (maybe it relies on library code?).

For cncjs-shopfloor-tablet, I ended up writing my own arc bounding box calculation code, since the 2D canvas library has no such function.  In case it should be useful, that code is in https://github.com/cncjs/cncjs-shopfloor-tablet/blob/master/src/toolpath-displayer.js starting at line 151 - bboxhandlers .. addArcToCurve .  It's surprisingly tricky because you have to work out which quadrants the arc touches and there are a lot of cases to consider.

rkoe, 2018-04-25:
I don't think that arcs are the problem, since:

- cnc.js draws arcs as several lines (you can see it by zooming in)
- cnc.js autozoom also does not show the complete drawing for a simple line:

```
G0 X0 Y0
G0 X100 Y100
```
[image]