cncjs/cncjs #318: Marlin Inch Mode still shows MM in DRO

FlightFixit, 2018-04-26:
#### Description

When I am connected to a marlin control board and send a g20 to convert to inches this does not update in the DRO and it still remains mm. I think that marlin does not send over it's current position in inches, only mm. Is there a way that cncjs could have an option, or do it by default, that when G20 is enabled and marlin is connected to divide the DRO values by 25.4 so that they read inches? 

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2018-05-05:
What's the Marlin firmware version in your control board? G20 commnad was initially supported since v1.1.0. http://marlinfw.org/docs/gcode/G020.html

FlightFixit, 2018-05-05:
1.1.8, g20 is enabled, but the dro is receiving and displaying the dimensions in mm, even when the label on the dro is in inches.

aczanderna, 2019-06-25:
I have the same issue with Marlin on the latest version of cnc.js.  Was a solution ever found for this? Would it be easy to add a simple unit conversion for the DRO output values?

MitchBradley, 2019-06-25:
It is highly unlikely that a given Marlin controller supports G20 / inch mode.  Support for G20 was added in version 1.1.10 of the Marlin source code, but the feature is disabled by default.  Of the 96 different example Configuration.h files in the current Marlin source tree, only one machine - Micromake/C1 - has that feature enabled.

You can test whether your Marlin controller supports inch mode by typing G20 in the console window.  Mine does not:
[image]

To enable inch mode for your controller, you would need to get the Marlin source code from GitHub, find the Configuration.h file for your machine, remove the // comment characters at the beginning of the line "//#define INCH_MODE_SUPPORT", recompile the firmware and download it to your controller.  Even then, it is possible that there might be problems, because inch mode is so rarely used with Marlin that it probably has not been tested extensively.

ac2678, 2019-06-25:
That’s not the issue. I have inch mode support enabled, the controller/machine work correctly in inches, accurately. I make parts all the time to accurate dimensions in inches.  As noted by the original poster, the only issue is that the cnc.js UI displays the axis positions in mm regardless of the mode it’s toggled to. 

For example, I can issue G20, successfully entering inch mode. The “axis” widget shows inches as the unit and x,y,z positions of (0,0,0). I can issue a G0 X1.0 Y1.0 to my controller and the machine will move 1 inch in the X and Y. However, the display on cnc.js will now read (25.4, 25.4, 0).  

If I were to then issue an M114 command (in marlin this requests the machine position from the controller, not sure if this is true for GRBL etc..) and the controller will respond saying it’s at (1,1,0). There simply seems to be a scalar unit conversion missing in the cnc.js front end that shows the machine position. I hope that’s a little more clear. Thanks.

MitchBradley, 2019-06-25:
Yes, it is now clear.  The original poster said, in a followup, that they are using Marlin 1.1.8, so presumably inch mode could not be enabled in the firmware.  They also said "g20 is enabled", but that is ambiguous as to whether "inch mode is enabled in the Marlin firmware configuration" or "g20 was clicked in the cncjs Axis widget". Given 1.1.8, the latter is more likely.
I just configured INCH_MODE_SUPPORT in my Marlin controller, so I am now able to reproduce the problem, and am investigating.
All this illustrates just how much time is typically wasted by brief problem reports that give too little contextual information.

MitchBradley, 2019-06-25:
@ac2678 - You said that you did (essentially) "G20 .. G0 X1 Y1 .. M114" and the reported position was (1,1,0).  When I try that (with my INCH_MODE_SUPPORT Marlin), I see:

```
> G20
ok
> G1 X1 Y1
ok
> M114
X:25.40 Y:25.40 Z:0.00 E:0.00 Count X:2032 Y:2032 Z:0
ok
>
```
That suggests, as is confirmed by reading the Marlin source code, that Marlin reports are always in mm.  Do you really see reports in inches?

cheton, 2019-06-26:
Fixed in PR https://github.com/cncjs/cncjs/pull/498 by @MitchBradley

cheton, 2020-03-07:
Fixed in [1.9.21](https://github.com/cncjs/cncjs/releases/tag/v1.9.21)

closed 2020-03-07 by cheton