cncjs/cncjs #319: AppImage does not work due to trying to modify its read-only filesystem

interfect, 2018-04-28:
#### Description

Running the AppImage starts up a browser window, but it can't connect to the backend because the server appears to crash. The server crashes because it tries to create `./sessions`, but I guess it is inside the read-only AppImage filesystem when it does that and the attempt to make the directory fails.

```
Gtk-Message: Failed to load module "overlay-scrollbar"
2018-04-28T03:10:23.628Z - info: init Loading configuration from "/z/home/anovak/.cncrc"
Error: EROFS: read-only file system, mkdir './sessions'
    at Object.fs.mkdirSync (fs.js:891:18)
    at appMain (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/app/index.js:10595:22)
    at server (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/app/index.js:7316:38)
    at /tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/dist/app/app.multihost.js:132:25
    at Array.forEach (<anonymous>)
    at Object.appMain [as default] (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/dist/app/app.multihost.js:97:31)
    at createServer (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/dist/app/server.js:62:49)
    at serverMain (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/dist/app/server.js:114:9)
    at createServer (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/dist/app/index.js:57:33)
    at module.exports (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/index.js:13:12)
    at Object.createServer (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/app/index.js:7320:32)
    at /tmp/.mount_cncjs-gAlZxJ/app/resources/app/cnc.js:100:26
    at Promise (<anonymous>)
    at F (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/core-js/library/modules/_export.js:35:28)
    at cnc (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/cnc.js:96:12)
    at _callee$ (/tmp/.mount_cncjs-gAlZxJ/app/resources/app/main.js:94:50)
2018-04-28T03:10:24.050Z - error: The multi-host server does not exist: {
    "type": "server",
    "route": "/"
}     at /tmp/.mount_cncjs-gAlZxJ/app/resources/app/node_modules/webappengine/dist/app/app.multihost.js:138:27 id=0, pid=22251
2018-04-28T03:10:24.077Z - info: init Starting the server at http://127.0.0.1:46779
```

#### Versions

- CNCjs: 1.9.15 from https://github.com/cncjs/cncjs/releases/download/v1.9.15/cncjs-app-1.9.15-linux-x86_64.AppImage

#### How Do You Install CNCjs?

- [ ] NPM
- [X] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [X] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [X] Linux (Ubuntu 17.10)

splitn2, 2018-05-21:
Hi, I have the same issue here I tried with both the 1.9.14 and 1.9.15 appimage files and it failed. I tried installing as my own user name and as root and neither worked same symptoms as above. I couldn't see any permissions issues but its a bit above me to troubleshoot further. Thanks, RIchard

cheton, 2018-07-05:
See https://github.com/cncjs/cncjs/issues/347#issuecomment-402619732

cheton, 2018-07-05:
Fixed in PR #348 

You can try the prebuilt image and let me know the result:
https://github.com/cncjs/cncjs/releases/tag/319-320-347-sessions-latest

cheton, 2019-03-07:
Fixed in [1.9.16](https://github.com/cncjs/cncjs/releases/tag/1.9.16)

closed 2019-03-07 by cheton