cncjs/cncjs #322: Error on Debian 9 installation

quiliro, 2018-05-04:
#### Description

When installing via npm. I get an error. This is the installation process I followed:

$ git clone https://github.com/creationix/nvm.git ~/.nvm
Cloning into '/home/fablabo/.nvm'...
remote: Counting objects: 7039, done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 7039 (delta 33), reused 53 (delta 32), pack-reused 6979
Receiving objects: 100% (7039/7039), 2.21 MiB | 509.00 KiB/s, done.
Resolving deltas: 100% (4408/4408), done./      
$ cd .nvm/
$ git checkout `git describe --abbrev=0 --tags`
Note: checking out 'v0.33.11'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 93990ab... v0.33.11
$  cd ..
$ . ~/.nvm/nvm.sh

$ nvm install 6
Downloading and installing node v6.14.2...
Downloading https://nodejs.org/dist/v6.14.2/node-v6.14.2-linux-x86.tar.xz...
######################################################################## 100,0%
Computing checksum with sha256sum
Checksums matched!
Now using node v6.14.2 (npm v3.10.10)
Creating default alias: default -> 6 (-> v6.14.2)

$ nano ~/.bashrc  
fablabo@netlibro:~$ nvm use 6
Now using node v6.14.2 (npm v3.10.10)
$ export NVM_DIR="$HOME/.nvm"
$ [ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm

$ npm install -g cncjs
/home/fablabo/.nvm/versions/node/v6.14.2/bin/cnc -> /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/bin/cnc
/home/fablabo/.nvm/versions/node/v6.14.2/bin/cncjs -> /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/bin/cnc
/home/fablabo/.nvm/versions/node/v6.14.2/bin/cncjs-server -> /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/bin/cnc

> serialport@6.1.1 install /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/node_modules/serialport
> prebuild-install || node-gyp rebuild

/home/fablabo/.nvm/versions/node/v6.14.2/lib
`-- cncjs@1.9.15 
  +-- @trendmicro/react-anchor@0.5.6 
  +-- @trendmicro/react-breadcrumbs@0.5.5 
  +-- @trendmicro/react-buttons@1.3.0 
  +-- @trendmicro/react-dropdown@1.2.3 
  | +-- dom-helpers@3.3.1 
  | +-- uncontrollable@5.1.0 
  | `-- warning@3.0.0 
  +-- @trendmicro/react-iframe@0.3.1 
  +-- @trendmicro/react-interpolate@0.5.5 
  | `-- lodash.omit@4.5.0 
  +-- @trendmicro/react-modal@2.0.5 
  | `-- @trendmicro/react-portal@0.4.3 
  +-- @trendmicro/react-navs@0.11.0 
  +-- @trendmicro/react-notifications@1.0.1 
  +-- @trendmicro/react-paginations@0.6.1 
  +-- @trendmicro/react-popover@0.4.0 
  | `-- rc-align@2.3.6 
[... 727 lines cut ...]
  | `-- stack-trace@0.0.10 
  `-- xterm@3.0.2 

npm WARN react-grid-system@3.1.3 requires a peer of react@^16.0.0 but none was installed.

#### Versions

- CNCjs: 1.9.15
- Node.js: 6.14.2
- NPM: 3.10.10

#### How Do You Install CNCjs?

- [x] NPM

#### CNC Controller

- [X] TinyG

#### Hardware

- [X] Laptop

#### Operating System
- [X] GNU/linux-Libre

quiliro, 2018-05-04:
Updated NPM to version 6.0.0 with npm install npm@latest -g. I still get an error:

$ npm install -g cncjs/home/fablabo/.nvm/versions/node/v6.14.2/bin/cnc -> /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/bin/cnc
/home/fablabo/.nvm/versions/node/v6.14.2/bin/cncjs -> /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/bin/cnc
/home/fablabo/.nvm/versions/node/v6.14.2/bin/cncjs-server -> /home/fablabo/.nvm/versions/node/v6.14.2/lib/node_modules/cncjs/bin/cnc
npm WARN react-grid-system@3.1.3 requires a peer of react@^16.0.0 but none is installed. You must install peer dependencies yourself.

+ cncjs@1.9.15
updated 1 package in 111.995s

cheton, 2018-05-05:
The warning message won't affect your installation, the installation seems to be successful. Are you able to run `cncjs -vv` from the command line?

quiliro, 2018-05-06:
Updated NPM to version 6.0.0 with:
npm install npm@latest -g
It works now!

closed 2018-05-06 by quiliro