cncjs/cncjs #323: Improve API Documentation

proffalken, 2018-05-05:
#### Description

I want to create an integration for HomeAssistant so I can monitor my CNC machine in the same way that I monitor my 3D printer using OctoPrint, however I am unable to find any documentation on the wiki that describe the endpoints available or how to authenticate.

#### Versions

N/A
#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

N/A

#### Operating System

- [x] Not Applicable

cheton, 2018-05-05:
Sorry for that I don't have enough time to write full API documentation recently. To authenticate with CNCjs, you can use cncjs pendant, the sample code is available at https://github.com/cncjs/cncjs-pendant-boilerplate

flynneva, 2020-04-13:
this is exactly the same problem I am trying to solve....has anyone found a solution?

alexives, 2021-01-13:
@proffalken not sure if you're still looking for a solution here, but I also wanted a home assistant solution. So I wrote an cncjs->MQTT connector based on the pendant with documentation for adding the sensors in home assistant. 

https://gitlab.com/alexives/cncjs-mqtt/

Definitely still a work in progress, but it has some of the basics (current state, position, execution information) in place.

proffalken, 2021-01-14:
@alexives perfect timing - although I've moved away from HomeAssistant to Hubitat.com these days, my entire workshop has been on hold for over a year and I'm just in the process of setting it up again!

Thanks for this, it's really appreciated, I'll see if I can do something similar for Hubitat!

Moskus, 2022-01-24:
@alexives I want to try cncjs-mqtt, but as the CNCjs amateur I am I can't find the .cncjs file (and its secret)... where would this be located on my Raspberry Pi?

MitchBradley, 2022-01-24:
.cncrc is in the home directory which is usually /home/pi.cncrc unless you have installed under a different user than the default user "pi"

Moskus, 2022-01-24:
Thanks for the help, @MitchBradley! Finally found it! :)